StevenMHernandez/ESP32-CSI-Tool #1: Unable to obtain CSI data even after connection to another ESP32 AP

kaustubhKVS, 2020-07-14:
STA LOG: After displaying monitor using make monitor
`Toolchain path: /home/kaustubh/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /home/kaustubh/esp/esp-idf/requirements.txt are satisfied.
MONITOR
--- idf_monitor on /dev/ttyUSB0 1552000 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (38) boot: ESP-IDF v3.3.1 2nd stage bootloader
I (38) boot: compile time 00:14:49
I (38) boot: Enabling RNG early entropy source...
I (38) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (39) boot: SPI Flash Size : 4MB
I (39) boot: Partition Table:
I (39) boot: ## Label            Usage          Type ST Offset   Length
I (40) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (40) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (41) boot:  2 factory          factory app      00 00 00010000 00100000
I (42) boot: End of partition table
I (42) boot_comm: chip revision: 1, min. application chip revision: 0
I (42) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x18de0 (101856) map
I (79) esp_image: segment 1: paddr=0x00028e08 vaddr=0x3ffb0000 size=0x03020 ( 12320) load
I (84) esp_image: segment 2: paddr=0x0002be30 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/kaustubh/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (85) esp_image: segment 3: paddr=0x0002c238 vaddr=0x40080400 size=0x03dd8 ( 15832) load
I (92) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x67fe8 (425960) map
0x400d0018: _stext at ??:?

I (242) esp_image: segment 5: paddr=0x00098008 vaddr=0x400841d8 size=0x0bcf4 ( 48372) load
0x400841d8: malloc_internal_wrapper at /home/kaustubh/esp/esp-idf/components/esp32/esp_adapter.c:262

I (271) boot: Loaded app from partition at offset 0x10000
I (272) boot: Disabling RNG early entropy source...
I (272) cpu_start: Pro cpu up.
I (272) cpu_start: Application information:
I (272) cpu_start: Project name:     active-sta
I (273) cpu_start: App version:      c9b5dc9
I (273) cpu_start: Compile time:     Jul 15 2020 00:14:46
I (273) cpu_start: ELF file SHA256:  d1d07a2eafeadc4f...
I (274) cpu_start: ESP-IDF:          v3.3.1
I (274) cpu_start: Starting app cpu, entry point is 0x40081120
0x40081120: call_start_cpu1 at /home/kaustubh/esp/esp-idf/components/esp32/cpu_start.c:269

I (0) cpu_start: App cpu up.
I (275) heap_init: Initializing. RAM available for dynamic allocation:
I (275) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (276) heap_init: At 3FFB8F38 len 000270C8 (156 KiB): DRAM
I (276) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (277) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (277) heap_init: At 4008FECC len 00010134 (64 KiB): IRAM
I (278) cpu_start: Pro cpu start user code
I (288) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (362) wifi: wifi driver task: 3ffc0770, prio:23, stack:3584, core=0
I (363) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (363) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (388) wifi: wifi firmware version: ac331d7
I (389) wifi: config NVS flash: enabled
I (389) wifi: config nano formating: disabled
I (389) wifi: Init dynamic tx buffer num: 32
I (389) wifi: Init data frame dynamic rx buffer num: 32
I (389) wifi: Init management frame dynamic rx buffer num: 32
I (390) wifi: Init management short buffer num: 32
I (390) wifi: Init static rx buffer size: 2212
I (391) wifi: Init static rx buffer num: 10
I (391) wifi: Init dynamic rx buffer num: 32
I (489) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
I (491) wifi: mode : sta (30:ae:a4:99:6b:30)
[... 81 lines cut ...]
I (395) wifi: Init management short buffer num: 32
I (395) wifi: Init static rx buffer size: 2212
I (396) wifi: Init static rx buffer num: 10
I (396) wifi: Init dynamic rx buffer num: 32
I (514) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
I (515) wifi: mode : softAP (30:ae:a4:9c:e2:41)
I (519) wifi: Total power save buffer number: 16
I (519) wifi: Init max length of beacon: 752/752
I (520) wifi: Init max length of beacon: 752/752
I (521) wifi: Set ps type: 0

I (522) Active CSI collection (AP): softap_init finished. SSID:avatarmeherbaba password:7777777777
I (675) wifi: new:<8,1>, old:<8,0>, ap:<8,1>, sta:<255,255>, prof:8
I (677) wifi: station: 30:ae:a4:99:6b:30 join, AID=1, bgn, 40U
I (700) Active CSI collection (AP): station:30:ae:a4:99:6b:30 join, AID=1
I (731) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
I (27556) wifi: new:<8,1>, old:<8,1>, ap:<8,1>, sta:<255,255>, prof:8
I (27557) wifi: station: 30:ae:a4:99:6b:30 join, AID=1, bgn, 40U
I (27578) Active CSI collection (AP): station:30:ae:a4:99:6b:30 join, AID=1
I (27685) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
`

------------------------------------------ END------------------------------------------------------

Please help me out here. I am willing to co-ordinate closely with you, I have acquired hardware just for executing your work of which I am a big fan of !! Thank You!!

EffronLu, 2020-08-28:
MakeFiles/__idf_main.dir/main.c.obj   -c ../main/main.c
../main/main.c:16:10: fatal error: ../../_components/nvs_component.h: No such file or directory
 #include "../../_components/nvs_component.h"
------------------------------------------ END------------------------------------------------------
When I  flash onto the device, I found this problem. Can you help me? Thank you!!

StevenMHernandez, 2020-08-28:
@kaustubhKVS We discussed a solution for this through email. If we keep terminals open for both AP and STA, we can look for any errors or any CSI data output. In some cases, you may need to press the reset button on the AP or the STA button to ensure the devices are able to communicate successfully.

StevenMHernandez, 2020-08-28:
@EffronLu Your question is unrelated to this issue. Does the file exist based on the relative path? i.e. what directory are you running `make flash` in?

closed 2020-08-28 by StevenMHernandez