StevenMHernandez/ESP32-CSI-Tool #2: CSI capture method

samehkhalfaoui, 2020-11-30:
Hey, thanks for the tool, i have a question about your tool. Do we need to create a UDP traffic in order to capture CSI as recommended by NEXMON and WI-ESP tools.

StevenMHernandez, 2020-11-30:
Hello. The [active_sta](https://github.com/StevenMHernandez/ESP32-CSI-Tool/tree/master/active_sta) sub-project automatically creates this required traffic. However, it is possible to generate the traffic from a non-ESP32 too.

StevenMHernandez, 2020-12-01:
I will close this issue for now. Let me know if you have any other issues.

closed 2020-12-01 by StevenMHernandez

roger-, 2021-02-22:
Related question: I'm trying to test this using a single ESP32. I have an active_sta instance connected to an AP but I'm not seeing any CSI data. I tried running a netcat UDP server and modified the socket code to connect to the correct server IP/port and I see traffic (777...) but still no CSI data. 

Any ideas?

StevenMHernandez, 2021-02-23:
The active_ap is the one that is typically set to receive CSI. What are your settings with `make menuconfig`? 

Do you have an example of the UDP server code that you built? There are some cases where CSI is actually not sent with the transmitter. It could depend on the packet rate as well as the transmitter hardware.

What is the `777` you are referencing?

roger-, 2021-02-23:
Oh I thought the station could receive it too. Is that not the case, and if so what's the best configuration for a single ESP32?

My settings are as per the readme. 

The server is just the command `netcat -ul 2223` and it gets some version of `data` from sockets_component.h (mangled for some reason). I noticed [this fork](https://github.com/huyrua291996/ESP32-CSI-Tool/commits/master) made some similar changes. 

The AP is just my regular router. How do I configure the packet rate?

roger-, 2021-02-23:
I just noticed [this commented line](). Enabling it gives me some output in sta mode:

```
CSI_DATA,STA,34:97:F6:96:51:50,-42,11,1,6,0,1,1,0,1,0,0,-93,0,8,0,52190062,0,114│····················································································
,0,0,52.406875,384,[114 -96 6 0 -18 38 -18 38 -18 38 -18 36 -18 33 -17 28 -16 24│····················································································
-15 19 -14 15 -12 9 -10 4 -7 0 -4 -4 -3 -7 -1 -10 0 -12 2 -13 3 -13 3 -14 3 -13│···················································································· 
1 -13 -1 -14 -3 -14 -5 -17 -5 -20 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 2│····················································································
3 -4 21 0 18 3 14 6 10 7 5 8 0 7 -4 6 -7 4 -11 3 -15 2 -17 0 -19 0 -21 0 -22 0 -│····················································································
22 0 -23 2 -24 5 -23 7 -23 10 -21 13 -20 18 -20 22 -19 25 -18 29 -17 33 ]
```

Didn't even need to run the server. Lots of zeros though, not   sure if the data is okay.

StevenMHernandez, 2021-02-23:
I was just about to mention that you might need to uncomment a line (https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/master/active_sta/main/main.c#L146). 

Glad it works. Zeros should be null subcarriers. It was disabled because it can slow down the esp32 in some cases.

If you have a good reproducible method for using netcat, you can make a pull request with steps for the Readme, I would be very happy to add this information. :)

roger-, 2021-02-23:
I'm getting CSI data even without the server -- is that expected? Would I get more CSI data with it?

Another question: is there any reason why I wouldn't want to just send the CSI over the network in sta mode (vs over UART)? 

Sure I can collect my findings in a pull request!

StevenMHernandez, 2021-02-23:
Without the server? Are you saying without the netcat server?
You can try to get data over the network, but that will require some more work. It actually might be a really good idea to evaluate if we can get the same or quicker speeds over the network. However, remember, more network usage means less bandwidth for receiving CSI data. So there will need to be a balance.

roger-, 2021-02-23:
Yes without the netcat server.  Assume that's just due to latent traffic. 

I'm thinking a simple UDP broadcast would be easy to add. Might take a stab at it.