StevenMHernandez/ESP32-CSI-Tool #3: How to adjust TX rate?

ZouXiang2019, 2020-12-10:
I have read your paper about the IOT based WIFI sensing. I want to know how to adjust the TX rate in active_sta project. Thank you very much!

StevenMHernandez, 2020-12-10:
Please see item number 8 in https://github.com/StevenMHernandez/ESP32-CSI-Tool#configuration-esp-idf

closed 2020-12-10 by StevenMHernandez

ZouXiang2019, 2020-12-12:
I adjust TX from 100 to 1000, but the received data at Rx is still 20. Why？
I flashed the active_ap and active_sta on two different esp32 devices.
Thank you very much!

ZouXiang2019, 2020-12-12:
additionally, when I adjust the Baud rate to 1552000, the system make error: A fatal error occurred: Failed to set baud rate 1552000. The driver may not support this rate.

StevenMHernandez, 2020-12-12:
Setting a high baud rate is very important to achieve high RX rate. I am guessing that there is something with your hardware/software (computer/OS) which prevents this baud rate. I use macOS, so that is what I have been able to test with. With the information I have, I cannot say why it is not working on your system. If you can do more debugging to track down any issues, I would like to hear what you can find though.

ZouXiang2019, 2021-01-05:
Finally, I bought a new ESP device, TTGO T8 V1.7, and the Baud rate can adjust to 1552000. I compared the schematic of the two devices in detail, the difference is the programming chip. The problem is solved by replacing cp2102 chip with cp2104. Next, I measured the RX rate at AP mode on Windows and Raspberrypi, but the RX rate is still very low at about 50HZ. I think that using macOS may improve performance. Inevitably, the RX rate can achieve 240HZ when I programming the esp32 devices with macOS.
However, the RX rate is still low when I only save CSI data to the SD card (about 50HZ).  How to solve this problem? On the other hand, how to measure the RX rate into 600HZ as your paper shows?

StevenMHernandez, 2021-01-09:
The difference between the cp2102 chip and the cp2104 chip is an interesting observation. Thank you for sharing this information.

Because the Serial baud rate is one bottleneck, to achieve higher data rates, you will need to do most processing on the ESP32. By performing computations on the device, you don't need to transmit data over Serial and thus higher data rates should be possible.