StevenMHernandez/ESP32-CSI-Tool #4: Support to IEEE 802.11 protocol

ENB213-xiaoh, 2021-01-05:
Hi Dr.Hernandez

Thank you for making this amazing tool in advance. I'm wondering which IEEE802.11 protocols this tool supports. As I believe Intel 5300 and Atheros only support HT IEEE802.11n and Nexmon supports all protocols with limits. I didn't see any of it in your paper. That would be perfect if it supports all protocols. Please advise. Thanks.

StevenMHernandez, 2021-01-09:
What specific protocols are you looking for?

You say that Intel 5300 and Atheros are only High-throughput. 
This tool _should_ allow HT and non-HT as well as some other settings. Does this table help? https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-channel-state-information

ENB213-xiaoh, 2021-01-25:
Hi. I got to come back and see if this tool works for me. So, I need to collect csi from beacon frame and prob response. Does it work for me? I think it is from the table, am I right? thanks

StevenMHernandez, 2021-02-02:
@ENB213-xiaoh I can try to test if you have a script example of what you are trying to figure out. For example, how are you generating the beacon frames (program and hardware).

closed 2021-02-16 by ENB213-xiaoh