StevenMHernandez/ESP32-CSI-Tool #5: baud rate 1552000 does not work

Jiachen-Zhang, 2021-02-17:
Connecting.....
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a8:03:2a:68:69:bc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 1552000
Changed.
Configuring flash size...

A fatal error occurred: Failed to write target memory. Only got 1 byte status response.
make: *** [flash] Error 2

StevenMHernandez, 2021-02-17:
Have you tried any other baud rates? Does flashing the esp32 work in general?

Jiachen-Zhang, 2021-02-18:
> Have you tried any other baud rates? Does flashing the esp32 work in general?

baud rate 115200 works and other buad rates can be used to flash but the output is wrong, even if I've set the same rate to read the serial port...

StevenMHernandez, 2021-02-18:
To help solve this, I need more information. I need to know more details about the problem including what steps you tried to solve the issue (Did you find anybody else with a similar problem with the esp32?). From my perspective, I am looking at the problem blind. You mention that the output is "wrong", but there are many ways that the output could be wrong.
What is wrong about the output with baud rate of 115200?

Thanks.

Jiachen-Zhang, 2021-02-21:
> To help solve this, I need more information. I need to know more details about the problem including what steps you tried to solve the issue (Did you find anybody else with a similar problem with [...]

It works when I use baud rate of 115200.
But for other baud rates, the output is garbled, it should be that the baud rate setting does not take effect, causing the encoding and parsing error

StevenMHernandez, 2021-02-21:
Oh ok. We have been seeing this issue. Do you happen to use a windows computer? I am unable to test with windows myself. Can you try with a baud rate of 1500000 and report back? Alternatively, try 1000000.

roger-, 2021-02-22:
1552000 failed for me too during flashing (in Ubuntu):

```
Running stub...
Stub running...
Changing baud rate to 1552000
Changed.
Configuring flash size...

A fatal error occurred: Timed out waiting for packet header
```

I assumed at first you meant 115200 (which does work).

StevenMHernandez, 2021-02-22:
On linux, I have tried `1500000` and it seems to work. If this is true (maybe you can confirm?), I should update the readme to match.

roger-, 2021-02-22:
Nope, same issue. 

Any reason for the higher baud rate? It flashes and runs with 115200 (but I'm having other issues getting CSI data).

StevenMHernandez, 2021-02-23:
Higher baud rate is useful to allow more data through serial. More data throughput ensure that CSI frames aren't dropped due to buffer overflow issues. (In general this translates to: higher baud rate means more CSI throughput as discussed in our paper http://www.people.vcu.edu/~ebulut/wowmom20-WiFiSensing.pdf)

How many places did you set the baud rate? There should be 3 distinct places in `make menuconfig` where you set this value.

Are you sure the time out is not related to you needing to press and hold the "FLASH" button. Some ESP32s require this extra step (https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/)

roger-, 2021-02-23:
I think I set the baud in all the right places. My dev board flashes correctly at other rates, not sure if it that issue is baud dependent. 

I also just upgraded to the latest IDF because I was getting a weird message before (`wpa: PTK has been installed, it may be an attack`) that might be a bug in the recommended IDF.

StevenMHernandez, 2021-02-23:
Actually when I was using the higher baud rate on Linux, I was using stty (as allowed here: https://github.com/wertarbyte/coreutils/blob/master/src/stty.c#L1772). But maybe it won't work with the ESP-IDF. You can try one of the lower values until it works: 1152000, 921600, and 576000 are good candidates I think

I don't know about the PTK thing. What version of IDF did you upgrade to?

roger-, 2021-02-23:
Thanks I'll give it another shot and see. 

I upgraded to the latest stable version (4.1). The changes were trivial, think I just had to comment out an `include` statement. 

The PTK thing might have been related to a KRACK mitigation bug, I read. Might not have been necessary to upgrade though.

Jiachen-Zhang, 2021-02-23:
One more thing ...

Jiachen-Zhang, 2021-02-23:
One more thing...
The packet rate is at about only 25 every second, even if I commented on the delay part in sockets_component.h.
I used the default csi output format to write into the serial port.
Is it caused by the low baud rate or other reason ( I'm not good at ROS development and thanks for your help again

StevenMHernandez, 2021-02-23:
@Jiachen-Zhang You are correct. You will need high baud rate for high packet rate. Your welcome, I am glad you are working with my tool :)

Jiachen-Zhang, 2021-03-01:
I was able to change the different baud rate by keep 
`Serial flasher config > 'make monitor' baud rate ` be `115200` and then successfully flash into it!

I think I can use `make monitor` after flashing and with changeing this parameter (As I read the serial port by python with the same baud rate 921600

StevenMHernandez, 2021-07-29:
I need to figure out what baud rate works best across platforms. It sounds like 921600 worked in this case, but will other people have problems with that? 
After figuring out a better baud rate, the documentation should be updated to reflect the new value.

aaeshaz, 2021-08-09:
I am using MacOS. I used the baud rate 1843200 as per [x](https://opensource.apple.com/source/IOSerialFamily/IOSerialFamily-11/IOSerialFamily.kmodproj/IOSerialBSDClient.cpp.auto.html) and it worked! had to use stty to set the terminal setting to get it to work. I haven't verified if the output is as expected though.

xiaokangkang6, 2023-01-13:
> Higher baud rate is useful to allow more data through serial. More data throughput ensure that CSI frames aren't dropped due to buffer overflow issues. (In general this translates to: higher baud [...]

Excuse me, which three places, I only see two places to fill in 1552000.

StevenMHernandez, 2023-01-15:
@xiaokangkang6 I think the information from that message is dated (the message was from 2021). In the current version of esp-idf, it should only be two places. You are correct. Please see (https://github.com/StevenMHernandez/ESP32-CSI-Tool/#configuration-esp-idf) 

I will also mentioned that the old method was `make menuconfig` and the current method is `idf.py menuconfig`.

satya2891, 2023-09-08:
@StevenMHernandez 
1. How to reduce the number of packets recieved?
2. By changing the tickrate it affects the packets recieved?
Can u pls clarify this.
Currently I am getting at around 1000 packets / min. I want to reduce it to 100 - 200.

mtijan, 2024-03-08:
C:\Espressif\frameworks\esp-idf-v4.4.6\esp32-csi-tool\active_sta>idf.py flash monitor
Executing action: flash
Serial port COM9
Connecting.......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Running ninja in directory C:\Espressif\frameworks\esp-idf-v4.4.6\esp32-csi-tool\active_sta\build
Executing "ninja flash"...
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/Espressif/frameworks/esp-idf-v4.4.6/esp32-csi-tool/active_sta/sdkconfig
-- App "active-sta" version: cce5b1a
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/esp32-csi-tool/active_sta/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/soc/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Espressif/frameworks/esp-idf-v4.4.6/components/app_trace C:/Espressif/frameworks/esp-idf-v4.4.6/components/app_update C:/Espressif/frameworks/esp-idf-v4.4.6/components/asio C:/Espressif/frameworks/esp-idf-v4.4.6/components/bootloader C:/Espressif/frameworks/esp-idf-v4.4.6/components/bootloader_support C:/Espressif/frameworks/esp-idf-v4.4.6/components/bt C:/Espressif/frameworks/esp-idf-v4.4.6/components/cbor C:/Espressif/frameworks/esp-idf-v4.4.6/components/cmock C:/Espressif/frameworks/esp-idf-v4.4.6/components/coap C:/Espressif/frameworks/esp-idf-v4.4.6/components/console C:/Espressif/frameworks/esp-idf-v4.4.6/components/cxx C:/Espressif/frameworks/esp-idf-v4.4.6/components/driver C:/Espressif/frameworks/esp-idf-v4.4.6/components/efuse C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp-tls C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp32 C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_adc_cal C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_common C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_eth C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_event C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_gdbstub C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_hid C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_http_client C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_http_server C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_https_ota C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_https_server C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_hw_support C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_ipc C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_lcd C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_local_ctrl C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_netif C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_phy C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_pm C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_ringbuf C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_serial_slave_link C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_system C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_timer C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_websocket_client C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_wifi C:/Espressif/frameworks/esp-idf-v4.4.6/components/espcoredump C:/Espressif/frameworks/esp-idf-v4.4.6/components/esptool_py C:/Espressif/frameworks/esp-idf-v4.4.6/components/expat C:/Espressif/frameworks/esp-idf-v4.4.6/components/fatfs C:/Espressif/frameworks/esp-idf-v4.4.6/components/freemodbus C:/Espressif/frameworks/esp-idf-v4.4.6/components/freertos C:/Espressif/frameworks/esp-idf-v4.4.6/components/hal C:/Espressif/frameworks/esp-idf-v4.4.6/components/heap C:/Espressif/frameworks/esp-idf-v4.4.6/components/idf_test C:/Espressif/frameworks/esp-idf [... cut ...]
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
��������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������

sorry, but can you tell me this is error or no?

mtijan, 2024-03-08:
hai, i can not found this 
Component config > Common ESP32-related > Channel for console output > Custom UART

i just find 
Component config > Common ESP-related >

there is something wrong or not i don't know, but i use 
version (v4.4) of ESP-IDF.

mtijan, 2024-05-13:
i'm know what is problem, but i've question, how do I set the sample rate for esp32, is it by changing the 

Component config > FreeRTOS > Tick rate (Hz) > 1000

Ash96wani3, 2025-02-18:
> i'm know what is problem, but i've question, how do I set the sample rate for esp32, is it by changing the
> 
> Component config > FreeRTOS > Tick rate (Hz) > 1000
> 
> thanks

did you got it ? same issue this side

mtijan, 2025-02-18:
Got it