StevenMHernandez/ESP32-CSI-Tool #7: Can we obtain the Power Delay Profile from the ESP32 Wifi?

rtmtree, 2021-03-08:
I've been using this repo to obtain CSI data successfully. But I'm curious if the data can be applied more.

Power delay profile (PDP) is a time-domain measurement of signal power. It can be applied to many fields of work related to multi-path transmission.
 e.g. 
[Fig 1](https://i.imgur.com/fxcUAbe.png) 

(ref : gaussianwaves)
Let this is the PDP measured at the ESP32 as a receiver. If we move Access point further away , the P (blue line) should be at further time ( more right on x-axis) .

Now, I'm struggling with finding a way to obtain the Power Delay Profile from ESP32. Surely, I attached an external antenna to the ESP32 to enhance its signal precision.  

First, I discovered that the Power Delay Profile (PDP) can be resulted from the Inverse fast Fourier transform (IFFT) of CSI data. But, the thing is the CSI that can be worked that way should be aligned in frequency domain so it can be turned into the time domain (PDP) . I coded with 'esp-idf' and obtain CSI data from callback function as many work does. Unfortunately, the CSI obtained from ESP32 is in subcarrier format and contains some null.

[Fig 2](https://i.imgur.com/QbMEhNd.png)

 So, I need to convert ESP32 RAW CSI (Fig 2) (that is in subcarrier domain) to frequency domain CSI which can probably be done by some Frequency Matching algorithm. So , I posted this to ask if there is already a way to match subcarriers to frequency or to **get a precise  Power Delay Profile**. May you share to be a big contribution to my work, please?

PS:
I know that subcarrier represents a frequency. But I don't know exactly which subcarrier represents which frequency.
And also there are many subcarriers (around 12) that always return 0. So, I want a document or a matching algorithm to tell me i-th subcarrier = what frequency band. and Also eliminate those unused subcarriers.

StevenMHernandez, 2021-03-12:
[image]

Here is a table showing which subcarriers from the ESP32-CSI-Tool correspond to which subcarriers in the frequency domain. You have a center subcarrier frequency (0) and then other subcarrier frequencies to the left (-) and the right (+) of the center frequency.

Null subcarriers (subcarrier # 0 as well as the 12 subcarriers you identified) are expected to be 0. This is an artifact of the use of FFT/IFFT which requires the number of subcarriers to be `2^N`. Because there are really only 52 subcarriers with data, FFT/IFFT increases the size to be `64` by filling with these null subcarriers.

yujianyuanhaha, 2021-06-28:
@StevenMHernandez meanwhile, does the CSI phase reliable? does the phase information include unknown offset and tweaks? the delay profile require precise phase information, I do see in some of your work the amplitude is pretty stable and good to go.

StevenMHernandez, 2021-06-28:
The tool does not directly add any information about the offsets in phase. This is something that could be added after receiving the raw phase data.

closed 2021-07-23 by StevenMHernandez