StevenMHernandez/ESP32-CSI-Tool #8: CSI with packet payload

ENB213-xiaoh, 2021-03-11:
Hi All

I'm wondering if it is possible to get the packet payload along with CSI. Because I need the sequence number. I saw some other small projects that did wifi sniffing and get payload, so I want to know if I can combine these two projects. Please kindly advise

ENB213-xiaoh, 2021-03-11:
Ok. Never mind. I have made it. Next is replacing one item in the control block to PN.

StevenMHernandez, 2021-03-12:
Good idea! What did you change in the code to make it work? It would be really nice to be able to add that feature to the library (as a option for others).

What is PN?

ENB213-xiaoh, 2021-03-12:
So, PN is the sequence number. The wifi chipset uses it to count the number of transmitted frames. I think you already know it. It resides in the payload, contains two bytes right after the transmitter and source mac address. So, I basically copied a promiscuous mode callback function from some other sniffer codes.

StevenMHernandez, 2021-03-12:
Were you able to write some clean code for the implementation? I don't have time to implement this into the project at this time, but I would be so happy if you could try creating a pull request with this. The feature would very helpful to other users :)

ENB213-xiaoh, 2021-03-12:
Okay. My pleasure. I can see your email address. So, I will email you what I have done precisely, maybe tonight or tomorrow. Then, we can discuss it, to see if you are okay with the modification.

closed 2021-03-19 by ENB213-xiaoh

tobiaswer, 2022-05-31:
Hi, I'm interested in receiving the wifi sequence number along with the CSI as well. Can you share how you were able to obtain it? Thanks!

ENB213-xiaoh, 2022-05-31:
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"
#include "freertos/event_groups.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_http_server.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "math.h"

#include "../../_components/nvs_component.h"
#include "../../_components/sd_component.h"
#include "../../_components/csi_component.h"
#include "../../_components/time_component.h"
#include "../../_components/input_component.h"

static void extract_macaddr(char *addr, uint8_t* data, uint16_t offset)
{
    sprintf(addr, "%02X:%02X:%02X:%02X:%02X:%02X",
            data[offset+0],
            data[offset+1],
            data[offset+2],
            data[offset+3],
            data[offset+4],
            data[offset+5]);
}
// #define LOG_TAG "CSI-DATA"

static void wifi_monitor_cb(void *recv_buf, wifi_promiscuous_pkt_type_t type)
{
    if(type != WIFI_PKT_MGMT) return; // Management frames
    wifi_promiscuous_pkt_t *recv_pkt = (wifi_promiscuous_pkt_t *)recv_buf;

    // Extract MAC address and print out the result
    if (recv_pkt->rx_ctrl.rate == 0) return; // Exclude IEEE802.11b
    char addr[] = "00:00:00:00:00:00";
    extract_macaddr(addr, recv_pkt->payload, 10);
    uint16_t offset = 22;

    outprintf("CSI_PN,");
    outprintf("%s,", addr);
    outprintf("0X%02X%02X,", recv_pkt->payload[offset+0], recv_pkt->payload[offset+1]);
    outprintf("%02X%02X", recv_pkt->payload[offset+2], recv_pkt->payload[offset+3]);
    outprintf("%02X%02X", recv_pkt->payload[offset+4], recv_pkt->payload[offset+5]);
    outprintf("%02X%02X", recv_pkt->payload[offset+6], recv_pkt->payload[offset+7]);
    outprintf("%02X%02X", recv_pkt->payload[offset+8], recv_pkt->payload[offset+9]);
    outprintf("\n");
    sd_flush();
    vTaskDelay(0);
    // ESP_LOGI(LOG_TAG, "MAC: %s, RSSI: %d, PN: 0x%x%x", addr, recv_pkt->rx_ctrl.rate, recv_pkt->payload[22], recv_pkt->payload[23]);
}

void passive_init() {
    tcpip_adapter_init();
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_NULL));
    ESP_ERROR_CHECK(esp_wifi_start());

    const wifi_promiscuous_filter_t filt = {
            .filter_mask = WIFI_PROMIS_FILTER_MASK_MGMT
    };

    int curChannel = 3;

    ESP_ERROR_CHECK(esp_wifi_set_promiscuous_rx_cb(wifi_monitor_cb));
    esp_wifi_set_promiscuous(true);
    esp_wifi_set_promiscuous_filter(&filt);
    esp_wifi_set_channel(curChannel, WIFI_SECOND_CHAN_NONE);
}

void app_main() {
    nvs_init();
    sd_init();
    passive_init();
    csi_init("PASSIVE");
    input_loop();
}

Hi, tobiaswer. This is the modified main code under passive mode I used to get SN with CSI. I didn't look into the station or AP modes. You will get 2 lines of information for each CSI sequence, one line tagged with "CSI_DATA" is the collected CSI, and another line tagged with "CSI_PN" is the sequence number of the corresponding packet.