StevenMHernandez/ESP32-CSI-Tool #10: Error when running in Station mode

Tremito96, 2021-03-26:
[image]
[image]
This error appeared when running in station mode. but the AP mode works fine with no issue. This error occurs in STA mode only. What can i do?

StevenMHernandez, 2021-03-27:
What version of esp-idf are you using?

Tremito96, 2021-03-27:
> 
> 
> What version of esp-idf are you using?

Tremito96, 2021-03-27:
> 
> 
> What version of esp-idf are you using?

V 4.2

StevenMHernandez, 2021-03-27:
Yeah, in its current state, the codebase is designed for v3.3.1 (It should be mentioned in the README: https://github.com/StevenMHernandez/ESP32-CSI-Tool#installation)

Can you use that version?

(Note to self: I need to update this project to use more recent esp-idf versions.)

Tremito96, 2021-03-28:
V3.3.5 worked fine. Thank you very much. And i tried connecting to my smartphone's hot spot and also my home router. it says it connected but did not output any CSI info. Does this mean that my phone and router does not send csi information? (Welll i didnt have a second ESP32 to test it with two ESP32s)

StevenMHernandez, 2021-03-28:
Try uncommenting this: https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/master/active_sta/main/main.c#L146

Tremito96, 2021-03-29:
> 
> 
> Try uncommenting this: https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/master/active_sta/main/main.c#L146

yes. that was the problem. thank you very much