StevenMHernandez/ESP32-CSI-Tool #11: Update to more recent version of ESP-IDF

StevenMHernandez, 2021-03-27:
Upgrade to V4.2 (current stable version)

See errors such as #10

Gorgeous2017, 2021-03-30:
Did you working for this now? 

I am getting familiar with developing with esp32 recently and I want to use your CSI Tool. My ESP-IDF version is v4.2. I don't want reset to v3 because I don't know what will happen... So many error when I set up the development environment of v4.2 ;(

So I want to know when this will be developed. Thank you!

StevenMHernandez, 2021-03-31:
Currently, I am not planning to work on this immediately. I have other projects that I need to focus on right now. If anyone is willing to work on this open source project, contributions are always welcome of course!

StevenMHernandez, 2021-04-04:
OK. Here are some of the changes I made a long while ago to update from v3 to v4 (https://github.com/StevenMHernandez/ESP32-CSI-Tool/tree/espidf_v4). 

It should work, but I have not tested it thoroughly as I mentioned. So if you are interested in testing the functionality to make sure it works and fixing any issues, that would be useful to the project. Thanks

pcossutta, 2021-06-25:
I just tested the STA. It works but it connects to an IP that is hardcoded in socket_components.h

StevenMHernandez, 2021-06-25:
Yes. the hardcoded IP is for the AP. Did it cause issues for you? Which version did you test with? Thanks.

I have tested the branch on v4.0 but I have not sat down to make sure it works on higher versions yet.

StevenMHernandez, 2021-07-30:
Version 4.0 is now available the in the master branch.

StevenMHernandez, 2021-11-11:
#39 makes it possible to successfully compile the project with ESP-IDF v4.3.

There are a few more tasks that will need to be completed before v4.3 changes are merged into the master branch.

StevenMHernandez, 2022-01-07:
The project is setup to run with ESP-IDF v4.3.

Thanks @xudesheng for contributing a commit towards this effort 👍

closed 2022-01-07 by StevenMHernandez