StevenMHernandez/ESP32-CSI-Tool #12: Subcarrier Order

RikeshMMM, 2021-04-06:
Whats the order of CSI data being received? Does it follow the same format at the attached image? 

[image]

Im trying to understand the different sub-carrier index (LLTF, HT-LTF) as mentioned in the espressif documentation. It seems that when you receieve HT-LTF signal(s) the len of the data should be 384 bytes but you are only getting 128 bytes from the buffer.

https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/48854bf612785598b75d2b2b654d5abd2be7eaa5/_components/csi_component.h#L52-L59

StevenMHernandez, 2021-06-01:
I missed this, sorry about that. 

I need to add better documentation for this, but essentially, the first 64 subcarriers (128 bytes) are the result from legacy long training field (LLTF). All CSI frames have this but not all CSI frames have (for example) HT-LTF. So to make it simple for new users it just returns those first 128 bytes. 

You are free to increase this number as you see fit!

YXW-123, 2021-12-05:
I have the same problem.Is there a way to know  the order of CSI data being received? （The sub-carrier index corresponding to the CSI data）

StevenMHernandez, 2021-12-07:
@YXW-123 here is a table that shows the sort-order of the subcarriers. After sorting, the null/data subcarriers are laid out like in the figure at the top of this thread.

https://github.com/StevenMHernandez/ESP32-CSI-Tool/issues/7#issuecomment-797199668

 If you have any suggestions on how I could represent this in my README documentation, suggestions would be helpful. I was thinking to create a script to automatically label the null-subcarriers and the data subcarriers, but I am not sure if this the most helpful way to handle this.

YXW-123, 2021-12-09:
Thanks for your help.May I understand it as the 128-byte data (64 sub-carriers) I received has a corresponding serial number of 0 to 31 and -32 to -1?If so，It matched well with the  conclusion I consulted,which is the sub-carrier with data are 1 to 26，-1 to -26. But I found that there were only 11 null/data subcarriers(data is 0,0),sub-carrier 0 wasn't null.

StevenMHernandez, 2021-12-09:
That sounds correct.

Note: although the value for subcarrier-0 is not `0`, this subcarrier is mark as potentially having "LO Leakage". (see the figure at the top of this thread). 

Here are some resources that I have found:

[https://cdn.rohde-schwarz.com/pws/dl_downloads/dl_application/application_notes/1ma192/1MA192_7e_80211ac_technology.pdf](https://cdn.rohde-schwarz.com/pws/dl_downloads/dl_application/application_notes/1ma192/1MA192_7e_80211ac_technology.pdf) says:

> The subcarriers that are not used for transmitting the signal are null subcarriers which are used for DC subcarrier(s) or guard subcarriers. The DC subcarrier (subcarrier 0) is nulled to reduce problems from analog/digital converters and carrier feedthrough.

[https://download.ni.com/evaluation/rf/Introduction_to_WLAN_Testing.pdf](https://download.ni.com/evaluation/rf/Introduction_to_WLAN_Testing.pdf) says:

> Most notable is the use of a null carrier at the center of the channel. This eliminates issues related to the center frequency contaminating the OFDM modulation in that area, namely LO leakage from the mixer. 

So, even though the subcarrier value is not `0`, it is common to ignore the subcarrier or set the value of the subcarrier to `0`.

YXW-123, 2021-12-12:
Thank you for your information about subcarrier 0.

And I found some relevant [information](https://www.electronics-notes.com/articles/connectivity/wifi-ieee-802-11/802-11n.php)，which said that:"In this mode the 802.11n signal is divided into 64 sub-carriers. 4 pilot signals are inserted in sub-carriers -21, -7, 7 and 21. In the legacy mode, signal is transmitted on sub-carriers -26 to -1 and 1 to 26, with 0 being the centre carrier. In the HT modes signal is transmitted on sub-carriers -28 to -1 and 1 to 28."

Unfortunately, in my latest acquisition attempt, I found that the My previous guess（the 128-byte data (64 sub-carriers) I received has a corresponding serial number of 0 to 31 and -32 to -1） was not quite correct.

The latest CSI data I received：
CSI_DATA | AP | my MAC address | -39 | 11 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | -96 | 0 | 6 | 1 | 893615 | 0 | 30 | 0 | 0 | 1.07185 | 128 | [30 | -32 | 1 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 18 | 7 | 18 | 6 | 17 | 7 | 16 | 7 | 17 | 7 | 17 | 8 | 17 | 8 | 17 | 8 | 18 | 9 | 19 | 10 | 20 | 11 | 22 | 12 | 23 | 13 | 22 | 13 | 21 | 12 | 20 | 12 | 21 | 12 | 21 | 11 | 20 | 12 | 20 | 12 | 21 | 11 | 20 | 11 | 21 | 11 | 20 | 12 | 20 | 12 | 19 | 11 | 0 | 0 | 19 | 11 | 19 | 11 | 19 | 11 | 20 | 12 | 20 | 11 | 19 | 11 | 19 | 10 | 20 | 11 | 19 | 10 | 19 | 10 | 19 | 10 | 20 | 10 | 19 | 10 | 20 | 10 | 19 | 10 | 18 | 9 | 17 | 8 | 16 | 7 | 14 | 7 | 14 | 6 | 14 | 6 | 13 | 6 | 14 | 6 | 13 | 6 | 14 | 5 | 14 | 5 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | ]

As your paper "Adversarial Occupancy Monitoring using One-Sided.Through-Wall WiFi Sensing" mentions "For each frame, we receive values for 64 subcarriers where 52 of them contain actual CSI data.” Did this mean that only 52 sets of data you receive are not （0,0）? Or how do you know which 52 of them contain actual CSI data?Could you give me some advice about this ?Thanks in advance！

AsciiPro99, 2022-11-09:
This  is a list of the subcarriers containing no data in a 384 byte long CSI entry.

 NULL_SUBCARRIERS = [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 64, 65, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127,128,129, 130, 131, 246, 247, 248, 249, 250, 251, 252, 253, 254, 255, 256, 257, 258, 259, 260, 261,262,263, 264, 265, 266, 267, 382, 383]
Obviously sets of imaginary and complex so remove them before amp og phase calculations.

StevenMHernandez, 2022-11-09:
Thanks for mentioning these null_subcarriers.

I wonder if we can create a "reference script" for parsing, sorting and labeling the subcarrier number for each real/imaginary number using this table:
[image]

For example, we already have reference scripts such as (https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/master/python_utils/serial_append_time.py) which reads CSI-lines. And we have (https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/master/python_utils/serial_plot_csi_live.py) which transforms the CSI into amplitude/phase. I think we should create a script that reads the incoming CSI lines and then returns a JSON style dictionary for each CSI line.

For example, using the table above, if we have a 128 bytes of CSI, and secondary_channel = 'none', then we can return a string like this:

```
{'LLTF': {'-32': amplitudes[32], '-31': amplitudes[33], '-30': amplitudes[34], ..., '-2': amplitudes[62], '-1': amplitudes[63], '0': amplitudes[0], '1': amplitudes[1], '2': amplitudes[2], ..., '29': amplitudes[29], '30': amplitudes[30], '31': amplitudes[31]}, ''HT-LTF': null, ''STBC-HT-LTF': null}
```
`HT-LTF` and `STBC-HT-LTF` are empty based on the table above. If `signal_mode` is set to `HT`, or if `stbc` is set to true, then these would be non-empty.

Unfortunately I do not have time to work on this currently, but if anyone is interested in building this script, please share a pull request. Thanks!

Sisyphus-pandapia, 2023-07-03:
> 我可以解决这个问题，似乎很有趣。我自己的项目即将结束，所以可能需要等到之后。

Excuse me, has the script about effective subcarrier index been released? Could you please tell me where I can find it? Thank you.