StevenMHernandez/ESP32-CSI-Tool #13: HELP! Unable to get CSI values

zeeshan0309, 2021-05-29:
I'm using an ESP32 as AP, and another ESP32 as STA. I've flashed the respective programs onto both of them but as i monitor the STA's serial output i do not understand what is happening. These are some of the output on serial monitor:

```
I (920) Active CSI collection (Station): connect to ap SSID:myssid password:mypassword
waiting
I (1043) wifi:new:<8,1>, old:<1,0>, ap:<255,255>, sta:<8,1>, prof:1
I (1044) wifi:state: init -> auth (b0)
I (1053) wifi:state: auth -> assoc (0)
I (1073) wifi:state: assoc -> run (10)
I (1125) wifi:connected with myssid, aid = 1, channel 8, 40U, bssid = f0:08:d1:d3:02:d1
I (1126) wifi:security: WPA2-PSK, phy: bgn, rssi: -65
I (1131) wifi:pm start, type: 0

I (1167) wifi:AP's beacon interval = 102400 us, DTIM period = 2
waiting
I (2208) esp_netif_handlers: sta ip: 192.168.4.2, mask: 255.255.255.0, gw: 192.168.4.1
I (2209) Active CSI collection (Station): got ip:192.168.4.2
I (35731) wifi:bcn_timout,ap_probe_send_start
I (65171) wifi:bcn_timout,ap_probe_send_start
I (116584) wifi:bcn_timout,ap_probe_send_start
```

this last phrase  `wifi:bcn_timout,ap_probe_send_start` keeps repeating every few seconds. 
What is happening here? Please help!

StevenMHernandez, 2021-05-31:
Are you getting any useful information from the Active AP serial output?

zeeshan0309, 2021-06-01:
Yes the output at AP side is indeed the required CSI values. Sorry for asking such a silly question lol

StevenMHernandez, 2021-06-01:
Do not worry about it. It is not a silly question. It caused some confusion with other people too. I removed the code-comment which disabled CSI collection for the STA codebase (https://github.com/StevenMHernandez/ESP32-CSI-Tool/commit/31f9f1be9f803e7770e568cd9eb951b4887bbb68) .

Glad it is working for you now.

closed 2021-06-01 by StevenMHernandez