StevenMHernandez/ESP32-CSI-Tool #14: Something went wrong while building the STATION mode

R88624332, 2021-06-03:
[image]
I'm using esp32 as a station and trying to get the CSI from a router, but there seems some errors when building the ACTIVE_STA.  But I can successfully build, flash and monitor it using the ACTIVE_AP and get the correct CSI.  What can I do?  Please help!
```
[3/151] Building C object esp-idf/libsodium/CMakeFiles/idf_component_libsodium.di...dium/crypto_box/curve25519xchacha20poly1305/box_curve25519xchacha20poly1305.c.objFAILED: esp-idf/libsodium/CMakeFiles/idf_component_libsodium.dir/libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_curve25519xchacha20poly1305.c.obj
ccache E:\esp_test_2\.espressif\tools\xtensa-esp32-elf\1.22.0-97-gc752ad5d-5.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DCONFIGURED -DESP_PLATFORM -DGCC_NOT_5_2_0=0 -DHAVE_CONFIG_H -DHAVE_WEAK_SYMBOLS -DIDF_VER=\"v3.3.5\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DNATIVE_LITTLE_ENDIAN -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -IE:/esp_test/esp-idf/components/libsodium/libsodium/src/libsodium/include -IE:/esp_test/esp-idf/components/libsodium/port_include -IE:/esp_test/esp-idf/components/libsodium/libsodium/src/libsodium/include/sodium -IE:/esp_test/esp-idf/components/libsodium/port_include/sodium -IE:/esp_test/esp-idf/components/libsodium/port -Iconfig -IE:/esp_test/esp-idf/components/mbedtls/port/include -IE:/esp_test/esp-idf/components/mbedtls/mbedtls/include -IE:/esp_test/esp-idf/components/esp32/include -IE:/esp_test/esp-idf/components/driver/include -IE:/esp_test/esp-idf/components/esp_ringbuf/include -IE:/esp_test/esp-idf/components/tcpip_adapter/include -IE:/esp_test/esp-idf/components/lwip/include/apps -IE:/esp_test/esp-idf/components/lwip/include/apps/sntp -IE:/esp_test/esp-idf/components/lwip/lwip/src/include -IE:/esp_test/esp-idf/components/lwip/port/esp32/include -IE:/esp_test/esp-idf/components/lwip/port/esp32/include/arch -IE:/esp_test/esp-idf/components/lwip/include_compat -IE:/esp_test/esp-idf/components/lwip/port/esp32/tcp_isn -IE:/esp_test/esp-idf/components/vfs/include -IE:/esp_test/esp-idf/components/esp_event/include -IE:/esp_test/esp-idf/components/log/include -IE:/esp_test/esp-idf/components/efuse/include -IE:/esp_test/esp-idf/components/efuse/esp32/include -IE:/esp_test/esp-idf/components/newlib/platform_include -IE:/esp_test/esp-idf/components/newlib/include -IE:/esp_test/esp-idf/components/freertos/include -IE:/esp_test/esp-idf/components/app_trace/include -IE:/esp_test/esp-idf/components/heap/include -IE:/esp_test/esp-idf/components/soc/esp32/include -IE:/esp_test/esp-idf/components/soc/include -mlongcalls   -mlongcalls -Og -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -std=gnu99 -Wno-old-style-declaration -MD -MT esp-idf/libsodium/CMakeFiles/idf_component_libsodium.dir/libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_curve25519xchacha20poly1305.c.obj -MF esp-idf\libsodium\CMakeFiles\idf_component_libsodium.dir\libsodium\src\libsodium\crypto_box\curve25519xchacha20poly1305\box_curve25519xchacha20poly1305.c.obj.d -o esp-idf/libsodium/CMakeFiles/idf_component_libsodium.dir/libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_curve25519xchacha20poly1305.c.obj   
-c E:/esp_test/esp-idf/components/libsodium/libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_curve25519xchacha20poly1305.c
ccache: error: Failed to create temporary file for esp-idf/libsodium/CMakeFiles/idf_component_libsodium.dir/libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_curve25519xchacha20poly1305.c.obj: No such file or directory
[4/151] Building C object esp-idf/libsodium/CMakeFiles/idf_component_libsodium.di...sodium/crypto_box/curve25519xsalsa20poly1305/box_curve25519xsalsa20poly1305.c.objFAILED: esp-idf/libsodium/CMakeFiles/idf_component_libsodium.dir/libsodium/src/libsodium/crypto_box/curve25519xsalsa20poly1305/box_curve25519xsalsa20poly1305.c.obj 
ccache E:\esp_test_2\.espressif\tools\xtensa-esp32-elf\1.22.0-97-gc752ad5d-5.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DCONFIGURED -DESP_PLATFORM -DGCC_NOT_5_2_0=0 -DHAVE_CONFIG_H -DHAVE_WEAK_SYMBOLS -DIDF_VER=\"v3.3.5\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DNATIVE_LITTLE_ENDIAN -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -IE:/esp_test/esp-idf/components/libsodium/libsodium/src/libsodium/include -IE:/esp_test/esp-idf/components/libsodium/port_include -IE:/esp_test/esp-idf/components/libsodium/libsodium/src/libsodium/include/sodium -IE:/esp_test/esp-idf/components/libsodium/port_include/sodium -IE:/esp_test/esp-idf/components/libsodium/port -Iconfig -IE:/esp_test/esp-idf/components/mbedtls/port/include -IE:/esp_test/esp-idf/components/mbedtls/mbedtls/include -IE:/esp_test/esp-idf/components/esp32/include -IE:/esp_test/esp-idf/components/driver/include -IE:/esp_test/esp-idf/components/esp_ringbuf/include -IE:/esp_test/esp-idf/components/tcpip_adapter/include -IE:/esp_test/esp-idf/components/lwip/include/apps -IE:/esp_test/esp-idf/components/lwip/include/apps/sntp -IE:/esp_test/esp-idf/components/lwip/lwip/src/include -IE:/esp_test/esp-idf/components/lwip/port/esp32/include -IE:/esp_test/esp-idf/components/lwip/port/esp32/include/arch -IE:/esp_test/esp-idf/components/lwip/include_compat -IE:/esp_test/esp-idf/components/lwip/port/esp32/tcp_isn -IE:/esp_test/esp-idf/components/vfs/include -IE:/esp_test/esp-idf/components/esp_event/include -IE:/esp_test/esp-idf/components/log/include -IE:/esp_test/esp-idf/components/efuse/include -IE:/esp_test/esp-idf/components/efuse/esp32/include -IE:/esp_test/esp-idf/components/newlib/platform_include -IE:/esp_test/esp-idf/components/newlib/include -IE:/esp_test/esp-idf/components/freertos/include -IE:/esp_test/esp-idf/components/app_trace/include -IE:/esp_test/esp-idf/components/heap/include -IE:/esp_test/esp-idf/components/soc/esp32/include -IE:/esp_test/esp-idf/components/soc/include -mlongcalls   -mlongcalls -Og -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -std=gnu99 -Wno-old-style-declaration -MD -MT esp-idf/libsodium/CMakeFiles/idf_component_libsodium.dir/libsodium/src/libsodium/crypto_box/curve25519xsalsa20poly1305/box_curve25519xsalsa20poly1305.c.obj -MF esp-idf\libsodium\CMakeFiles\idf_component_libsodium.dir\libsodium\src\libsodium\crypto_box\curve25519xsalsa20poly1305\box_curve25519xsalsa20poly1305.c.obj.d -o esp-idf/libsodium/CMakeFiles/idf_component_libsodium.dir/libsodium/src/libsodium/crypto_box/curve25519xsalsa20poly1305/box_curve25519xsalsa20poly1305.c.obj   -c E:/esp_test/esp-idf/components/libsodium/libsodium/src/libsodium/crypto_box/curve25519xsalsa20poly1305/box_curve25519xsalsa20poly1305.c
ccache: error: Failed to create temporary file for esp-idf/libsodium/CMakeFiles/idf_component_libsodium.dir/libsodium/src/libsodium/crypto_box/curve25519xsalsa20poly1305/box_curve25519xsalsa20poly1305.c.obj: No such file or directory
[16/151] cmd.exe /C "cd /D E:\espressif\ESP32-CSI-Tool-master-two\ESP32-CSI-Tool-... *******************************************************************************"Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
ninja: build stopped: subcommand failed.
The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake --build ." terminated with exit code: 1.
```

StevenMHernandez, 2021-06-14:
This is strange. What is the exact command that you ran? Is there a way to make the error more verbose (something like adding a `--verbose` flag)?

StevenMHernandez, 2021-07-29:
I have not heard any response, so I am going to assume that the issue was resolved on your own, so I will close the issue for now. If you still need help, please reopen or continue the conversation

closed 2021-07-29 by StevenMHernandez