StevenMHernandez/ESP32-CSI-Tool #16: pass the cross session test with ESP32 CSI

yujianyuanhaha, 2021-06-28:
greetings,

thanks for the good work!

@StevenMHernandez  can  ESP32 CSI pass the  cross session test, first list in [reproduction of CSI in same location but different time session ](https://github.com/seemoo-lab/nexmon_csi/issues/158), Is it possible to see the nearly same CSI waveform, from a receiver in the same location(all the environment, i.e. stuff in the house is almost the same) but different time session? So that the CSI can be used for some fingerprint-based application like localization.

Best

Jet

StevenMHernandez, 2021-06-28:
I do not work directly with fingerprint based research, so I do not have any data about this question. 

It sounds like somebody else in the thread that you linked is looking at methods for mitigating these sorts of issues. Is this "cross-session test" a test you came up with? From your results you can see that sometimes the test works and sometimes it does not. Why do you expect that this is happening? Are you certain that nothing in the environment has changed?

yujianyuanhaha, 2021-06-28:
@StevenMHernandez I am pretty sure environment does NOT change, I guess it maybe channel fading [Unstable CSI amplitude](https://github.com/xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src/issues/18),  or chip temperature changes.

Best

Jet

StevenMHernandez, 2021-06-28:
I am a bit confused. What is the goal for this issue? This does not sound like a problem with this tool specifically. Instead it sounds like a general issue with CSI-based fingerprinting. In which case, you would assume that the ESP32 will have very similar issues right?

yujianyuanhaha, 2021-06-28:
@StevenMHernandez Yes, i feel like it is a general issue with CSI-based fingerprinting. I do not have ESP32 yet but not surprise if it cannot pass cross-session check.

closed 2021-06-29 by yujianyuanhaha