StevenMHernandez/ESP32-CSI-Tool #18: Test with Windows

StevenMHernandez, 2021-07-02:
We need to make sure that the project works well with Windows machines. However, I do not have access to a Windows computer to test.

It would be a huge help to this project if anyone in the community can confirm the following:

1. Does the project work well with Windows?
2. Are there any extra steps that need to be done to get it working on Windows?
3. Can you add additional windows-specific information to the `readme.md`?

Here are some issues that seem to be Windows related
* #6 
* #14

Thanks!

zeeshan0309, 2021-07-03:
I am working on Windows10 with this tool. Here are my observations till now

- able to obtain and monitor CSI data live through serial monitor
- for storing/logging these CSI values I used PuTTY serial monitor and it's logging feature. Using this CSI can easily be stored in CSV/text format
- still cant find a way to utilise these CSI values in "realtime"

StevenMHernandez, 2021-07-03:
Thank you for sharing. Did you use the commands like `grep` on the data for real-time filtering of the data (#6 had an issue with that on windows I believe). Does PuTTY give these extra utilities or maybe you downloaded it previously?

Here are some other comments I have for you (which are unrelated to this issue):

> Thanks for bringing up the "realtime" thing. Processing CSI in "realtime" can mean a lot of things. For example: 
> 1. maybe you just want to see the CSI as a plot as the data comes in (for debugging). This would be generic enough to be useful for this project
> 2. Alternatively, some people will want to do data processing and data prediction on the "realtime" data. There are so many different ways to process the CSI that it would not make sense to me (at this time) to add this. If somebody wants to do this, they will need to have the knowledge to engineer their own unique solution. Hmm.. Maybe providing a small proof-of-concept for processing data in realtime could be useful for people to. I will think about this.

zeeshan0309, 2021-07-04:
I didnt use `grep`, but i think that command wont work directly anyways. PuTTY just logs any and everything (including error lines) into text/csv. With PuTTY also i didn't try realtime parsing. 

Actually i am personally concerned with only storing/saving CSI values which can be easily done with putty, and to filter/operate on them later using python, thats why i didn't try realtime filtering.

ahmetanbar, 2021-07-05:
ESP-IDF adds "winpty" command before of "python" for windows at this line. "idf_monitor.py" is run by a command like "winpty python" so that the error "stdout is not a tty" is throwed. To overcome this, the parameter for winpty "-Xallow-non-tty" could be used like that "winpty -Xallow-non-tty python" at that line. Because of this parameter, I think the logs is broken. The new lines are broken. For example, I want to obtain this "CSI_DATA,STA,XX:XX:XX:XX:XX:XX,-56,11,1,6,1,1,1,1,0,0,0,-93,1,8,1,16715674,0,110,0,0,16.937968,384,[110 96 6 0 0 0 0 0 0 0 0 0 12 -8 12 -7 13 -7 13 -7 13 -6 13 -6 13 -6 13 -5 13 -5 13 -4 14 -4 13 -4 13 -3 13 -3 12 -3 13 -3 12 -3 12 -3 12 -4 12 -4 12 -3 11 -4 12 -4 11 -3 12 -3 11 -4 0 0 11 -4 11 -4 11 -4 10 -5 10 -5 10 -5 10 -5 9 -6 10 -6 10 -6 10 -6 10 -6 10 -6 9 -7 10 -7 10 -7 9 -7 9 -8 9 -8 9 -9 9 -8 9 -9 9 -9 9 -9 9 -9 9 -9 0 0 0 0 0 0 0 0 0 0 ],1625492304.85" , but I could just obtain just that "CSI_DATA,STAXX:XX:XX:XX:XX:XX,-56,11,1,6,1,1,1,1,0,0,0,-93,1,8,1,16719468,0,110,0,0,16.943324,384,[110 9". I don't know why but I think it is related with this.

I also have thought removing the adding "winpty" for windows. I try the "python" code directly on msys32, and I see it works. So I remove the "winpty" command from that line. I see that it works clearly. For using grep command line, I take a error "Binary file (standard input) matches". I overcome with "-a" parameter. "make monitor | grep -a "CSI_DATA" > my-experiment-file.csv". It works clearly. For python script command, it works without any error.

Those are my results. But I am not sure about the accuracy of my results. So I can't talk absolutely.

arcgolatkar, 2021-07-09:
The windows equivalent to grep is findstr. I am able to store the CSI data in windows using the following command:
idf.py -p PORT -b BAUD monitor | findstr "CSI_DATA" > my-experiment-file.csv
OR
idf.py -p PORT -b BAUD monitor | python ../_utils/serial_append_time.py > my-experiment-file.csv

StevenMHernandez, 2021-08-11:
This is a very late response, but I wonder if anyone has ideas of how to handle using `grep` for macOS and linux and using `findstr` in windows? Should it be automated somehow, or can we just update the documentation to make note of this?

Any suggestions for updating the documentation through a pull-request are greatly appreciated!

Loki-moon, 2022-01-24:
It's a late answer,
I write a comment because I think it can be helpful to window users who find the problem.

In a window environment, you can use `find` instead of `grep`.

------------------------
> Example

- linux, osx : 
`idf.py monitor | grep "CSI_DATA" > my-experiment-file.csv `

same event 
- windows esp.idf cmd :
(For example, your ESP port is COM12)
```
idf.py monitor | find "CSI_DATA" > my-experiment-file.csv : **NOT WORK** 
idf.py monitor -p com12 | find "CSI_DATA" > my-experiment-file.csv ""WORK"" 
```

Tisha36, 2022-07-09:
My csv file is not being created .I used these commands but didn't work.
I am working on windows 10. ESP-idf-v4.4.1-3
https://github.com/StevenMHernandez/ESP32-CSI-Tool/issues/18#issuecomment-1019824808
[image]