StevenMHernandez/ESP32-CSI-Tool #19: CSI reading process is extremely jerky, stops unexpectedly many times

zeeshan0309, 2021-07-05:
I have faced a sequence of issues over this month, (apparently) related to `timeout`. 

1. Initially, all my readings were perfectly continuous i.e no pauses anywhere even for half a second. Instead of pauses, i used to get some error messages which indicated Brownout/timeout detected (I can't recreate the error messages because I don't get these anymore). 
2. Now I face the issue of of abrupt pauses between readings, but no error messages. Serial monitor shows that the output pauses for varying (~1 second pause to even ~20 seconds pause) amounts of time in between readings. This happens on both AP and STA, but much more on the STA side. These abrupt pause/delays have been so bad lately that in 10 Minutes of collection, only 800-900 readings are registered while ideally i should've gotten atleast 5k readings in that amount of time.

Also, these pauses (almost)never leave any error messages, but on rare occasions this has been logged: 
`I (31603) wifi:bcn_timeout,  ap_probe_send_start`. 

What do i do?

StevenMHernandez, 2021-07-06:
Which version of this project are you using? `git rev-parse HEAD`

Did this happen after updating to a git commit?

I know 'Task watchdog got triggered' happens when `vTaskDelay(0)` is not called within an infinite while loop (or a very long running task). I thought this had been fixed throughout the project.

zeeshan0309, 2021-07-07:
I cloned latest version once again hoping it would resolve the issue but it didn't.

`git rev-parse HEAD` returned `f8f95a5edca2309fd9ac1ff57828acdc8893f05d`.

I haven't updated it up until now so i don't think any updates might have done this.. Today i did clone it once again but pausing problem is still there

[Actually here's a short screen recording of the issue](https://drive.google.com/file/d/1DgmhHq8w64S4JFFmI8oAn3_CpiGFaZLX/view?usp=sharing). As you  can see here, it just stops and resumes on its own and doesn't even throw any error message

StevenMHernandez, 2021-07-09:
That is very strange. What specific ESP32 module are you using? I wish you could narrow down through process of elimination what is specifically causing this issue for you. Because I do not have the same issues, I am not sure what to do for this :(

zeeshan0309, 2021-07-09:
okay. I'm using ESP32 devkit, the one which most people use. I'll see what can i do

arcgolatkar, 2021-07-22:
@zeeshan0309 I am facing the same issue. Did you find any reason why it is happening?

zeeshan0309, 2021-07-22:
@arcgolatkar no unfortunately i haven't found anything which could eliminate the lags. I was thinking perhaps using esp32 as STA and some other commercial WiFi AP, instead of both AP & STA being esp32, may solve it. but i haven't tried this yet...

StevenMHernandez, 2021-07-23:
What is the baud rate that is being used?

StevenMHernandez, 2021-07-25:
OK. I have some ideas. Note for myself: I need to update the project to be better for both 1.) brand new esp32-csi-tool users as well as 2.) power users.

Try the following:

1. Comment out `csi_init()` here (https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/f8f95a5edca2309fd9ac1ff57828acdc8893f05d/active_sta/main/main.c#L146)
2. Make sure the for-loop here (https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/f8f95a5edca2309fd9ac1ff57828acdc8893f05d/_components/csi_component.h#L56) has `int i = 0; i < 128; i++;`

FadedCoder, 2021-07-28:
Facing the same issue with both AP and STA. Although in my case I'm using just a single ESP (TTGO T8) and phone/router. And the delay is much more on STA.

For the first few seconds, it sends a lot of data and then it becomes jerky. Sometimes it'll continuously send data or pause for a few seconds.

Using baud rate 1552000.

StevenMHernandez, 2021-08-02:
I updated the project on the `master` branch (using ESP-IDF v4.0) last week and it sounds like others have tried it and this issue is no longer happening. 

I will likely close this issue soon _unless_ the issue continues to persist for any users.

zeeshan0309, 2021-08-04:
using latest version of tool with baudrate 921600 eliminated the lag and stopping behaviour. Solved

closed 2021-08-04 by zeeshan0309