StevenMHernandez/ESP32-CSI-Tool #20: Error while Flashing

arcgolatkar, 2021-07-08:
I am getting the following error while flashing the ESP32. (shown in img)
I have followed the readme for the menuconfig and have ESP-IDF v4.1. 
Working on a single ESP in station mode and moble's hotspot as AP.  I have changed 'myssid' and 'mypassword' to the mobile hotspot ssid and pwd. Wondering if doing this can connect my ESP to the hotspot ?.
[image]

StevenMHernandez, 2021-07-09:
"A fatal error occurred: Timed out waiting for packet header"

Do you need to hold a "flash" button to flash this ESP32?

arcgolatkar, 2021-07-09:
Yes

StevenMHernandez, 2021-07-10:
Is the issue resolved? Can I close this issue?

closed 2021-07-11 by arcgolatkar