StevenMHernandez/ESP32-CSI-Tool #21: Inconsistent packet recieve rate

aaeshaz, 2021-07-13:
I've set the packet receive rate as 20 packets per second. It seems inconsistent as it receives up to 23 pps in some instances. I use the baud rate of 115200. Is there a way to make set the receive rate as 20 pps throughout? Is this an issue with the ESP generally?

StevenMHernandez, 2021-07-13:
There are always some time-offset with any sort of wireless communications systems (ESP32 and other CSI Tools included). One source could be _Packet Detection Delay (PDD)_ or _Transmission Delay_ or even some other delays caused by the operating system that is computing the pps rate. 

How are you calculating the pps? Is it running on a real-time operating system where events are processed (somewhat) immediately? Alternatively, can you look at the time between each packet using the provided timestamps?

StevenMHernandez, 2021-07-29:
I happened to test with baud rate 115200 and that value is far too low. I suggest baud rates of at least 1000000 which is ten times higher. Specifically, baud rate 1552000 is known to work the best.

If you are unable to get higher baud rates to work you can get some ideas from #5.

I have not heard any response from you, so I am assuming that you solved your problem. Alternatively, I believe you will be able to solve the problem with suggestions from #5. 
So I will close this issue. If you still have issues, please re-open or ask more questions. Thanks.

closed 2021-07-29 by StevenMHernandez

aaeshaz, 2021-08-09:
I am using MacOS. I used the baud rate 1843200 as per [Link](https://opensource.apple.com/source/IOSerialFamily/IOSerialFamily-11/IOSerialFamily.kmodproj/IOSerialBSDClient.cpp.auto.html) and it worked! had to use stty to set the terminal setting as per [Link](https://stackoverflow.com/questions/5539192/how-to-set-the-baud-rate-for-macs-in-a-terminal) though. 

I updated from ESP-IDF from v3.3 to v4.0 and updated the CSI tool accordingly. However, I found an issue is that it doesn't connect to the correct SLAB_USBtoUART port. How to set this to a known USB port in the new version of the esp32-csi-tool?

StevenMHernandez, 2021-08-09:
Woah cool, I have never used a baud rate that was so high.

You used `stty` with `cat` or with `idf.py monitor`?

This should help you: `idf.py flash monitor -p /dev/.../SLAB_USBtoUART_abc123`
mainly the `-p` is what you should need.

aaeshaz, 2021-08-13:
Thank you! that worked well. 

I used stty with cat on a different terminal.