StevenMHernandez/ESP32-CSI-Tool #22: can we use only one esp32 board?

BidyutSaha, 2021-07-15:


BidyutSaha, 2021-07-19:
I am using only one esp32 board.
My setup is like this :
One Host (My PC ) is connected with One Access Point  ( 2.4 GHz , channel 1 ).
I am using the ESP32 as passive csi collector. I set the channel at 1.

I am not getting any output. 

------------------------------------------------------------------------------------------------------------------------------------------------------------

root@bidyut-System-Product-Name:/home/bidyut/ESP32-CSI-Tool/passive# make monitor
Toolchain path: /root/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp-2021r1
Compiler version: 8.4.0
Python requirements from /root/esp/esp-idf/requirements.txt are satisfied.
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6380
ho 0 tail 12 room 4
load:0x40078000,len:14236
load:0x40080400,len:3820
0x40080400: _init at ??:?

entry 0x40080698
I (28) boot: ESP-IDF v4.4-dev-2128-gf45b60b28 2nd stage bootloader
I (28) boot: compile time 01:06:11
I (28) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=171ach ( 94636) map
I (146) esp_image: segment 1: paddr=000271d4 vaddr=3ffb0000 size=038f8h ( 14584) load
I (152) esp_image: segment 2: paddr=0002aad4 vaddr=40080000 size=05544h ( 21828) load
I (161) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=7d9e0h (514528) map
I (348) esp_image: segment 4: paddr=000ada08 vaddr=40085544 size=100d4h ( 65748) load
I (375) esp_image: segment 5: paddr=000bdae4 vaddr=50000000 size=00010h (    16) load
I (386) boot: Loaded app from partition at offset 0x10000
I (386) boot: Disabling RNG early entropy source...
I (397) cpu_start: Pro cpu up.
I (398) cpu_start: Starting app cpu, entry point is 0x400811b8
0x400811b8: call_start_cpu1 at /root/esp/esp-idf/components/esp_system/port/cpu_start.c:155

I (0) cpu_start: App cpu up.
I (412) cpu_start: Pro cpu start user code
I (412) cpu_start: cpu freq: 160000000
I (412) cpu_start: Application information:
I (416) cpu_start: Project name:     passive
I (421) cpu_start: App version:      f8f95a5
I (426) cpu_start: Compile time:     Jul 20 2021 01:06:09
I (432) cpu_start: ELF file SHA256:  0884d64dc63e16d6...
I (438) cpu_start: ESP-IDF:          v4.4-dev-2128-gf45b60b28
I (445) heap_init: Initializing. RAM available for dynamic allocation:
I (452) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (458) heap_init: At 3FFB7690 len 00028970 (162 KiB): DRAM
I (464) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (471) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (477) heap_init: At 40095618 len 0000A9E8 (42 KiB): IRAM
I (484) spi_flash: detected chip: generic
I (488) spi_flash: flash io: dio
W (492) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (506) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (547) vfs_fat_sdmmc: slot init failed (0x103).
E (547) sd.h: Failed to initialize the card (ESP_ERR_INVALID_STATE).   If you do not have an SD card attached, please ignore this message.  Make sure SD card lines have pull-up resistors in place.
W (561) wifi_init: Failed to set default Wi-Fi event handlers (0x103)
I (568) wifi:wifi driver task: 3ffbeed4, prio:23, stack:6656, core=0
I (573) system_api: Base MAC address is not set
I (578) system_api: read default base MAC address from EFUSE
I (589) wifi:wifi firmware version: 4c9ede1
I (589) wifi:wifi certification version: v7.0
I (592) wifi:config NVS flash: enabled
I (596) wifi:config nano formating: disabled
I (600) wifi:Init data frame dynamic rx buffer num: 32
I (605) wifi:Init management frame dynamic rx buffer num: 32
I (610) wifi:Init management short buffer num: 32
I (614) wifi:Init dynamic tx buffer num: 32
I (619) wifi:Init static rx buffer size: 2212
I (623) wifi:Init static rx buffer num: 10
I (626) wifi:Init dynamic rx buffer num: 32
I (631) wifi_init: rx ba win: 6
I (634) wifi_init: tcpip mbox: 32
I (638) wifi_init: udp mbox: 6
I (642) wifi_init: tcp mbox: 6
I (645) wifi_init: tcp tx win: 5744
I (650) wifi_init: tcp rx win: 5744
I (654) wifi_init: tcp mss: 1440
I (658) wifi_init: WiFi IRAM OP enabled
I (662) wifi_init: WiFi RX IRAM OP enabled
I (668) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (776) wifi:mode : null
I (777) wifi:ic_enable_sniffer
type,role,mac,rssi,rate,sig_mode,mcs,bandwidth,smoothing,not_sounding,aggregation,stbc,fec_coding,sgi,noise_floor,ampdu_cnt,channel,secondary_channel,local_timestamp,ant,sig_len,rx_state,real_time_set,real_timestamp,len,CSI_DATA

--- after this no output is comming.

StevenMHernandez, 2021-07-21:
How long did you keep the ESP32 running? Are you sure that CSI is being generated by your host and access point? Overall, your setup of the tool sounds correct to me, there must be an issue with the host to access point part.

BidyutSaha, 2021-07-21:
Thank you for your response, Sir.
Q. **How long did you keep the ESP32 running?**
Ans. : 4 to 5 minutes. Not working. 
          But after few trials, it's showing the CSI data ( In passive mode ). Sometimes It stopped and does not show any received 
          CSI  data.
Q.  **Are you sure that CSI is being generated by your host and access point?** 
         How do I confirm it ? will It works on AP Beacon and STA probe frame?

StevenMHernandez, 2021-07-23:
From my understanding beacon frames will not work, it needs to be actual DATA frames. However, I do not currently have a list of what frames will or will not generate CSI in passive mode.

Here is my request for you: do you think you can you help figure out what sort of frames will generate CSI in passive mode? That would be a huge help towards this project!

BidyutSaha, 2021-07-27:
Hi StevenMHernandez,

Yes , beacon frames will not work. Even Control and Management Frames are not generating CSI Data. I tested it with different ways. After receiving the Data Frames  [ void  _wifi_csi_cb(void *ctx, wifi_csi_info_t *data) ] callback method is invoked. So we only need to data frames.
_wifi_csi_cb(void *ctx, wifi_csi_info_t *data)  method doesn't provide any frame header details. So we may not verify the frame type of extracted CSI. What do you thing ?

StevenMHernandez, 2021-07-28:
What ways did you test beacon, control and management frames? Did you use some specific software?

BidyutSaha, 2021-07-29:
I deployed a RPI  with packet sniffer mode and using tcp dump I can see Beacon and other frames (except data frame) but at the same time no CSI data is received in the esp 32 model (esp32 is configured in same chanel No. )

BidyutSaha, 2021-07-29:
I am also frequently seeing this error :

E (10578) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10578) task_wdt:  - IDLE (CPU 0)
E (10578) task_wdt: Tasks currently running:
E (10578) task_wdt: CPU 0: wifi
E (10578) task_wdt: CPU 1: IDLE
E (10578) task_wdt: Print CPU 0 (current core) backtrace

**** Any advice to  fix this Sir?

StevenMHernandez, 2021-07-29:
If I was to recreate your setup, all I need to run is `tcpdump` on the command line? Are there any other options/flags that you used? You are not actively sending specific packet types with the raspberry pi are you?

I forgot to mention: (https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/master/passive/main/main.c#L27) only allows DATA frames to work for CSI. Please add more flags and report back to let me know if this helps you.

For the watchdog error:
1. what is your baud rate
2. what is the git-hash of esp32-csi-tool
3. what is your esp-idf version
4. is this for the `passive` project?

BidyutSaha, 2021-07-29:
Setup :

Device #1 :  Wifi Router ( without internet connectivity , channel = 1) 
Device #2 :  Raspberry Pi W 0 ( in wifi monitor mode)
Device #3 :  Esp32 (in CSI Pasive mode ,channel = 1 , )

            const wifi_promiscuous_filter_t filt = {
            .filter_mask = WIFI_PROMIS_FILTER_MASK_ALL
            };

            in main.c to capture all types of frame ( control, management and Data )

Device #4 : Laptop ( UDP Server)
Device #5 : Laptop ( UDP client)

In this scenerio , 

            I run "tcpdump -i mon0" in rpi terminal where mon0 is the monitor mode interface and saw 
            diffrent becoan frames and other frames . But at that time No csi data is capturing by the esp ( serial port )

            But if I connect 2 pc to that network and create a udp client server and exchange data between the client and server 
            I can see the csi data captured by ESP ( serial port ), though the filter mask = WIFI_PROMIS_FILTER_MASK_ALL ( which is supposed to intercept all 
            types of frames) in the esp32 board. But if I stop the UDP data communication between the client and server then CSI captured also paused.

            So I thought only Data frames can gives CSI data to us what ever the filtermask is used in "wifi_promiscuous_filter_t".

            void _wifi_csi_cb(void *ctx, wifi_csi_info_t *data)  : this is the only call back method which is called after the CSI is captured/estimated by the ESP but unfortunatly
            no parameter is here to confirm from which types of packet this CSI is estimated.

Q. You are not actively sending specific packet types with the raspberry pi are you? 
NO i am not sending any packet from RPI.

Q.  all I need to run is tcpdump on the command line? Are there any other options/flags that you used? 
Yes in cmd line I run . I had not used any other flag.

whenn I apply filter_mask = WIFI_PROMIS_FILTER_MASK_ALL then It should capture all types of frame in promisous mode right?
according to the API it is says that CSI is estimated whenever a packet is recived by the esp from the transmitter. I am little bit confused 
with it ( only data packet can estimate CSI)

***** Actually I am giving more weight in becoan frame because if esp estimates CSI data  from  the becoan frame send by  the acess point, then there is no headace for  deploying udp server and client for data traffic. We can use only on single esp32 bord and there should not be any dependency for data traffic generation , by sending becoan frame by any access point , esp32 recives that farme and estimates its CSI.

BidyutSaha, 2021-07-29:
For the watchdog error:

    what is your baud rate : 115200
    what is the git-hash of esp32-csi-tool : f8f95a5edca2309fd9ac1ff57828acdc8893f05d
    what is your esp-idf version : ESP-IDF v4.4-dev-2128-gf45b60b28
    is this for the passive project? : Yes

BidyutSaha, 2021-07-31:
Hi, 
Is there any documentation to find the indices of null and pilot sub carriers so that we can filter them out and only consider data sub carrier?
We need to use only data sub carriers for processing.

StevenMHernandez, 2021-08-02:
See (https://github.com/StevenMHernandez/ESP32-CSI-Tool/issues/26#issuecomment-890544170). ESP-IDF v4.0 is tested but >v4.0 is unknown. You can test each version (v4.0, v4.1, v4.2, v4.3, up to your current version v4.4) to figure out when this watchdog error started appearing and report back in #11. That would be helpful to improve this project. But currently, I am not working on updating the version.

I do not have specific documentation about null/pilot subcarriers currently. Null subcarriers should be obvious in the data, they are all zeros. Did you do some research on your own to figure out the pilot subcarrier indices? If so, please share what you found. 
If you think it would be useful to other users of the tool, please send a pull request with this information added to the Readme.md file. Thanks!

NiyamPoudel, 2021-08-03:
I am using only one esp board with active_sta project. However, I am not getting any data. 
```
I (4240) tcpip_adapter: sta ip: 192.168.1.59, mask: 255.255.255.0, gw: 192.168.1.1
I (4241) Active CSI collection (Station): Got ip: 192.168.1.59
initial wifi connection established.
sending frames.
```
It stops at this. I have changed WIFI_CHANNEL and my SSID in main.cc
```
#define EXAMPLE_ESP_WIFI_SSID      "niyamssid"
#define EXAMPLE_ESP_WIFI_PASS      "niyampassword"
#define EXAMPLE_ESP_MAXIMUM_RETRY  10

#ifdef CONFIG_WIFI_CHANNEL
#define WIFI_CHANNEL CONFIG_WIFI_CHANNEL
#else
#define WIFI_CHANNEL 1
#endif
```

NiyamPoudel, 2021-08-03:
Also when I am using one board with passive project, I also dont get any data,
```
W (692) wifi_init: Failed to set default Wi-Fi event handlers (0x103)
I (713) wifi: wifi driver task: 3ffbac3c, prio:23, stack:3584, core=0
I (713) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (718) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (745) wifi: wifi firmware version: 581f422
I (746) wifi: config NVS flash: enabled
I (746) wifi: config nano formating: disabled
I (746) wifi: Init dynamic tx buffer num: 32
I (750) wifi: Init data frame dynamic rx buffer num: 32
I (755) wifi: Init management frame dynamic rx buffer num: 32
I (761) wifi: Init management short buffer num: 32
I (765) wifi: Init static rx buffer size: 2212
I (769) wifi: Init static rx buffer num: 10
I (773) wifi: Init dynamic rx buffer num: 32
I (881) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (882) wifi: mode : null
I (883) wifi: ic_enable_sniffer
type,role,mac,rssi,rate,sig_mode,mcs,bandwidth,smoothing,not_sounding,aggregation,stbc,fec_coding,sgi,noise_floor,ampdu_cnt,channel,secondary_channel,local_timestamp,ant,sig_len,rx_state,real_time_set,real_timestamp,len,CSI_DATA

```
Do I have to set the BASE MAC ADDRESS somewhere?

StevenMHernandez, 2021-08-11:
@NiyamPoudel Likely there is nothing nearby that is transmitting CSI-enabled OFDM frames for the passive device to collect. I would recommend reading this thread and looking at what questions I asked in the past (i.e. https://github.com/StevenMHernandez/ESP32-CSI-Tool/issues/22#issuecomment-884219664). If you answer these questions, I am more likely able to help you out.

StevenMHernandez, 2021-08-11:
For now, the original problem for this github-issue is "why is there no csi when in passive-mode?".

**For future readers**, the answer likely is:
- There are no devices transmitting CSI-enabled OFDM frames nearby (Maybe devices are not using 2.4GHz or the devices are using a different wifi channel than what you set for the ESP32)
- Keep the device running for a long time to see if there are any CSI is eventually collected

I am going to close this issue since it seems the issue was solved and now it might be diverging into unrelated support-style questions. If there are any other questions, please check other issues or create your own issue.

closed 2021-08-11 by StevenMHernandez