StevenMHernandez/ESP32-CSI-Tool #23: What each value in obtained CSI indicates?

dev-sora, 2021-07-21:
Hi,

I'm new to your tool and not sure what each value in obtained CSI indicates.

I use my ESP32 board as the STA, connecting to the ELECOM AP, 20 MHz, 2.4 GHz with encryption.
The following is the obtained CSI example (MAC address is hidden for security purpose):

```
CSI_DATA,STA,XX:XX:XX:XX:XX:XX,-68,11,1,0,0,1,1,0,1,0,0,-93,0,1,0,44107670,0,138,0,0,44.674807,384,[-118 32 8 0 4 -25 3 -22 3 -19 2 -12 2 -8 2 -3 3 2 3 6 3 10 4 13 5 15 6 18 7 19 8 19 7 19 7 19 6 17 5 17 4 15 2 13 1 12 -3 10 -4 8 -6 5 -7 3 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 -1 1 -5 1 -10 3 -13 3 -16 4 -19 5 -20 7 -22 7 -22 8 -22 9 -20 9 -18 9 -15 9 -12 7 -9 5 -6 3 -3 -1 0 -5 4 -8 5 -13 7 -16 8 -21 10 -25 10 -27 9 -29 9 -30 ]
```

I have two question about the obtained result.
1. What the values up to the front of the left square bracket mean? (In the example above, -68,11,...,44.674807,384)
2. What the values between the square brackets mean?  I suppose that the CSI indicates the channel frequency response and consists of complex values, and should be numbered based on the number of antennas.

I'm really looking forward to your reply.

StevenMHernandez, 2021-07-23:
1.) The values before the square brackets are described here: (https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/master/_components/csi_component.h#L85)
2.) The image below shows what each index of the subcarrier should contain. There are 64 subcarriers. The even values are imaginary and the odd indices are the real components.

[image]

I have not used multiple antennas with this project. But since you seem to have some knowledge about it, can you help me understand something? If there are two transmit antennas and one receive antenna, how can the receiver distinguish the two transmitting antenna signals? Would they be transmitted with different carrier frequencies or would each antenna send with a delay? If you can help me understand this, that would be very appreciated!

dev-sora, 2021-07-23:
Hi,
Thank you for your reply.  I've understood what the values indicates.

As for multiple antennas, in MIMO system, the size of matrix H (or CSI matrix) has N rows and M columns with N receiving antennas and M transmitting antennas.  Each element h_{ij} indicates the channel frequency response between i-th receiving antenna and j-th transmitting antenna.  CSI matrix should be obtained from each subcarrier, that is, if the communication channel includes 64 subcarriers, you can get 64 CSI matrices.  This matrices are shared between a receiver and a transmitter.

NiyamPoudel, 2021-07-23:
Hello Sir @dev-sora and @StevenMHernandez  . I am a quite new to this field. I am trying to use these CSI data for human fall detection. I am using a router with two antennas. I would be grateful if you could help me with these two queries. 
1) As far as I understand, the data enclosed in the bracket is a matrix.  It a single row matrix (because we have 1 receiving antenna) with 128 columns (because we have 64 sub carriers and 2 transmitting antennas). Am I correct? 

2) What would be the best way to analyze/visualize any change in the CSI_DATA because of some external factors? Thanks

StevenMHernandez, 2021-07-23:
@dev-sora Yes, I understand that the CSI is represented as a NxM matrix N receiving and M transmitting antennas. This matrix size is well known.

My main question is: how is CSI calculated at the physical layer? Most importantly **"If there are two transmit antennas and one receive antenna, how can the receiver separate and distinguish the two transmitting antenna signals?"** Can you share your own knowledge/research about this? Thanks!

@NiyamPoudel My comment above answers your first question. Analyzing CSI data is up to you. This tool currently is focused on _collecting_ CSI data. As a user, you need to figure out what the problem is that you are trying to solve and then narrow down to the best solution for your own use case.

NiyamPoudel, 2021-07-23:
@StevenMHernandez , Thank you for the answer. I am also eagerly waiting for @dev-sora to kindly answer what the CSI matrix mean in the case of two transmitting antennas.

StevenMHernandez, 2021-07-24:
@NiyamPoudel Can you look into how multiple antennas works at the _signals level_ as well and report back, that would be helpful. Thanks!

dev-sora, 2021-07-24:
Hi,
I don't fully understand how the CSI is calculated at the physical layer, but the least I can say is that even if there are multiple transmitting antennas, the receiver does not separate the signals transmitted from them.
Since MIMO communication is represented by a model of linear mixing, the receiver can only estimate the frequency response, and precoding of the transmitted signal is required if you want to separate the individual signals.

StevenMHernandez, 2021-07-25:
@dev-sora Thanks for this information! Precoding sounds interesting. 
Do you have a good source for where you found this information? Thanks again!

StevenMHernandez, 2021-07-29:
I will close this because I think the original question has been answered. Please reopen if needed.

closed 2021-07-29 by StevenMHernandez