StevenMHernandez/ESP32-CSI-Tool #24: Visualizing CSI amplitude Data in Real Time

arcgolatkar, 2021-07-21:
I want to visualize the CSI Amplitude Data in real-time. Is there any way to do that? 
So far, I am able to plot the Amplitude data from a CSV file. 
Thanks! Looking forward to your reply.

StevenMHernandez, 2021-07-23:
Great! I think there are some other people that would like to do this as well!

Currently I have not built any functionality for this.

However, if you can help me out by possibly doing the following:
1.) Can you share some suggestions on how you would recommend building this feature?
2.) Then, if you are up to the challenge, you could also try implementing this feature and trying out contributing to this open source project. That would be great!

Thanks!

arcgolatkar, 2021-07-24:
I  converted the raw data into amplitude data using _utils/parse.py. Basically, I wanted to convert the data and visualize it while the new data is getting stored in the csv. So, I made some changes to your parse.py and was able to visualize the data in real-time. I have mailed you the copy. The only thing is that we have to run this file on a different terminal.

NiyamPoudel, 2021-08-04:
Here is what I did to visualize the data in real time
```
import sys
import time
import datetime as dt
import matplotlib.pyplot as plt
import matplotlib.animation as animation
import math
import numpy as np

# Create figure for plotting
plt.close('all')
plt.figure()
plt.ion()
plt.show()

def process(res):
    # print("res",res)
    all_data = res.split(',')
    # print(all_data[25])
    csi_data = all_data[25].split(" ")
    csi_data[0] = csi_data[0].replace("[", "")
    csi_data[-1] = csi_data[-1].replace("]", "") 
    csi_data.pop()
    csi_data = [int(c) for c in csi_data if c]
    # print(csi_data)    
    imaginary = []
    real = []
    for i, val in enumerate(csi_data):
        if i % 2 == 0:
            imaginary.append(val)
        else:
            real.append(val)
    
    print(imaginary)
    print(real)
    # print("real", len(real))
    csi_size = 128
    x = []
    y = []
    amplitudes = []
    phases = []
    csi = []
    if len(imaginary) > 0 and len(real) > 0:
        for j in range(int(csi_size / 2)):  
            amplitudeCalc = math.sqrt(imaginary[j] ** 2 + real[j] ** 2)
            phaseCalc = math.atan2(imaginary[j], real[j])  
            amplitudes.append(amplitudeCalc)
            phases.append(phaseCalc)
            x.append(j)

        y = csi
        print("x:", x)
        print("y:", y)
        plt.cla()
        plt.plot(x,y)
        plt.pause(0.0001)

while True:
    line = sys.stdin.readline()
    if "CSI_DATA" in line:
        process(line)

```

kallu1294, 2021-12-28:
> Here is what I did to visualize the data in real time
>
> ```
> import sys
> import time
> import datetime as dt
> import matplotlib.pyplot as plt
> import matplotlib.animation as animation
> import math
> import [...]

I am not able plot real time CSI data using code which you mentioned above, If I run that its just showing blank figure.
Can you please explain how and when to run your code.

kallu1294, 2022-01-04:
> I converted the raw data into amplitude data using _utils/parse.py. Basically, I wanted to convert the data and visualize it while the new data is getting stored in the csv. So, I made some changes [...]

I tried to visualize CSI data, but I am not able to plot real time data. Can you please send a copy of the code which you have done.

NJUPT-Sivan, 2022-04-03:
> 以下是我为实时可视化数据所做的工作
> 
> ```
> import sys
> import time
> import datetime as dt
> import matplotlib.pyplot as plt
> import matplotlib.animation as animation
> import math
> import numpy as np
> 
> # Create figure for plotting
> plt.close('all')
> plt.figure()
> plt.ion()
> plt.show()
> 
> def process(res):
>     # print("res",res)
>     all_data = res.split(',')
>     # print(all_data[25])
>     csi_data = all_data[25].split(" ")
>     csi_data[0] = csi_data[0].replace("[", "")
>     csi_data[-1] = csi_data[-1].replace("]", "") 
>     csi_data.pop()
>     csi_data = [int(c) for c in csi_data if c]
>     # print(csi_data)    
>     imaginary = []
>     real = []
>     for i, val in enumerate(csi_data):
>         if i % 2 == 0:
>             imaginary.append(val)
>         else:
>             real.append(val)
>     
>     print(imaginary)
>     print(real)
>     # print("real", len(real))
>     csi_size = 128
>     x = []
>     y = []
>     amplitudes = []
>     phases = []
>     csi = []
>     if len(imaginary) > 0 and len(real) > 0:
>         for j in range(int(csi_size / 2)):  
>             amplitudeCalc = math.sqrt(imaginary[j] ** 2 + real[j] ** 2)
>             phaseCalc = math.atan2(imaginary[j], real[j])  
>             amplitudes.append(amplitudeCalc)
>             phases.append(phaseCalc)
>             x.append(j)
> 
>         y = csi
>         print("x:", x)
>         print("y:", y)
>         plt.cla()
>         plt.plot(x,y)
>         plt.pause(0.0001)
> 
> while True:
>     line = sys.stdin.readline()
>     if "CSI_DATA" in line:
>         process(line)
> ```

I tried to use your code in pycharm to visualize the data in real time, but it shows a blank. What is the reason for this?

Tisha36, 2022-07-14:
> I converted the raw data into amplitude data using _utils/parse.py. Basically, I wanted to convert the data and visualize it while the new data is getting stored in the csv. So, I made some changes [...]

can you send me the copy

AsciiPro99, 2022-10-04:
> Here is what I did to visualize the data in real time
>
> ```
> import sys
> import time
> import datetime as dt
> import matplotlib.pyplot as plt
> import matplotlib.animation as animation
> import math
> import [...]

Solid code, i fixed the plotting to get live plot. Not sure if made in old version or other hardware, but needed some tweaks to get it working. Is live plotting still something you are interested in @StevenMHernandez ? The code is not amazing, it works tho.

StevenMHernandez, 2022-10-05:
Absolutely! Please create a pull request and I will be glad to give comments and suggestions to make the code better/more understandable. The new python file can be store in (https://github.com/StevenMHernandez/ESP32-CSI-Tool/tree/master/python_utils).

Thanks!

AsciiPro99, 2022-10-10:
Alright uploaded it, sorry for the delay busy week. Feel free to correct any code.

xiaokangkang6, 2022-10-18:
> 我使用_utils/parse.py将原始数据转换为幅度数据。基本上，我希望在新数据存储在csv中的同时转换数据并将其可视化。因此，我对您的parse.py做了一些更改，能够实时可视化数据。我已经把副本寄给你了。唯一的问题是我们必须在不同的终端上运行这个文件。

Excuse me, could you please give me your code for reference? Thank you very much!

StevenMHernandez, 2022-11-02:
Closed by #67. If anyone has other ideas for how to visualized the CSI data, we can create another pull request or another github issue.

Thanks again for your help @AsciiPro99!

closed 2022-11-02 by StevenMHernandez