StevenMHernandez/ESP32-CSI-Tool #26: Understanding the lines between the CSI data feed

arcgolatkar, 2021-07-28:
I get these lines sometimes between the CSI feed. Can anyone tell me what they mean? I am also observing some pauses between the data packets? Are these lines anyhow related to that?

CSI_DATA,AP,F0:9E:4A:81:04:BB,-56,11,0,0,0,0,0,0,0,0,0,-95,0,8,1,2688908,0,30,0,0,3.482471,128,[30 -32 1 0 0 0 0 0 0 0 0 0 6 -15 7 -15 7 -15 8 -16 7 -14 8 -15 8 -15 8 -14 9 -16 9 -16 11 -18 12 -19 12 -19 12 -20 12 -20 12 -19 13 -20 12 -19 14 -20 13 -19 13 -19 13 -19 12 -19 13 -19 13 -19 13 -20 0 0 13 -20 13 -19 13 -20 13 -19 13 -20 13 -20 13 -21 13 -20 13 -21 12 -20 13 -21 13 -21 13 -21 13 -22 12 -19 12 -20 11 -18 11 -18 10 -16 9 -16 9 -16 9 -16 9 -16 9 -16 9 -17 9 -16 0 0 0 0 0 0 0 0 0 0 ]
W (3928) wifi:<ba-del>idx
I (3931) wifi:station: f0:9e:4a:81:04:bb leave, AID = 1, bss_flags is 658547, bss:0x3ffb9ff0
I (3937) wifi:new:<8,0>, old:<8,1>, ap:<8,1>, sta:<255,255>, prof:8
W (3944) wifi:<ba-del>idx
I (3947) Active CSI collection (AP): station:f0:9e:4a:81:04:bbleave, AID=1
E (3984) wifi:addba response cb: ap bss deleted
E (4001) wifi:addba response cb: ap bss deleted
I (7195) wifi:new:<8,1>, old:<8,0>, ap:<8,1>, sta:<255,255>, prof:8
I (7197) wifi:station: f0:9e:4a:81:04:bb join, AID=1, bgn, 40U
I (7224) Active CSI collection (AP): station:f0:9e:4a:81:04:bb join, AID=1
CSI_DATA,AP,F0:9E:4A:81:04:BB,-51,11,1,3,1,1,1,0,1,0,0,-95,0,8,1,6377972,0,414,0,0,6.831762,612,[-98 97 25 0 0 0 0 0 0 0 0 0 -13 -10 -12 -9 -11 -9 -10 -9 -9 -7 -8 -6 -7 -5 -7 -3 -6 -2 -6 0 -7 1 -7 3 -7 4 -8 5 -9 7 -10 9 -13 10 -14 11 -16 13 -17 13 -17 12 -18 12 -18 12 -18 11 -18 10 -17 9 0 0 -15 7 -14 7 -12 6 -10 6 -8 5 -6 5 -4 5 -1 6 1 6 3 7 5 8 7 9 8 10 9 12 10 13 11 15 11 15 12 16 11 17 11 17 10 18 9 18 9 18 8 17 7 16 6 15 0 0 0 0 0 0 0 0 0 0 ]

And sometimes these.

CSI_DATA,AP,F0:9E:4A:81:04:BB,-52,11,1,7,1,1,1,0,1,0,0,-95,0,8,1,7632692,0,110,0,0,8.318688,612,[110 96 6 0 0 0 0 0 0 0 0 0 1 -12 0 -14 0 -15 1 -16 1 -17 1 -18 2 -19 2 -19 3 -19 4 -19 5 -18 5 -18 5 -17 6 -16 5 -15 6 -15 5 -15 5 -15 4 -14 3 -12 0 -11 -1 -10 -4 -9 -6 -8 -7 -8 -10 -9 0 0 -13 -9 -15 -10 -16 -11 -18 -12 -18 -13 -18 -14 -18 -15 -18 -16 -18 -16 -17 -17 -16 -17 -15 -16 -13 -16 -12 -15 -11 -13 -10 -12 -9 -10 -8 -9 -8 -6 -8 -4 -7 -1 -7 1 -8 3 -9 6 -10 7 -12 9 0 0 0 0 0 0 0 0 0 0 ]
W (8765) wifi:<ba-del>idx
W (8766) wifi:<ba-del>idx
W (8768) wifi:<ba-del>idx
W (8770) wifi:<ba-del>idx
W (8773) wifi:<ba-del>idx

StevenMHernandez, 2021-07-29:
Does this happen immediately after monitoring begins?
Does this happen if no STAs are connected?

I see (https://github.com/espressif/esp-idf/issues/6107) mentions `wifi:<ba-del>idx`, have you found any other helpful links on the internet that have that same error message?

What is the STA that is connecting? Is it another ESP32 or a smartphone or something else?

arcgolatkar, 2021-08-01:
After 4-5 seconds of monitoring. I was using a smartphone as STA.

I was working with IDF 4.3 and may be due to that, I was getting this error. 
After seeing the updated readme, I shifted to IDF 4.0 and I didn't see these lines again.
The error is resolved. Also, I am getting a smooth stream of CSI data without any pauses after the update.
Thanks a lot and sorry for the trouble!

closed 2021-08-01 by arcgolatkar

StevenMHernandez, 2021-08-02:
Thank you for letting me know! So this is a potential issue when we update to ESP-IDF 4.3
I need to keep this in mind. I am glad it is working well now!

May I ask, are you running any specific software on the smartphones to generate specific traffic rates or are you just looking at the natural wifi rate?

arcgolatkar, 2021-08-16:
The data stream is smooth for around 8-10 seconds. After that, it starts to jitter again! I have tried both laptop and smartphone as STA. I am not running any specific software to generate traffic.
I used your python file serial_append_time.py to check the interval between 2 packets. (mostly the pauses are for around 1 or 2 sec, but sometimes it takes 20 seconds to get new data packet)
Also after the update, the 1552000 baud rate is working for me.

StevenMHernandez, 2021-08-16:
If you are not controlling the traffic on the laptop or smartphone, should you be expecting a consistent traffic rate?

There are 3 areas that could causes issues here (I will assume STA is laptop)
1. **STA is not sending consistent data rate.** To evaluate this, can you run a program like wireshark on the laptop to see how much traffic is being send over wifi and what types or packets are being sent?
2. **ESP32 is causing some lag.** I doubt this is the issue unless you did not install the system correctly based on the README. To evaluate this you need two esp32s running active_ap and active_sta.
3. **Your computer cannot handle the incoming serial data**. Maybe you have a very old/slow computer that can not handle the incoming serial data from the esp32 on the USB port.

I think it is the first one. 

The **easiest method** for running this project is to use two ESP32 running the `active_ap` and `active_sta` applications. There are many other ways to use the project (i.e. with laptop as STA), but if you are not sure how to break down the problem to debug the many unique problems on your own, I would *really* recommend going with the easiest method.