StevenMHernandez/ESP32-CSI-Tool #27: Amplitude/phase oscillates rapidly between upper and lower value

FadedCoder, 2021-07-29:
**Describe the bug**
The amplitude and phase keeps on rapidly oscillating (often alternatively) between two values, one higher and one lower. This is visible in example CSV inside the _utils folder too. What exactly causes this? What's the best way to mitigate? 

**Tool Settings**
- Baud Rate: 1552000
- WiFi Channel per Device: The default (4?)
- FreeRTOS > Tick rate (Hz): 1000

**To Reproduce**
Steps to reproduce the behavior:
1. Use active_ap along with an external device (laptop/phone etc) and record data.
2. Observe amplitude/phase values oscillating between an upper value and a lower value.

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - ESP-IDF Version: 3.3.1
 - ESP32-CSI-Tool Git Hash: f8f95a5edca2309fd9ac1ff57828acdc8893f05d

StevenMHernandez, 2021-08-02:
Good question. Typically this CSI will go through a smoothing/denoising process which will eliminate high frequency changes like this, but I do not have any comment about why this should or should not be happening.

I think this is a very good question, and if you have any ideas of what might be causing this I would like to hear your thoughts. Can you help me and help this open source project by spending some time thinking/researching why something like this might be happening? That would be very helpful to the project. Thanks!

Paco-Danes, 2023-09-23:
Hello, same issue here. The extracted amplitudes are very very variable over subcarriers but also over time (considering a single subcarrier). 
[image]

Thise is a plot for a single subcarrier over ONLY 1 second (100 packets, since the rate is 100pk/s). Is it normal that it changes so rapidly every 10ms ?