StevenMHernandez/ESP32-CSI-Tool #28: Build stopped subcommand failed?

TheNightMaster, 2021-08-08:
I'm not very familiar with the process being followed here, however I followed all the steps and am unable to get the project to build. I've attached the console log below, can someone please have a look? 

```
F:\Favourites\Projects\SOP\esp32-csi-tool\active_sta>idf.py -p COM3 flash monitor
Executing action: flash
Running ninja in directory f:\favourites\projects\sop\esp32-csi-tool\active_sta\build
Executing "ninja flash"...
[1/10] Performing build step for 'bootloader'
ninja: no work to do.
[2/8] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/main.cc.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.cc.obj
ccache C:\Users\aksha\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -IC:/Users/aksha/Desktop/esp-idf-2/components/newlib/platform_include -IC:/Users/aksha/Desktop/esp-idf-2/components/freertos/include -IC:/Users/aksha/Desktop/esp-idf-2/components/freertos/port/xtensa/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_hw_support/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_hw_support/port/esp32/. -IC:/Users/aksha/Desktop/esp-idf-2/components/heap/include -IC:/Users/aksha/Desktop/esp-idf-2/components/log/include -IC:/Users/aksha/Desktop/esp-idf-2/components/lwip/include/apps -IC:/Users/aksha/Desktop/esp-idf-2/components/lwip/include/apps/sntp -IC:/Users/aksha/Desktop/esp-idf-2/components/lwip/lwip/src/include -IC:/Users/aksha/Desktop/esp-idf-2/components/lwip/port/esp32/include -IC:/Users/aksha/Desktop/esp-idf-2/components/lwip/port/esp32/include/arch -IC:/Users/aksha/Desktop/esp-idf-2/components/soc/include -IC:/Users/aksha/Desktop/esp-idf-2/components/soc/esp32/. -IC:/Users/aksha/Desktop/esp-idf-2/components/soc/esp32/include -IC:/Users/aksha/Desktop/esp-idf-2/components/hal/esp32/include -IC:/Users/aksha/Desktop/esp-idf-2/components/hal/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_rom/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_rom/esp32 -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_common/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_system/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp32/include -IC:/Users/aksha/Desktop/esp-idf-2/components/driver/include -IC:/Users/aksha/Desktop/esp-idf-2/components/driver/esp32/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_ringbuf/include -IC:/Users/aksha/Desktop/esp-idf-2/components/efuse/include -IC:/Users/aksha/Desktop/esp-idf-2/components/efuse/esp32/include -IC:/Users/aksha/Desktop/esp-idf-2/components/xtensa/include -IC:/Users/aksha/Desktop/esp-idf-2/components/xtensa/esp32/include -IC:/Users/aksha/Desktop/esp-idf-2/components/espcoredump/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_timer/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_ipc/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_pm/include -IC:/Users/aksha/Desktop/esp-idf-2/components/vfs/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_wifi/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_wifi/esp32/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_event/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_netif/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_eth/include -IC:/Users/aksha/Desktop/esp-idf-2/components/tcpip_adapter/include -IC:/Users/aksha/Desktop/esp-idf-2/components/app_trace/include -IC:/Users/aksha/Desktop/esp-idf-2/components/mbedtls/port/include -IC:/Users/aksha/Desktop/esp-idf-2/components/mbedtls/mbedtls/include -IC:/Users/aksha/Desktop/esp-idf-2/components/mbedtls/esp_crt_bundle/include -IC:/Users/aksha/Desktop/esp-idf-2/components/app_update/include -IC:/Users/aksha/Desktop/esp-idf-2/components/spi_flash/include -IC:/Users/aksha/Desktop/esp-idf-2/components/bootloader_support/include -IC:/Users/aksha/Desktop/esp-idf-2/components/nvs_flash/include -IC:/Users/aksha/Desktop/esp-idf-2/components/pthread/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_gdbstub/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_gdbstub/xtensa -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_gdbstub/esp32 -IC:/Users/aksha/Desktop/esp-idf-2/components/wpa_supplicant/include -IC:/Users/aksha/Desktop/esp-idf-2/components/wpa_supplicant/port/include -IC:/Users/aksha/Desktop/esp-idf-2/components/wpa_supplicant/include/esp_supplicant -IC:/Users/aksha/Desktop/esp-idf-2/components/perfmon/include -IC:/Users/aksha/Desktop/esp-idf-2/components/asio/asio/asio/include -IC:/Users/aksha/Desktop/esp-idf-2/components/asio/port/include -IC:/Users/aksha/Desktop/esp-idf-2/components/cbor/port/include -IC:/Users/aksha/Desktop/esp-idf-2/components/unity/include -IC:/Users/aksha/Desktop/esp-idf-2/components/unity/unity/src -IC:/Users/aksha/Desktop/esp-idf-2/components/cmock/CMock/src -IC:/Users/aksha/Desktop/esp-idf-2/components/coap/port/include -IC:/Users/aksha/Desktop/esp-idf-2/components/coap/port/include/coap -IC:/Users/aksha/Desktop/esp-idf-2/components/coap/libcoap/include -IC:/Users/aksha/Desktop/esp-idf-2/components/coap/libcoap/include/coap2 -IC:/Users/aksha/Desktop/esp-idf-2/components/console -IC:/Users/aksha/Desktop/esp-idf-2/components/nghttp/port/include -IC:/Users/aksha/Desktop/esp-idf-2/components/nghttp/nghttp2/lib/includes -IC:/Users/aksha/Desktop/esp-idf-2/components/esp-tls -IC:/Users/aksha/Desktop/esp-idf-2/components/esp-tls/esp-tls-crypto -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_adc_cal/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_hid/include -IC:/Users/aksha/Desktop/esp-idf-2/components/tcp_transport/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_http_client/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_http_server/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_https_ota/include -IC:/Users/aksha/Desktop/esp-idf-2/components/protobuf-c/protobuf-c -IC:/Users/aksha/Desktop/esp-idf-2/components/protocomm/include/common -IC:/Users/aksha/Desktop/esp-idf-2/components/protocomm/include/security -IC:/Users/aksha/Desktop/esp-idf-2/components/protocomm/include/transports -IC:/Users/aksha/Desktop/esp-idf-2/components/mdns/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_local_ctrl/include -IC:/Users/aksha/Desktop/esp-idf-2/components/sdmmc/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_serial_slave_link/include -IC:/Users/aksha/Desktop/esp-idf-2/components/esp_websocket_client/include -IC:/Users/aksha/Desktop/esp-idf-2/components/expat/expat/expat/lib -IC:/Users/aksha/Desktop/esp-idf-2/components/expat/port/include -IC:/Users/aksha/Desktop/esp-idf-2/components/wear_levelling/include -IC:/Users/aksha/Desktop/esp-idf-2/components/fatfs/diskio -IC:/Users/aksha/Desktop/esp-idf-2/components/fatfs/vfs -IC:/Users/aksha/Desktop/esp-idf-2/components/fatfs/src -IC:/Users/aksha/Desktop/esp-idf-2/components/freemodbus/common/include -IC:/Users/aksha/Desktop/esp-idf-2/components/idf_test/include -IC:/Users/aksha/Desktop/esp-idf-2/components/idf_test/include/esp32 -IC:/Users/aksha/Desktop/esp-idf-2/components/jsmn/include -IC:/Users/aksha/Desktop/esp-idf-2/components/json/cJSON -IC:/Users/aksha/Desktop/esp-idf-2/components/libsodium/libsodium/src/libsodium/include -IC:/Users/aksha/Desktop/esp-idf-2/components/libsodium/port_include -IC:/Users/aksha/Desktop/esp-idf-2/components/mqtt/esp-mqtt/include -IC:/Users/aksha/Desktop/esp-idf-2/components/openssl/include -IC:/Users/aksha/Desktop/esp-idf-2/components/spiffs/include -IC:/Users/aksha/Desktop/esp-idf-2/components/ulp/include -IC:/Users/aksha/Desktop/esp-idf-2/components/wifi_provisioning/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -std=gnu++11 -fno-exceptions -fno-rtti -D_GNU_SOURCE -DIDF_VER=\"v4.3-dirty\" -DESP_PLATFORM -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/main.cc.obj -MF esp-idf\main\CMakeFiles\__idf_main.dir\main.cc.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/main.cc.obj -c ../main/main.cc
In file included from ../main/main.cc:8:
[... cut ...]
         esp_err_t __err_rc = (x);                                       \
                               ^
In file included from ../main/main.cc:7:
C:/Users/aksha/Desktop/esp-idf-2/components/esp_wifi/include/esp_wifi.h:799:48: note:   initializing argument 1 of 'esp_err_t esp_wifi_set_config(wifi_interface_t, wifi_config_t*)'
 esp_err_t esp_wifi_set_config(wifi_interface_t interface, wifi_config_t *conf);
                               ~~~~~~~~~~~~~~~~~^~~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

```

StevenMHernandez, 2021-08-08:
You need to complete ALL information here (https://raw.githubusercontent.com/StevenMHernandez/ESP32-CSI-Tool/master/.github/ISSUE_TEMPLATE/bug_report.md)

StevenMHernandez, 2021-08-08:
With out seeing the above information, my first guess is you are not using esp-idf v4.0

StevenMHernandez, 2021-11-15:
The same bug was found in issue #37, I will close this issue since there is already some discussion on the other issue.

The issue is: esp-idf version was greater than v4.0.

closed 2021-11-15 by StevenMHernandez