StevenMHernandez/ESP32-CSI-Tool #29: Not able to collect CSI in active_sta mode

tusharjain1003, 2021-08-08:
I am not able to collect CSI data while working with the active_sta mode. I can establish a connection with my phone's hotspot, but the process gets stuck after that.
[image]
Another issue that I am facing is related to the active_ap mode. The CSI collection process is not smooth and gets stuck randomly. 
Could you please help me with these issues? Thanks!

StevenMHernandez, 2021-08-08:
You need to complete ALL information here (https://raw.githubusercontent.com/StevenMHernandez/ESP32-CSI-Tool/master/.github/ISSUE_TEMPLATE/bug_report.md)

StevenMHernandez, 2021-08-08:
Hint: you are probably using an old version of this project. Update to the master branch and follow all steps in the readme again.

ahmed1hisham, 2021-11-13:
I'm encountering the same issue, any instructions please?

- Baud Rate: `1500000`  (tried 1552000, 921600 as well)
- WiFi Channel per Device: `6` (tried 1 & 11 as well)
- FreeRTOS > Tick rate (Hz): `1000`

Steps to reproduce the behavior:
1. Build project(s) `active_sta`
2. Flash and monitor project(s) `active_sta`
3. See error

**Desktop (please complete the following information):**
 - OS: macOS
 - ESP-IDF Version: 4.0
 
 
 Thanks in advance 🙏

StevenMHernandez, 2021-11-15:
STA does not normally display CSI (typically active_ap would be setup to display CSI). That is why it _appears_ to be not working.

**Please check your tool settings to make sure that `SHOULD_COLLECT_CSI` is enabled in `idf.py menuconfig > ESP32 CSI Tool Config`.**

@ahmed1hisham, if this fixes your issue please let me know so that I can close this issue (The original issue is from Aug. so I am assuming that @tusharjain1003 no longer has this problem.)

ahmed1hisham, 2021-11-15:
@StevenMHernandez I already have this enabled, but I managed to fix the issue. I tried multiple wifi channels till it worked.

Thanks a lot!