StevenMHernandez/ESP32-CSI-Tool #30: Active_ap mode: CSI collection process not smooth

tusharjain1003, 2021-08-17:
In active_ap mode, the CSI collection process is not smooth and gets stuck randomly.

- Baud Rate: 1552000 
- WiFi Channel per Device: 6
- FreeRTOS > Tick rate (Hz): 1000

Steps to reproduce the behavior:
1. Build project(s) `...` active_ap 
2. Flash project(s) `...` active_ap
3. External device setup (if applicable)
3. See error

 - OS: Windows
 - ESP-IDF Version: 4.0

StevenMHernandez, 2021-08-23:
At what rate do yuo have the smartphone sending CSI frames? If you are not controlling the data output by the smartphone, then it is likely the smartphone is not sending any frames OR the frames being sent do not generate CSI.

kallu1294, 2021-12-03:
> At what rate do yuo have the smartphone sending CSI frames? If you are not controlling the data output by the smartphone, then it is likely the smartphone is not sending any frames OR the frames being sent do not generate CSI.

Sir can we make our smartphone as STA and  esp32 as AP.? Are we able to see CSI frames.??

StevenMHernandez, 2021-12-07:
@kallu1294 Yes this is possible. However, I cannot guarantee consistent CSI frame-rate. This is an **advanced** usage of this ESP-CSI-Tool and I cannot promise that I will be able to debug any issues from "non-supported hardware" like smartphones. If you have advanced understanding of what is going on, you should be able to identify methods to get this working. I just do not know how much engineering effort would be required.

kallu1294, 2021-12-07:
Sir for this then I can't use my smartphone or wifi (router) as AP .?

So I should use two esp32 one for AP and another one for STA right.??

StevenMHernandez, 2021-12-08:
Two esp32s is the **recommended** way with this tool. Yes.

It is possible to use a smartphone or an AP, but you will need advanced knowledge.

kallu1294, 2021-12-08:
Ok thank you sir for the reply

What if I make esp32 as STA and my wifi router as AP will I'm able to
capture CSI data..??

However CSI is coming but I'm not able to download those data

Is there any solution for this.??

StevenMHernandez, 2021-12-08:
Yes, that should work too. 

@kallu1294, I have already answered this question for you in the other thread 12 hours ago (#44) and in a different thread 9 days ago (#45). 

At this point I am afraid that you are not being respectful of my help and my time. It also seems like you are not trying to debug these issues yourself.

kallu1294, 2021-12-08:
I'm trying sir 😕 CSV file is getting created but It is empty

kallu1294, 2021-12-08:
In #44 serial port was an issue that was solved and at the end I mentioned
that too csv file is empty 😕

kallu1294, 2021-12-08:
> Yes, that should work too.
>
> @kallu1294, I have already answered this question for you in the other thread 12 hours ago (#44) and in a different thread 9 days ago (#45).
>
> At this point I am afraid [...]

literally sorry sir but I'm trying my best to figure it out, CSV file getting created but it is empty

StevenMHernandez, 2022-10-17:
Closing old issues.

closed 2022-10-17 by StevenMHernandez