StevenMHernandez/ESP32-CSI-Tool #31: active_sta mode: not able to collect CSI data

tusharjain1003, 2021-08-17:
I am not able to collect CSI data while working with the active_sta mode. I can establish a connection with my phone's hotspot, but the process gets stuck after that.

- Baud Rate: 1552000 
- WiFi Channel per Device: 6
- FreeRTOS > Tick rate (Hz): 1000

Steps to reproduce the behavior:
1. Build project(s) `...` active_sta
2. Flash project(s) `...` active_sta
3. External device setup (if applicable)
3. See error

 - OS: Windows
 - ESP-IDF Version: 4.0
[image]

StevenMHernandez, 2021-08-23:
Did you set the STA to collect CSI in the `idf.py menuconfig`? It is not on by default. If this was the issue, can you think of a way to update the README.md file to make this more clear? Thanks.

StevenMHernandez, 2021-11-15:
This is a duplicate of your previous issue: #29 
Closing as duplicate.

closed 2021-11-15 by StevenMHernandez