StevenMHernandez/ESP32-CSI-Tool #32: collect2.exe error

jhopino, 2021-08-22:
**Describe the bug**
I can not build. It shows collect2.exe error.

**Tool Settings**
- Baud Rate: 1522000
- WiFi Channel per Device: 6
- FreeRTOS > Tick rate (Hz): 1000

**To Reproduce**
Steps to reproduce the behavior:
1. Build project(s) `...` active_ap
2. Flash project(s) `...` active_ap
3. External device setup (if applicable)
3. See error

**Desktop (please complete the following information):**
 - OS: Windows
 - ESP-IDF Version: 4.0
 - ESP32-CSI-Tool Git Hash: ?

**Additional context**
Add any other context about the problem here.
c:\users\v330\desktop\esp-idf\examples\esp32-csi-tool\active_ap\build>C:\Users\V330\.espressif\python_env\idf4.0_py3.8_env\Scripts\python.exe C:/Users/V330/Desktop/esp-idf/tools/kconfig_new/confgen.py --kconfig C:/Users/V330/Desktop/esp-idf/Kconfig --sdkconfig-rename C:/Users/V330/Desktop/esp-idf/sdkconfig.rename --config C:/Users/V330/Desktop/esp-idf/examples/esp32-csi-tool/active_ap/sdkconfig --env-file C:/Users/V330/Desktop/esp-idf/examples/esp32-csi-tool/active_ap/build/config.env --env IDF_TARGET=esp32 --output config C:/Users/V330/Desktop/esp-idf/examples/esp32-csi-tool/active_ap/sdkconfig   || exit /b

C:\Users\V330\Desktop\esp-idf\examples\esp32-csi-tool\active_ap>idf.py build
Checking Python dependencies...
Python requirements from C:\Users\V330\Desktop\esp-idf\requirements.txt are satisfied.
Executing action: all (aliases: build)
Running ninja in directory c:\users\v330\desktop\esp-idf\examples\esp32-csi-tool\active_ap\build
Executing "ninja all"...
[0/1] Re-running CMake...
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- ccache will be used for faster recompilation
-- Project version: b2f0b28
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/Users/V330/Desktop/esp-idf/examples/esp32-csi-tool/active_ap/sdkconfig
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- Adding linker script C:/Users/V330/Desktop/esp-idf/examples/esp32-csi-tool/active_ap/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/V330/Desktop/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/V330/Desktop/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/Users/V330/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/V330/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/V330/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/V330/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/V330/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_ringbuf esp_rom esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/V330/Desktop/esp-idf/components/app_trace C:/Users/V330/Desktop/esp-idf/components/app_update C:/Users/V330/Desktop/esp-idf/components/asio C:/Users/V330/Desktop/esp-idf/components/bootloader C:/Users/V330/Desktop/esp-idf/components/bootloader_support C:/Users/V330/Desktop/esp-idf/components/bt C:/Users/V330/Desktop/esp-idf/components/coap C:/Users/V330/Desktop/esp-idf/components/console C:/Users/V330/Desktop/esp-idf/components/cxx C:/Users/V330/Desktop/esp-idf/components/driver C:/Users/V330/Desktop/esp-idf/components/efuse C:/Users/V330/Desktop/esp-idf/components/esp-tls C:/Users/V330/Desktop/esp-idf/components/esp32 C:/Users/V330/Desktop/esp-idf/components/esp_adc_cal C:/Users/V330/Desktop/esp-idf/components/esp_common C:/Users/V330/Desktop/esp-idf/components/esp_eth C:/Users/V330/Desktop/esp-idf/components/esp_event C:/Users/V330/Desktop/esp-idf/components/esp_gdbstub C:/Users/V330/Desktop/esp-idf/components/esp_http_client C:/Users/V330/Desktop/esp-idf/components/esp_http_server C:/Users/V330/Desktop/esp-idf/components/esp_https_ota C:/Users/V330/Desktop/esp-idf/components/esp_https_server C:/Users/V330/Desktop/esp-idf/components/esp_local_ctrl C:/Users/V330/Desktop/esp-idf/components/esp_ringbuf C:/Users/V330/Desktop/esp-idf/components/esp_rom C:/Users/V330/Desktop/esp-idf/components/esp_websocket_client C:/Users/V330/Desktop/esp-idf/components/esp_wifi C:/Users/V330/Desktop/esp-idf/components/espcoredump C:/Users/V330/Desktop/esp-idf/components/esptool_py C:/Users/V330/Desktop/esp-idf/components/expat C:/Users/V330/Desktop/esp-idf/components/fatfs C:/Users/V330/Desktop/esp-idf/components/freemodbus C:/Users/V330/Desktop/esp-idf/components/freertos C:/Users/V330/Desktop/esp-idf/components/heap C:/Users/V330/Desktop/esp-idf/components/idf_test C:/Users/V330/Desktop/esp-idf/components/jsmn C:/Users/V330/Desktop/esp-idf/components/json C:/Users/V330/Desktop/esp-idf/components/libsodium C:/Users/V330/Desktop/esp-idf/components/log C:/Users/V330/Desktop/esp-idf/components/lwip C:/Users/V330/Desktop/esp-idf/examples/esp32-csi-tool/active_ap/main C:/Users/V330/Desktop/esp-idf/components/mbedtls C:/Users/V330/Desktop/esp-idf/components/mdns C:/Users/V330/Desktop/esp-idf/components/mqtt C:/Users/V330/Desktop/esp-idf/components/newlib C:/Users/V330/Desktop/esp-idf/components/nghttp C:/Users/V330/Desktop/esp-idf/components/nvs_flash C:/Users/V330/Desktop/esp-idf/components/openssl C:/Users/V330/Desktop/esp-idf/components/partition_table C:/Users/V330/Desktop/esp-idf/components/protobuf-c C:/Users/V330/Desktop/esp-idf/components/protocomm C:/Users/V330/Desktop/esp-idf/components/pthread C:/Users/V330/Desktop/esp-idf/components/sdmmc C:/Users/V330/Desktop/esp-idf/components/soc C:/Users/V330/Desktop/esp-idf/components/spi_flash C:/Users/V330/Desktop/esp-idf/components/spiffs C:/Users/V330/Desktop/esp-idf/components/tcp_transport C:/Users/V330/Desktop/esp-idf/components/tcpip_adapter C:/Users/V330/Desktop/esp-idf/components/ulp C:/Users/V330/Desktop/esp-idf/components/unity C:/Users/V330/Desktop/esp-idf/components/vfs C:/Users/V330/Desktop/esp-idf/components/wear_levelling C:/Users/V330/Desktop/esp-idf/components/wifi_provisioning C:/Users/V330/Desktop/esp-idf/components/wpa_supplicant C:/Users/V330/Desktop/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/V330/Desktop/esp-idf/examples/esp32-csi-tool/active_ap/build
[4/818] cmd.exe /C "cd /D C:\Users\V330\Desktop\esp-idf\ex...*********************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[471/818] Performing configure step for 'bootloader'
-- Found Git: C:/Users/V330/.espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1")
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/V330/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/V330/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/V330/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
[... 206 lines cut ...]
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/no-rtti/libstdc++-v3/include/bits/char_traits.h:464: undefined reference to `wmemcpy'
[... cut ...] .0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(collate_members_cow.o): in function `std::collate<wchar_t>::_M_transform(wchar_t*, wchar_t const*, unsigned int) const':
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/no-rtti/libstdc++-v3/src/c++98/collate_members_cow.cc:69: undefined reference to `wcsxfrm'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

StevenMHernandez, 2021-08-23:
From the README.md 

> It is important that you are able to successfully build and flash the example project from the esp-idf guide onto your own esp32. If you have issues building the example project on your hardware, do not create an issue in this github repo. We will not be able to assist with general ESP32 issues (those issues that are unrelated to this project).

Were you able to build and flash examples from the esp-idf guide? This looks like an issue with esp-idf, not an issue with esp32-csi-tool.

StevenMHernandez, 2021-11-02:
I appreciate you commenting with this extra information. It sounds like reinstalling esp-idf fixed the issue. Since this issue is old and is unrelated to the ESP32-CSI-Tool, I am going to close the issue.

closed 2021-11-02 by StevenMHernandez