StevenMHernandez/ESP32-CSI-Tool #33: Which module should I go for?

bachejames, 2021-09-04:
Hi there!

Firstly, thank you for such a fantastic project! Really exciting work.

I'm wondering out of the three modules that have been tested (TTGO T8, NodeMCU & WINGONEER TS V1.2), which would people recommend please?

Aside from TTGO's obvious advantage of having an SD card, it seems like perhaps NodeMCU may be preferable as it seems like NodeMCU has 512KB of SRAM (over TTGO's 320KB of SRAM) and possibly even 16MB of memory (though [this page](https://joy-it.net/en/products/SBC-NodeMCU-ESP32) has conflicting info on that).

Has anyone tested out how important this performance difference is and whether there are any other consequential considerations please?

Thanks so much!

StevenMHernandez, 2021-09-15:
Hello @bachejames This is a great question. This open source project would benefit greatly if it had some documentation  comparing different ESP32 modules.

Since you have done a bit of research into this already, maybe you could start writing up an initial `esp32_board_comparison.md` file that has a table of these features that you described (has_sd_card, SRAM_size, cost, etc). Any initial research into this would be hugely beneficial to the project even if you have not actually tested the modules yourself. It will give others a nice starting point. Great idea!

bachejames, 2021-09-17:
Sure thing @StevenMHernandez! Will give this a go!

bachejames, 2021-09-17:
#36 For reference for people viewing this thread.

StevenMHernandez, 2021-11-15:
Thanks again @bachejames. I will close this issue, but if anyone else would like to contribute more information about different ESP32 models, please send a _pull request_ editing [esp32_module_comparison.csv](https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/master/docs/esp32_module_comparison.csv). Thanks!

closed 2021-11-15 by StevenMHernandez