StevenMHernandez/ESP32-CSI-Tool #34: Amplitude and phase plot looks inconsistent in comparison the plot captured via nexmon CSI or https://github.com/jonathanmuller/ESP32-gather-channel-state-information-CSI-/blob/master/plot_csi_data/amplitude.png

govindsi, 2021-09-14:
**Describe the bug**
Plot captured using sample data in github or with captured data b/w 2 ESP devices.
Is some configuration missing?

[image]

https://github.com/jonathanmuller/ESP32-gather-channel-state-information-CSI-/blob/master/plot_csi_data/amplitude.png capture:
[image]

**Tool Settings**
- Baud Rate: [e.g. 1000000]  (using a lower baud rate *WILL* cause issues. **this is the most common issue**)
- WiFi Channel per Device: [e.g. 6]
- FreeRTOS > Tick rate (Hz): [e.g. 500]

**To Reproduce**
Steps to reproduce the behavior:
1. Build project(s) `...` (active_ap, active_sta, passive)
2. Flash project(s) `...` (active_ap, active_sta, passive)
3. External device setup (if applicable)
3. See error

**Desktop (please complete the following information):**
 - OS: [e.g. macOS, Ubuntu]
 - ESP-IDF Version: [e.g. 4.0]
 - ESP32-CSI-Tool Git Hash: [e.g. 31f9f1be9f803e7770e568cd9eb951b4887bbb68]

**Additional context**
Add any other context about the problem here.

StevenMHernandez, 2021-09-15:
Please fill out the required tool settings in your issue.

Do you see any important differences in the codebase for the two projects that you used when you captured this data? I need you to break down the problem into what you think is an issue. For example:
- What exactly did you try, were there any signal processing steps (i.e. converting the complex CSI subcarrier data to Amplitude/Phase). 
- Is there a slight difference in the code-logic that is being performed that you think should be corrected.

It is a bit hard to tell what you have already tried and what you expect to be correct.

WGQYY, 2023-06-30:
> **描述**使用 github 中的示例数据或使用捕获的数据黑白 2 ESP 设备捕获的错误图。是否缺少某些配置？
> 
> [image]
> 
> https://github.com/jonathanmuller/ESP32-gather-channel-state-information-CSI-/blob/master/plot_csi_data/amplitude.png 捕获： [image]
> 
> **工具设置**
> 
> * 波特率：[例如 1000000]（使用较低的波特率_会导致_问题。**这是最常见的问题**)
> * 每台设备的 WiFi 频道：[例如 6]
> * FreeRTOS > 刻度率 （Hz）： [例如 500]
> 
> **重现重现**行为的步骤：
> 
> 1. 构建项目（active_ap、active_sta、被动）`...`
> 2. Flash 项目（active_ap、active_sta、被动）`...`
> 3. 外部设备设置（如果适用）
> 4. 查看错误
> 
> **桌面（请填写以下信息）：**
> 
> * 操作系统：[例如macOS，Ubuntu]
> * ESP-IDF 版本： [例如 4.0]
> * ESP32-CSI-Tool Git 哈希：[例如 [31f9f1b](https://github.com/StevenMHernandez/ESP32-CSI-Tool/commit/31f9f1be9f803e7770e568cd9eb951b4887bbb68)]
> 
> **其他**上下文 在此处添加有关该问题的任何其他上下文。

WGQYY, 2023-06-30:
Hello, I also encountered this problem, please tell me if you have solved it now

Paco-Danes, 2023-09-23:
Hello, same issue here. The extracted amplitudes are very very variable over subcarriers but also over time (considering a single subcarrier). 
[image]

Thise is a plot for a single subcarrier over ONLY 1 second (100 packets, since the rate is 100pk/s). Is it normal that it changes so rapidly every 10ms ?