StevenMHernandez/ESP32-CSI-Tool #35: Unicode decode error with the python utility, and weird logs in terminal

yc541, 2021-09-14:
Hello!

I am using a PC with Ubuntu 20.04, ESP-IDF 4.0 and NodeMCU kit. 

There are some unicode decode errors when I try to use the python utility to include timestamps (see attached).
[image]
There is a my-experiment-file-sta.csv created but it is empty. 

Also when I just run idf.py monitor without using the python utility, there are some incorrectly decoded logs in the terminal. 

Here are my outputs of locale
[image]

Any help would be appreciated!

StevenMHernandez, 2021-09-15:
This looks indicative of an incorrect serial baud rate can you double and triple check that the baud rate you set for the ubuntu 20.04 machine is correct

yc541, 2021-09-16:
@StevenMHernandez,

Thanks for this! 
I just checked both baud rate are set at 1552000 on the 20.04 machine (the 18.04 machine has the same baud rate). I tried 1152000 on the 20.04 machine as well but the result is the same.

[image]
[image]

StevenMHernandez, 2021-09-17:
> "It is important that you are able to successfully build and flash the example project from the esp-idf guide onto your own esp32."

I need you to try the above.

You need to figure out if it is a problem with the esp32-csi-tool or more likely an external problem (faulty computer hardware/issues with your installed ESP32-IDF/operating system issue/etc.)

Please run one of the example projects from the esp-idf guide (try using the low baud rates and high baud rates). If you are not able to get those working, than it is probably not an issue with this project specifically.

yc541, 2021-09-17:
@StevenMHernandez 

Thanks for the advice! 

I run the example hello world with default 115200 baudrate, it works, but with 1552000 baudrate (the 18.04 machine runs OK with this baudrate) I see similar garbled logs. So I try to reduce the baudrate and it seems that the maximum baudrate I can get is 921600 on the 20.04 machine. Both station and ap run OK with this baudrate. However I'm still getting the unicode error when using the python utility, I will try testing it on a third machine, this seems to be a machine specific issue.

StevenMHernandez, 2021-11-15:
I believe this problem was solved by changing the baud rates. In which case, issue #5 is related.

I will close this for now. Feel free to reopen if the issues were not solved.

closed 2021-11-15 by StevenMHernandez