StevenMHernandez/ESP32-CSI-Tool #37: active_sta mode build error : "main.cc:148:70: error: cannot convert 'esp_interface_t' to 'wifi_interface_t'"

Judgement9882, 2021-09-28:
An error occurred while building in active_sta mode.
(Previously, I successfully collected CSI data in passive mode.)

**Tool Settings**
- Baud Rate: 152000
- WiFi Channel per Device: 7
- FreeRTOS > Tick rate (Hz): 1000

**To Reproduce**
Steps to reproduce the behavior:
1. Build project(s) : active_sta
2. Flash project(s) : active_sta

**Desktop (please complete the following information):**
 - OS: Windows10
 - ESP-IDF Version: 4.4

**Additional context**

[image]

I'm not sure how this code works but this error can be resolved by simply commenting out the line.

[image]

After the above process, I got CSI data.
Hope this helps someone who has the same error.

StevenMHernandez, 2021-09-28:
Hello. Thanks for mentioning the esp-idf version you used.

Currently the project is known to work with v4.0 of ESP-IDF. Small changes may be required to use higher versions, however I have not yet fully tested these higher versions. 

From the project README.md:

> First, Install Espressif IoT Development Framework (ESP-IDF) by following their step by step installation guide. Notice, this project **requires version (v4.0) of ESP-IDF.**

There is an existing issue on this github project to update to a higher version of esp-idf (#11). However, I believe versions 4.3 is the current stable version. Since v4.4 is a stable version, I will not plan update to that version yet. If you would like to help update this open source project to a higher version of esp-idf, please feel free to leave some comments in issue #11. Thanks!

StevenMHernandez, 2021-11-02:
@xudesheng Thanks for identifying this. I am surprised that this is the only change that needs to happen to go from 4.1 to 4.3. There are no other errors?

Does the `passive` project need any updates too?

Please feel free to submit a PR. If you do, please reference #11

xudesheng, 2021-11-02:
> @xudesheng Thanks for identifying this. I am surprised that this is the only change that needs to happen to go from 4.1 to 4.3. There are no other errors?
> 
> Does the `passive` project need any updates too?
> 
> Please feel free to submit a PR. If you do, please reference #11

Thank you, @StevenMHernandez . My 3 ESP32 are now running as: 1 AP with 2 STA. 

I will test `passive` mode later this week and I will come back with the test result here along with PR.

lyndametref, 2021-11-20:
I have also found that the only line to change to make it work are the one mentioned by @xudesheng .
I have 1 device running with the sta version and another with the ap and does not have any error at runtime (both on ESP32-S2-Saola-1). I have also tried on a C3 but get an error about the image being to big for the device memory... but that's another issue and I am happy with the S2.

StevenMHernandez, 2021-11-20:
@lyndametref Thank you for checking. #39 has these changes pending to be merged. First, I want to fix some deprecation errors before updating the project to 4.3.

I have not worked with the S2 or C3 models. You might consider adding information to the [esp32_module_comparison.csv](https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/master/docs/esp32_module_comparison.csv). More info: #33. It would be very useful for everyone to be able to compare different ESP32 models in one place. Thanks!

Judgement9882, 2021-11-25:
> for active_sta, you can modify it like this (line 148):
> 
> ```
> //ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config));
> ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config));
> ```
> 
> for active_ap, you can modify line 115 like:
> 
> ```
> //ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config));
> ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_AP, &wifi_config));
> ```
> 
> Those two constants are different in version 4.3.
> 
> @StevenMHernandez Please let me know if I should submit a PR.

Thanks! It really helped a lot!!

kallu1294, 2021-11-29:
> for active_sta, you can modify it like this (line 148):
>
> ```
> //ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config));
> ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config)); [...]

it's really helped me too run with some warnings that's it but I'm not able to connect mobile in AP mode and could not able to scan the wifi devices in  STA mode

StevenMHernandez, 2022-01-07:
This project is now using esp-idf v4.3 which should have solved this issue. If the problem still persists, please reopen the issue and let me know. Thanks.

closed 2022-01-07 by StevenMHernandez