StevenMHernandez/ESP32-CSI-Tool #38: Unclear if CSI is being collected accurately

bachejames, 2021-10-30:
**Describe the bug**
I have consistently seen a series of subcarriers not giving any data, and the remaining subcarriers giving what seems to be very low levels of amplitude. Is this to be expected please? Here's a typical view of the amplitude/subcarrier (chart created using CSIKit):

[image]

During every single frame, these subcarriers never reported any data: 2, 3, 4, 5, 32, 59, 60, 61, 62, 63 (when counting up from 0).

Thanks so much!

**Tool Settings**
- Baud Rate: 1552000
- WiFi Channel per Device: 6
- FreeRTOS > Tick rate (Hz): 1000

**To Reproduce**
- TTGO T8 as the station
- NodeMCU V4 as the AP

**Desktop (please complete the following information):**
 - OS: macOS
 - ESP-IDF Version: 4.0
 - ESP32-CSI-Tool Git Hash: c9e66fe481b31afb70ca01a4fdd76729503bff44

**Additional context**
The two devices were approximately 1ft apart, with no objects between them and they were on the same horizontal plane.

StevenMHernandez, 2021-11-02:
That is correct. Those are "null subcarriers". They are expected to be empty. Furthermore, subcarrier 0 is the pilot subcarrier which you can also filter out because the amplitude is typically way higher than the other subcarriers. If you filter out subcarrier 0,1,2,3,4,5,6,...,etc and all of the null subcarriers that you had identified you should have 52 remaining subcarriers. There are the "data subcarriers" that you want to use.

I noticed that in your figure that the pilot subcarrier is actually labelled as 63. The numbering from CSIKit must be backwards to what I usually see.

bachejames, 2021-11-06:
This is super useful context @StevenMHernandez, thank you so much! Great spot regarding CSIKit seemingly reversing the data, I wondered what it was doing there.

closed 2021-11-06 by bachejames