StevenMHernandez/ESP32-CSI-Tool #40: When exporting CSI：BrokenPipeError: [Errno 32] Broken pipe

YXW-123, 2021-11-12:
**Describe the bug**
When using the command：idf.py -p /dev/ttyUSB0 monitor | grep "CSI_DATA" > my-experiment-file.csv ,there is a error:BrokenPipeError: [Errno 32] Broken pipe

**Tool Settings**
- Baud Rate: [1152000]  (using a lower baud rate *WILL* cause issues. **this is the most common issue**)
- WiFi Channel per Device: [ 6]
- FreeRTOS > Tick rate (Hz): [1000]

**To Reproduce**
Steps to reproduce the behavior:
1. cd ./active_ap
2. Recording to the README.md use the commond: idf.py menuconfig
3. idf.py -p /dev/ttyUSB0 flash
4. use the commond :idf.py -p /dev/ttyUSB0 monitor ,then let my phone connect to the AP,it can collect CSI successfully.
[image]

5. use the commond:idf.py -p /dev/ttyUSB0 monitor | grep "CSI_DATA" > my-experiment-file.csv , let my phone connect to the AP, then it printed the error like this :
[image]

**Desktop (please complete the following information):**
 - OS: [Ubuntu16.04]
 - ESP-IDF Version: [4.2.1]

StevenMHernandez, 2021-11-12:
Try each of these and report the results:

- `idf.py -p /dev/ttyUSB0 monitor`
- `idf.py -p /dev/ttyUSB0 monitor | grep "CSI_DATA"`
- `idf.py -p /dev/ttyUSB0 monitor | grep "CSI_DATA" > my-experiment-file.csv`

Which one causes the error?

Did you try to monitor the ESP32 from multiple terminals at the same time?

YXW-123, 2021-11-12:
> Try each of these and report the results:
> 
> 1. `idf.py -p /dev/ttyUSB0 monitor`
[image]

> 2. `idf.py -p /dev/ttyUSB0 monitor | grep "CSI_DATA"`
[image]

> 3. `idf.py -p /dev/ttyUSB0 monitor | grep "CSI_DATA" > my-experiment-file.csv`
> 
[image]

> Which one causes the error?
> The 2nd and 3rd

> I used a  terminal for AP and another terminal for sta.But when I only open a terminal for AP,the commond 2nd or 3rd still causes same error.

StevenMHernandez, 2021-11-12:
This ultimately looks like an issue with the esp-idf (not my project). Can you try flashing one of the esp-idf example projects and run `idf.py -p /dev/ttyUSB0 monitor | grep "CSI_DATA"`.

Piping data is a normal task in linux, I am not sure why this would be happening to you. I have found (https://github.com/pyserial/pyserial/issues/139) which has some information on broken pipes with esp-idf, however I am not sure that it will help you. What steps have you taken so far to debug this or to find solutions on the internet?

YXW-123, 2021-11-22:
> This ultimately looks like an issue with the esp-idf (not my project). Can you try flashing one of the esp-idf example projects and run `idf.py -p /dev/ttyUSB0 monitor | grep "CSI_DATA"`.
> 
> Piping data is a normal task in linux, I am not sure why this would be happening to you. I have found ([pyserial/pyserial#139](https://github.com/pyserial/pyserial/issues/139)) which has some information on broken pipes, however I am not sure that it will help you. What steps have you taken so far to debug this or to find solutions on the internet?

Thanks for your help! 
***Firstly,I have added some code like this to miniterm.py:
import sys

try:
   ******************
except (BrokenPipeError, IOError):
    pass

sys.stderr.close()

 I found it  didn't report error ，but it couldn't exporting data either.
***Secondly,I try to use the ESP-IDF v4.3.0,when generate bin，it reported error:
panda@panda-virtual-machine:~/ESP32-CSI-Tool-master/active_ap$  idf.py build

Executing action: all (aliases: build)
Running ninja in directory /home/panda/ESP32-CSI-Tool-master/active_ap/build
Executing "ninja all"...
[2/9] Performing build step for 'bootloader'
ninja: no work to do.
[2/7] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/main.cc.obj
FAILED: /home/panda/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++   -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I/home/panda/esp/esp-idf/components/newlib/platform_include -I/home/panda/esp/esp-idf/components/freertos/include -I/home/panda/esp/esp-idf/components/freertos/port/xtensa/include -I/home/panda/esp/esp-idf/components/esp_hw_support/include -I/home/panda/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/home/panda/esp/esp-idf/components/heap/include -I/home/panda/esp/esp-idf/components/log/include -I/home/panda/esp/esp-idf/components/lwip/include/apps -I/home/panda/esp/esp-idf/components/lwip/include/apps/sntp -I/home/panda/esp/esp-idf/components/lwip/lwip/src/include -I/home/panda/esp/esp-idf/components/lwip/port/esp32/include -I/home/panda/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/panda/esp/esp-idf/components/soc/include -I/home/panda/esp/esp-idf/components/soc/esp32/. -I/home/panda/esp/esp-idf/components/soc/esp32/include -I/home/panda/esp/esp-idf/components/hal/esp32/include -I/home/panda/esp/esp-idf/components/hal/include -I/home/panda/esp/esp-idf/components/esp_rom/include -I/home/panda/esp/esp-idf/components/esp_rom/esp32 -I/home/panda/esp/esp-idf/components/esp_rom/include/esp32 -I/home/panda/esp/esp-idf/components/esp_common/include -I/home/panda/esp/esp-idf/components/esp_system/include -I/home/panda/esp/esp-idf/components/esp32/include -I/home/panda/esp/esp-idf/components/driver/include -I/home/panda/esp/esp-idf/components/driver/esp32/include -I/home/panda/esp/esp-idf/components/esp_ringbuf/include -I/home/panda/esp/esp-idf/components/efuse/include -I/home/panda/esp/esp-idf/components/efuse/esp32/include -I/home/panda/esp/esp-idf/components/xtensa/include -I/home/panda/esp/esp-idf/components/xtensa/esp32/include -I/home/panda/esp/esp-idf/components/espcoredump/include -I/home/panda/esp/esp-idf/components/esp_timer/include -I/home/panda/esp/esp-idf/components/esp_ipc/include -I/home/panda/esp/esp-idf/components/esp_pm/include -I/home/panda/esp/esp-idf/components/vfs/include -I/home/panda/esp/esp-idf/components/esp_wifi/include -I/home/panda/esp/esp-idf/components/esp_wifi/esp32/include -I/home/panda/esp/esp-idf/components/esp_event/include -I/home/panda/esp/esp-idf/components/esp_netif/include -I/home/panda/esp/esp-idf/components/esp_eth/include -I/home/panda/esp/esp-idf/components/tcpip_adapter/include -I/home/panda/esp/esp-idf/components/app_trace/include -I/home/panda/esp/esp-idf/components/mbedtls/port/include -I/home/panda/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/panda/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/home/panda/esp/esp-idf/components/app_update/include -I/home/panda/esp/esp-idf/components/spi_flash/include -I/home/panda/esp/esp-idf/components/bootloader_support/include -I/home/panda/esp/esp-idf/components/nvs_flash/include -I/home/panda/esp/esp-idf/components/pthread/include -I/home/panda/esp/esp-idf/components/esp_gdbstub/include -I/home/panda/esp/esp-idf/components/esp_gdbstub/xtensa -I/home/panda/esp/esp-idf/components/esp_gdbstub/esp32 -I/home/panda/esp/esp-idf/components/perfmon/include -I/home/panda/esp/esp-idf/components/wpa_supplicant/include -I/home/panda/esp/esp-idf/components/wpa_supplicant/port/include -I/home/panda/esp/esp-idf/components/wpa_supplicant/include/esp_supplicant -I/home/panda/esp/esp-idf/components/protobuf-c/protobuf-c -I/home/panda/esp/esp-idf/components/wear_levelling/include -I/home/panda/esp/esp-idf/components/sdmmc/include -I/home/panda/esp/esp-idf/components/fatfs/diskio -I/home/panda/esp/esp-idf/components/fatfs/vfs -I/home/panda/esp/esp-idf/components/fatfs/src -I/home/panda/esp/esp-idf/components/esp_adc_cal/include -I/home/panda/esp/esp-idf/components/nghttp/port/include -I/home/panda/esp/esp-idf/components/nghttp/nghttp2/lib/includes -I/home/panda/esp/esp-idf/components/esp-tls -I/home/panda/esp/esp-idf/components/esp-tls/esp-tls-crypto -I/home/panda/esp/esp-idf/components/tcp_transport/include -I/home/panda/esp/esp-idf/components/esp_websocket_client/include -I/home/panda/esp/esp-idf/components/spiffs/include -I/home/panda/esp/esp-idf/c [... cut ...]
         esp_err_t __err_rc = (x);                                       \
                               ^
In file included from ../main/main.cc:7:
/home/panda/esp/esp-idf/components/esp_wifi/include/esp_wifi.h:800:48: note:   initializing argument 1 of 'esp_err_t esp_wifi_set_config(wifi_interface_t, wifi_config_t*)'
 esp_err_t esp_wifi_set_config(wifi_interface_t interface, wifi_config_t *conf);
                               ~~~~~~~~~~~~~~~~~^~~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

StevenMHernandez, 2021-11-23:
#37 deals with issues with esp-idf 4.3

Please review the thread. It will tell you how you can get rid of this issue in v4.3

StevenMHernandez, 2022-01-07:
This project is now using esp-idf v4.3 which should have solved this issue. If the problem still persists, please reopen the issue and let me know. Thanks.

YXW-123, 2022-01-09:
Thanks for your advice.I have tried to using esp-idf(release v.4.3.1),but the problem still persists.
[image]

YXW-123, 2022-01-23:
Thanks fo your advice .I solved this problem through the following steps：
1.OS: change [Ubuntu16.04] to 20.04 ( the python version changed from 3.7 to 3.8,I guess it has something to do with this)
2.ESP-IDF Version: change [4.2.1] to 4.3
3.Using the latest project you provided 
4.Run command`idf.py monitor | grep "CSI_DATA" > my-experiment-file.csv `，it can work correctly.
5.Run command `idf.py monitor | python ./python_utils/serial_append_time.py > my-experiment-file.csv`,[Errno 32]Broken pipe again,and it prompts that the file or directory ' ./python_utils/serial_append_time.py 'cannot be found.
6. Change the commond to `idf.py monitor | python ../python_utils/serial_append_time.py > my-experiment-file.csv`,it can work correctly.

StevenMHernandez, 2022-01-24:
Does this mean your issue has been resolved? Can I close the issue?

YXW-123, 2022-01-24:
Yes!

closed 2022-01-24 by StevenMHernandez