StevenMHernandez/ESP32-CSI-Tool #41: Not working with v4.0 since xtensa-esp32-elf tool version in esp-idf v4 has an issue

ahmed1hisham, 2021-11-15:
**Describe the bug**
When building the project using esp-idf v4.0, the build fails due to some undefined references in the Xtensa-esp32-elf tool children files (eg. collate_members_cow.cc)

**Tool Settings**
- Baud Rate: 1552000
- WiFi Channel per Device: 6
- FreeRTOS > Tick rate (Hz): 1000

**To Reproduce**
Steps to reproduce the behavior:
1. Install ESP-IDF v4.0 (while making sure example project (blink) builds and flashes successfully)
2. Build project(s) `active_sta`
3. See error

**Desktop (please complete the following information):**
 - OS: macOS
 - ESP-IDF Version: 4.0

**Additional context**
Upgrading to esp-idf v4.3 uses a newer version of the xtensa-esp32-elf tool, so the build error is solved... but the other upgrading issues described in [this PR](https://github.com/StevenMHernandez/ESP32-CSI-Tool/pull/39) still exist.

StevenMHernandez, 2021-11-15:
Do you get the same error log as #32? [One person](https://github.com/StevenMHernandez/ESP32-CSI-Tool/issues/32#issuecomment-950835602) said that they were able to solve this issue by "reinstalled everything from the beginning". 
If it is a different error, please paste and format the relevant error traces.

I use the same exact settings as you (v4.0, macOS), but I have never seen this error. What is the git-hash of the esp-idf project that you are using?

It is very strange that the example projects work fine from esp-idf, but not this project. `Xtensa-esp32-elf` and `collate_members_cow.cc` are not part of this ESP32-CSI-Tool project.

ahmed1hisham, 2021-11-15:
@StevenMHernandez yes it is the same error as #32 ... But "reinstalling everything" multiple times actually, still didn't work for me... Also, can you please guide me on where can I find the git has of the esp-idf I'm using?

Thanks a lot! 🙏

StevenMHernandez, 2021-11-16:
From your terminal:

```
cd ~/esp/esp-idf  # Where ever you installed (git-cloned) the esp-idf
git rev-parse HEAD
```

My version of 4.0 has the following git-hash `7504329ef9fb01f01a382416af7453c54b47fb2b`.

You should be able to run `git checkout 7504329ef9fb01f01a382416af7453c54b47fb2b` to use the *exact* same esp-idf version as me.

ahmed1hisham, 2021-11-17:
Turns out this Xtensa error happens if you're on the latest release/4.0 branch.
Checking out the hash you provided solved it for me! 🎉  Thanks a million! 🙏 

I have one follow up question though... I see that the csi_data is printed at random inconsistent times... isn't it supposed to be printed in a regular manner? like every second maybe?
I haven't collected realtime CSI data before so excuse me if that sounds trivial or obvious 🙏 

Thanks a lot for your help!

StevenMHernandez, 2021-11-17:
That is very strange. Do you see an issue created on the esp-idf github project? I really do not like that you need to use my exact version, because v4.0 should _just work_. However, I am very glad that you mentioned previously that this is not an issue with v4.3, so everything should be fixed once this project is finally upgraded.

For your next question, the active_sta prints any and all CSI that it receives as soon as it receives it. What device in your environment is sending CSI frames? If your device (i.e. active_ap) is not sending frames at a consistent rate, then the CSI printed will not be consistent either. I am guessing that you do not have an `active_ap`.

ahmed1hisham, 2021-11-17:
I couldn't find a related issue on the esp-idf project. I know that seems very strange, but I tested it multiple times on different operating systems and it was consistent. But yes, on the bright side, supporting 4.3 would solve that issue.

Oh I see. Yes you are right I don't have an `active_ap` ... is there any way you know of so that we can ping the router from the `active_sta` ? I need to work with the one esp32 I currently have.

Thanks a lot in advance! 🙏

StevenMHernandez, 2021-11-18:
I do not have a general method for working with one esp32. If you connect a device to the active_sta and send frames, you should be able to capture CSI **as long as the packets sent are able to generate CSI**. I cannot give any guidance on how you can accomplish this in a reliable manner.
However, if you do happen to find a **reliable** way to generate these CSI through pings, please let me know. It would be a good open source contribution to document how to do this.

StevenMHernandez, 2022-01-07:
This project is now using esp-idf v4.3 which should have solved this issue. If the problem still persists, please reopen the issue and let me know. Thanks.

closed 2022-01-07 by StevenMHernandez