StevenMHernandez/ESP32-CSI-Tool #42: why i can not able to access the settings which you are mentioned for starting

kallu1294, 2021-11-23:
**Describe the bug**
A clear and concise description of what the bug is.

**Tool Settings**
- Baud Rate: [e.g. 1000000]  (using a lower baud rate *WILL* cause issues. **this is the most common issue**)
- WiFi Channel per Device: [e.g. 6]
- FreeRTOS > Tick rate (Hz): [e.g. 500]

**To Reproduce**
Steps to reproduce the behavior:
1. Build project(s) `...` (active_ap, active_sta, passive)
2. Flash project(s) `...` (active_ap, active_sta, passive)
3. External device setup (if applicable)
3. See error

**Desktop (please complete the following information):**
 - OS: [e.g. macOS, Ubuntu]
 - ESP-IDF Version: [e.g. 4.0]
 - ESP32-CSI-Tool Git Hash: [e.g. 31f9f1be9f803e7770e568cd9eb951b4887bbb68]

**Additional context**
Add any other context about the problem here.

kallu1294, 2021-11-23:
why i can not able to access the settings which you are mentioned for starting..??

StevenMHernandez, 2021-11-23:
You need to provide more information. Specifically, you need to mention the "steps to reproduce" and other information such as "OS", "ESP-IDF", "ESP32-CSI-Tool Git Hash". 

Please edit the original issue message with relevant information. 

Let me know when you have edited the information. The more _useful_ information that you give, the easier it will be to help you.

closed 2021-11-29 by kallu1294