StevenMHernandez/ESP32-CSI-Tool #43: can't able to download csi data

ezio1294, 2021-11-25:
now im getting the msg like this

-- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6840
load:0x40078000,len:14836
load:0x40080400,len:3804
0x40080400: _init at ??:?

entry 0x40080698
0�hB1X�
          d���&�h�
 �L�&	$�d���!
�������@������
                  �0�������d��dd&�6,0,Ȍ`h�c2X�8�L(L�*d@����CZ(]Bid&+�	�ࡸ�px�v۾���PDE��*Q��dSH�v�.kH�b,ZKd��tb�̂�"�"f,�X�L���S�t����

what does it mean ? And i need to download the msg from my serial port which i can't able to do any solution for this.??
And I'm using esp32WROOM32 will it support for passive file which is there in this document.??

StevenMHernandez, 2021-11-26:
What baud rate are you using? When you use `idf.py menuconfig` how many places do you set the baud rate? Did you miss any? Go back through the README and redo every step to make sure you did not miss something.

I do not understand what the following sentence means.

> And i need to download the msg from my serial port which i can't able to do any solution for this.??

Also, there is no "esp-idf version 5" that I am aware of. The version should be in the form of `X.Y` where X is the major version and Y is the minor version. Both numbers are important.

StevenMHernandez, 2022-03-06:
It has been >3 months. I am guessing that you have solved this issue, so I will close it. Please feel free to open it if you have any further questions.

closed 2022-03-06 by StevenMHernandez