StevenMHernandez/ESP32-CSI-Tool #44: issue with downloading data from serial

kallu1294, 2021-11-25:
when i run the following command to download the data from serial

idf.py monitor | grep "CSI_DATA" > my-experiment-file.csv 

then it is showing as follows

"No serial ports found. Connect a device, or use '-p PORT' option to set a specific port"

if i dont have serial port connected means how can do flash monitor.I have serial port connected by the name of ttyUSB0

StevenMHernandez, 2021-11-26:
Did you add `-p` as recommended?

`idf.py monitor -p /your/path/to/ttyUSB0 | grep "CSI_DATA" > my-experiment-file.csv`

If that helps, please let me know and close the issue.

kallu1294, 2021-11-29:
Sir while collecting data i should use only passive file from your document right.??
if i use only passive file means I'm getting like below and its getting stop there itself. So what can i should do to collect the CSI data now.?? 

type,role,mac,rssi,rate,sig_mode,mcs,bandwidth,smoothing,not_sounding,aggregation,stbc,fec_coding,sgi,noise_floor,ampdu_cnt,channel,secondary_channel,local_timestamp,ant,sig_len,rx_state,real_time_set,real_timestamp,len,CSI_DATA

kallu1294, 2021-11-29:
Did you add -p as recommended?

idf.py monitor -p /your/path/to/ttyUSB0 | grep "CSI_DATA" > my-experiment-file.csv

If that helps, please let me know and close the issue.

Yes sir this command is working but I'm getting blank file after running the above command .So what should i do now to get the CSI data.??

kallu1294, 2021-11-29:
Did you add -p as recommended?

idf.py monitor -p /your/path/to/ttyUSB0 | grep "CSI_DATA" > my-experiment-file.csv

If that helps, please let me know and close the issue.

Thank you so much sir it's working fine :-)
But what is the password for AP file

kallu1294, 2021-11-29:
> Did you add `-p` as recommended?
> 
> `idf.py monitor -p /your/path/to/ttyUSB0 | grep "CSI_DATA" > my-experiment-file.csv`
> 
> If that helps, please let me know and close the issue.

Thank you so much sir it's working fine :-)
But what is the password for the AP file

kallu1294, 2021-11-29:
> Did you add `-p` as recommended?
> 
> `idf.py monitor -p /your/path/to/ttyUSB0 | grep "CSI_DATA" > my-experiment-file.csv`
> 
> If that helps, please let me know and close the issue.

Sir as you said ran command using -p but the following is popping up  :-(

Exception in thread Thread-2:
Traceback (most recent call last):
  File "/home/mwt/.espressif/python_env/idf5.0_py3.6_env/lib/python3.6/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/esp/esp-idf/active_ap/ttyUSB0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/mwt/esp/esp-idf/tools/idf_monitor_base/stoppable_thread.py", line 58, in _run_outer
    self.run()
  File "/home/mwt/esp/esp-idf/tools/idf_monitor_base/serial_reader.py", line 51, in run
    self.serial.open()
  File "/home/mwt/.espressif/python_env/idf5.0_py3.6_env/lib/python3.6/site-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /esp/esp-idf/active_ap/ttyUSB0: [Errno 2] No such file or directory: '/esp/esp-idf/active_ap/ttyUSB0'

StevenMHernandez, 2021-11-29:
@kallu1294 You are making a lot of new issues which makes it hard for me to keep track of which problems still exist and which problems you have solved on your own. 

For example, based on #45, it sounds like you were able to solve the issue in this current thread. Is that true?

If you were not able to solve the issue, the error that you posted says:

> No such file or directory: '/esp/esp-idf/active_ap/ttyUSB0'

What is the file path to `ttyUSB0`? You need the *full_path*. I assume it is `/dev/ttyUSB0`, but you will need to figure it out on your own computer.

kallu1294, 2021-11-30:
> @kallu1294 You are making a lot of new issues which makes it hard for me to keep track of which problems still exist and which problems you have solved on your own.
>
> For example, based on #45, it [...]

I'm extremely sorry sir :-( I'm trying to figure it out myself before i ask here but I'll try once again sir

kallu1294, 2021-11-30:
> @kallu1294 You are making a lot of new issues which makes it hard for me to keep track of which problems still exist and which problems you have solved on your own.
>
> For example, based on #45, it [...]

Yes sir this problem is solved for me but the csv txt file is empty  because I'm not transmitting or receiving any data.Sir pleasecan you rectify my #45 problem :-(

closed 2021-11-30 by StevenMHernandez