StevenMHernandez/ESP32-CSI-Tool #45: AP not connecting to other devices

kallu1294, 2021-11-29:
Hi everyone I'm really thankful to the steven sir and some other authors

Now I'm ready with steven sir's AP code and running it successfully but the problem arises when i try to connect any other devices with esp32 

The following use to display

I (947221) wifi:new:<6,1>, old:<6,0>, ap:<6,1>, sta:<255,255>, prof:6
I (947223) wifi:station: 00:1a:ef:58:29:85 join, AID=1, bgn, 40U
I (951227) wifi:station: 00:1a:ef:58:29:85 leave, AID = 1, bss_flags is 658530, bss:0x3ffbb84c
I (951228) wifi:new:<6,0>, old:<6,1>, ap:<6,1>, sta:<255,255>, prof:6
I (967688) wifi:new:<6,1>, old:<6,0>, ap:<6,1>, sta:<255,255>, prof:6
I (967690) wifi:station: 00:1a:ef:58:29:85 join, AID=1, bgn, 40U

please help me out and let me collect the csi data successfully :-(

StevenMHernandez, 2021-11-29:
I am not sure what the problem or bug is here. 

You are using the active_ap. Are you also using the active_sta on a second ESP32? The active_ap can **receive** CSI, but you also need a device to connect and **transmit** CSI. Otherwise the active_ap is just sitting and waiting for CSI.

The easiest way to accomplish this is through the use of a second ESP32 running the active_sta firmware. It is possible to use other devices as well, but this is an advanced use case that requires user's to have further advanced knowledge. This means: if you use anything other than an ESP32, I can not promise that I will be able to help you.

kallu1294, 2021-11-30:
> I am not sure what the problem or bug is here.
>
> You are using the active_ap. Are you also using the active_sta on a second ESP32? The active_ap can **receive** CSI, but you also need a device to [...]

Thank you for the reply actually I'm using one esp32 as active_ap and trying to connect that with other devices like mobile,laptop or desktop while connecting to those devices it is giving problem like above which i have mentioned

StevenMHernandez, 2021-11-30:
I am still not sure what the problem is though. You need to details the steps you have taken otherwise I cannot guess what your setup is. Are you connecting the external laptop device to the esp32 AP? How did you do this? Have you set up this external laptop device to transmit any packets? How did you do this? 

I will reiterate: if you use anything other than an ESP32 for **transmitting CSI**, I can not promise that I will be able to help you.

If you find any useful tools or have any recommendations for how to generate CSI packets from non-ESP32 devices, please feel free to share your discoveries.

kallu1294, 2021-11-30:
> I am still not sure what the problem is though. You need to details the steps you have taken otherwise I cannot guess what your setup is. Are you connecting the external laptop device to the esp32 [...]

sure sir

kallu1294, 2021-11-30:
> I am still not sure what the problem is though. You need to details the steps you have taken otherwise I cannot guess what your setup is. Are you connecting the external laptop device to the esp32 [...]

Sir i have one doubt can we scan the available wifi devices in our surroundings and connect to the particular wifi to esp32 in STA mode..??

kallu1294, 2021-12-01:
](url)

kallu1294, 2021-12-01:
> I think you need to review what the difference is between an AP and an STA.
> 
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html
> 
> https://defineinstruments.com/blog/networking-iot-and-wifi-101/
> 
> AP does not connect to other devices. Other devices connect to the AP.

Yes sir i know the difference between AP and STA quite well.
But when i made esp32 as STA there are many active wifi devices around me but still it's not connecting to any of open wifi

StevenMHernandez, 2021-12-01:
Please share all of the settings that you set with menuconfig. Specifically under `ESP32 CSI Tool Config > ****`.

kallu1294, 2021-12-01:
> Please share all of the settings that you set with menuconfig. Specifically under `ESP32 CSI Tool Config > ****`.

1) serial flasher config --> 'idf.py monitor' baud rate (921600 bps)
2) ESP32 CSI Tool Config >
     (6) WiFi Channel
     (myssid) WiFi SSID
     (mypassword) WiFi Password
     (100) Packet TX Rate
      [*] Should this ESP32 collect and print CSI data?
      [*]     (Advanced users only) Should we only collect LLTF?
      [*]     Send CSI data to Serial
      [ ]     Send CSI data to SD

3)component  config > 
   . ESP system settings > UART console baud rate >921600
   . wIFI > [*] WiFi CSI(Channel State Information)
   .FreeRTOS> (1000) Tick rate (Hz)

kallu1294, 2021-12-01:
> Please share all of the settings that you set with menuconfig. Specifically under `ESP32 CSI Tool Config > ****`.

Thank you sir i rectified my problem now it was a silly mistake i made but i learnt something from that
Now I'm able to connect esp32 to my wifi device(router)  and now it's showing like sending frames

kallu1294, 2021-12-01:
Sir i established a connection between esp32(STA) and my wifi router(AP) and now I'm getting some  values on my console by the name of CSI_DATA,STA are they really a CSI values ? If "YES" then why I'm not able to download those values.? If "NO" then what should i do to get CSI DATA  downloaded.?

Once again I apologize for a silly question

StevenMHernandez, 2021-12-01:
Are you deleting your messages? I received an email saying that you left a comment with your menuconfig parameters, but I cannot find the comment anymore. It makes it very confusing for me to help track down  and understand your problems when the messages keep disappearing.

In the future, other users might look through these conversations to solve their own problems. If you made a mistake, someone in the future will likely make the same mistake. You say:

> i rectified my problem it was just a silly mistake

but you did not share the mistake or the solution. If you share the mistake and the solution, then it could help user's in the future.

Also, there are no silly questions. We do not know how to walk or talk when we are born, we have to learn how to do these things.

StevenMHernandez, 2021-12-01:
Now on to your question. Yes the data labelled `CSI_DATA,STA,...` is the CSI data. What do you mean by:

> I'm not able to download those values?

What do you mean by "download"?

Share a small sample of the output lines that start with `CSI_DATA,STA` so that I can understand what you are looking at.

Imagine I am all the way on the other side of the planet and I am trying to help solve problems with you. Imagine that I cannot see what you have done. Imagine that I cannot know what you have tried or what you have not tried yet. Imagine that I cannot read your mind to understand what actual problems are happening. With that imagined scene in your mind, make sure to add as much **useful** information to your comments as you can. The better that you can explain what is happening, the better that I can help you. Read and reread your comments before you post them to make sure that they are readable and can be understood by someone that is not in the same room with you.

kallu1294, 2021-12-02:
> Are you deleting your messages? I received an email saying that you left a comment with your menuconfig parameters, but I cannot find the comment anymore. It makes it very confusing for me to help [...]

Sorry sir for this actually in the ESP32 CSI tool config i did not set my wifi(router) SSID name and PASSWORD that's why i was not able to connect my to wifi before . Before i rectify my mistake the settings were like this

ESP32 CSI tool config 
  

(6) WiFi Channel
(myssid) WiFi SSID
(mypassword)  WiFi Password
(100) Packet TX Rate
[ * ] Should this ESP32 collect and print CSI data?
[ * ]     (Advanced users only) Should we only collect LLTF?
[ * ]     Send CSI data to Serial
[   ]     Send CSI data to SD

kallu1294, 2021-12-02:
> Now on to your question. Yes the data labelled `CSI_DATA,STA,...` is the CSI data. What do you mean by:
>
>
> What do you mean by "download"?
>
> Share a small sample of the output lines that start with [...]

I mean I'm getting the values(data) like below if these are CSI data only then why I'm can not able to download those by using the command "idf.py moniotr -p /dev/ttyUSB0 |grep "CSI_DATA" > sta-file.csv"

initial wifi connection established.
sending frames.
CSI_DATA,STA,38:D5:47:42:F6:D0,-58,11,1,7,0,1,1,1,1,0,0,-96,1,1,0,1610937,0,134,0,0,1.76928,384,[-122 96 8 0 -19 3 -15 2 -11 2 -5 1 1 1 6 1 10 1 14 1 19 1 21 2 23 2 25 3 24 3 24 4 21 5 19 5 15 5 10 5 6 3 2 2 -3 0 -7 -3 -10 -6 -13 -8 -14 -11 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 5 -2 7 -5 10 -8 12 -12 15 -14 16 -16 18 -17 19 -18 20 -17 20 -17 19 -15 17 -13 14 -13 11 -10 6 -9 2 -6 -3 -5 -8 -3 -12 -1 -17 1 -20 3 -24 4 -27 4 -27 4 -28 5 -28 5 ]
CSI_DATA,STA,38:D5:47:42:F6:D0,-60,11,1,7,0,1,1,1,1,0,0,-96,1,1,0,5889409,0,98,0,0,6.04754,384,[98 32 6 0 -7 -6 -3 -4 1 -4 4 -5 7 -5 9 -6 11 -8 12 -10 12 -12 12 -12 11 -14 9 -15 8 -15 6 -15 5 -14 3 -13 2 -11 2 -9 1 -7 1 -5 1 -3 2 -1 3 1 3 2 4 3 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 -9 17 -13 19 -15 20 -16 21 -17 22 -17 21 -16 21 -15 19 -14 17 -13 14 -11 10 -10 6 -9 1 -8 -4 -8 -8 -9 -11 -9 -15 -11 -17 -13 -19 -14 -20 -16 -20 -18 -20 -18 -18 -18 -16 -17 -14 -15 -12 ]
CSI_DATA,STA,38:D5:47:42:F6:D0,-60,11,1,7,0,1,1,1,1,0,0,-96,1,1,0,5900711,0,98,0,0,6.05851,384,[98 32 6 0 -25 6 -25 7 -23 10 -22 11 -19 11 -16 11 -12 10 -10 9 -7 7 -5 5 -3 2 -2 -1 -2 -5 -1 -7 -2 -9 -3 -11 -3 -13 -5 -14 -6 -15 -7 -15 -8 -14 -8 -14 -8 -13 -8 -12 -7 -11 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 9 -20 10 -17 9 -12 10 -7 9 -1 9 4 9 9 9 14 10 19 11 23 12 25 13 28 14 28 14 28 14 26 14 23 13 20 11 16 9 13 6 10 2 7 -2 5 -7 3 -12 2 -16 2 -19 3 ]
CSI_DATA,STA,38:D5:47:42:F6:D0,-60,11,1,6,0,1,1,1,1,0,0,-96,1,1,0,5921241,0,98,0,0,6.07906,384,[98 32 6 0 17 -18 17 -20 15 -21 13 -21 10 -20 8 -18 5 -16 3 -13 1 -10 0 -7 0 -3 0 -1 1 3 2 5 3 8 5 10 6 11 8 12 9 12 10 12 10 11 10 11 10 10 9 10 8 9 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 12 19 8 18 4 15 0 11 -4 7 -8 4 -13 0 -17 -3 -22 -6 -25 -8 -28 -10 -30 -12 -31 -12 -31 -11 -30 -10 -28 -8 -25 -6 -22 -5 -17 -4 -12 -3 -7 -4 -2 -5 3 -6 8 -8 12 -11 14 -13 ]
CSI_DATA,STA,38:D5:47:42:F6:D0,-60,11,1,7,0,1,1,1,1,0,0,-96,1,1,0,5945501,0,100,0,0,6.10328,384,[100 64 6 0 9 -26 7 -26 6 -26 4 -25 2 -23 0 -21 -1 -18 -2 -14 -3 -10 -3 -7 -1 -3 0 0 2 3 3 5 6 7 8 9 9 10 11 10 12 10 13 10 13 9 13 9 13 9 12 8 10 7 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 23 4 20 5 14 6 9 7 3 6 -4 6 -9 7 -15 7 -20 7 -25 8 -29 8 -32 9 -33 10 -33 10 -31 10 -29 10 -25 10 -21 8 -16 6 -12 5 -8 1 -3 -2 1 -6 4 -11 7 -14 8 -19 ]

StevenMHernandez, 2021-12-02:
Great, that helps understand the problem a little bit better. However, I still do not understand:

> why I can not able to download those by using the command "idf.py moniotr -p /dev/ttyUSB0 |grep "CSI_DATA" > sta-file.csv"

Is there an error that is occurring? Does the command generate a blank file? What are you **_expecting_** to see specifically and what are you **_actually_** seeing? You need to be more specific about what problems are happening or I cannot help you. Here are some example of Good ways to ask technical questions and a Bad way to ask technical questions.

> Good Example: 
> I run `idf.py monitor -p /dev/ttyUSB0 | grep "CSI_DATA" > sta-file.csv`. 
>`sta-file.csv` is created, but the file is empty. I tried breaking down the problem myself by doing X, Y and Z. 
> X works fine
> Y works fine
> Z seems to cause an issue.

> Bad Example: 
> I run `idf.py monitor -p /dev/ttyUSB0 | grep "CSI_DATA" > sta-file.csv`. But data does not exist.

I hope you can recognize the difference in approaches for the two example.

Finally, please see the following: https://github.com/StevenMHernandez/ESP32-CSI-Tool/issues/6#issuecomment-801992612

kallu1294, 2021-12-02:
> Great, that helps understand the problem a little bit better. However, I still do not understand:
> 
> > why I can not able to download those by using the command "idf.py moniotr -p /dev/ttyUSB0 |grep "CSI_DATA" > sta-file.csv"
> 
> Is there an error that is occurring? Does the command generate a blank file? What are you **_expecting_** to see specifically and what are you **_actually_** seeing? You need to be more specific about what problems are happening or I cannot help you. Here are some example of Good ways to ask technical questions and a Bad way to ask technical questions.
> 
> > Good Example:
> > I run `idf.py monitor -p /dev/ttyUSB0 | grep "CSI_DATA" > sta-file.csv`.
> > `sta-file.csv` is created, but the file is empty. I tried breaking down the problem myself by doing X, Y and Z.
> > X works fine
> > Y works fine
> > Z seems to cause an issue.
> 
> > Bad Example:
> > I run `idf.py monitor -p /dev/ttyUSB0 | grep "CSI_DATA" > sta-file.csv`. But data does not exist.
> 
> I hope you can recognize the difference in approaches for the two example.
> 
> Finally, please see the following: [#6 (comment)](https://github.com/StevenMHernandez/ESP32-CSI-Tool/issues/6#issuecomment-801992612)

Thank you sir for guidelines 
Yes the command is generating the blank(empty) file it doesn't contain any data in it

kallu1294, 2021-12-02:
1) sir if run echo "testing" > new_file.csv 
I have CSI_DATA in my console and the csv is file created if i open that file in libreoffice it is empty with single name CSI_DATA  
[image]

2) sir if i run make monitor > new_file.csv the csv file is created but it is empty
[image]

3) if i run make monitor | grep "CSI_DATA" the csv file is not getting created 
4) and if i run "make monitor | grep "CSI_DATA" > new_file.csv" and "make monitor | cat > new_file.csv" the same thing happening like what i said in point no 2.

what is happening really i don't know

One more question sir why I'm getting csi data delayed by more than 5-8 seconds

kallu1294, 2021-12-06:
Sorry to disturb you sir. Can you please help me how can i download that csv file now ;-(

StevenMHernandez, 2021-12-07:
Sorry, I have been busy with my own work and I have been unable to look at issues. 

Take a look at learning about "[unix-pipes](https://www.tutorialspoint.com/unix/unix-pipes-filters.htm)" and "[unix-redirection](https://www.tutorialspoint.com/unix/unix-io-redirections.htm)". These concepts will be very useful for your future knowledge. It will also help understand _why_ I was asking you to perform the 4 steps and what you should expect to see after each step. 

For example: 1 and 3 are doing exactly what would be expected. Number 2 and 4 are not working as expected. Number 2 is a simplified version of number 4.

This means we should look at why number 2 is not working. Thank you for adding screenshots, this helps with this process. It looks like when you run `make monitor > sat_file.csv` the process stops with some (Error?) message:

> `...make target 'monitor'. Stop.

Why is the process stopping? Did you stop the process manually? Actually, I think I already helped you with this exact problem See #44.

kallu1294, 2021-12-08:
> Sorry, I have been busy with my own work and I have been unable to look at issues.
>
> Take a look at learning about "[unix-pipes](https://www.tutorialspoint.com/unix/unix-pipes-filters.htm)" and [...]

i didn't stop manually until it that csv file get created. I'm just stuck at the end point (not able to get CSI data in csv file) :-(

StevenMHernandez, 2021-12-08:
Do you understand unix pipes and redirection?

kallu1294, 2021-12-09:
> Do you understand unix pipes and redirection?

Yes sir I'm understanding it, Now my question is I'm using a single esp32 board(STA mode) is that the reason why I'm not able to get csi data in csv file.??
Sorry for asking again and again. If i get to know is that the reason means i could use another esp32 board as AP

StevenMHernandez, 2021-12-09:
Oh I see. Please use `idf.py` instead of `make`. 

Also, do not forget `-p` from #44

kallu1294, 2021-12-10:
> Oh I see. Please use `idf.py` instead of `make`.
> 
> Also, do not forget `-p` from #44

Yes sir i did that too idf.py adding -p to it
but once you check the images which i got  

when i run 

1. **echo "testing" > new_file.csv** 
   the csv file is created but no CSI data in that 
[image]

2.**make monitor > new_file.csv** 
csv file is created but showing something else which i don't have any idea 
[image]

3.when i run this command nothing is happening **make monitor | grep "CSI_DATA** 

4.**make monitor | grep "CSI_DATA" > new_file.csv**  
csv is getting created but it is empty
[image]

5.**make monitor | cat > new_file.csv** 
when i run this same thing happening as point 2
[image]

And finally when i run **idf.py monitor -p  /dv/ttyUSB0 | grep "CSI_DATA" > my-experiment-file.csv** here also csv file is getting created but it is empty

Sir please can you help me how can i download CSI DATA however CSI data is getting printed on serial but not on csv file

StevenMHernandez, 2021-12-10:
Did you read the error in `new_file1.csv` and `new_file3.csv`?

---------

Run these three steps and identify the errors:

1. `idf.py monitor -p /dev/ttyUSB0`
2. `idf.py monitor -p /dev/ttyUSB0 | grep "CSI_DATA"`
3. `idf.py monitor -p /dev/ttyUSB0 | grep "CSI_DATA" > my-experiment-file.csv`

kallu1294, 2021-12-13:
> Did you read the error in `new_file1.csv` and `new_file3.csv`?
> 
> Run these three steps and identify the errors:
> 
>     1. `idf.py monitor -p /dev/ttyUSB0`
> 
>     2. `idf.py monitor -p /dev/ttyUSB0 | grep "CSI_DATA"`
> 
>     3. `idf.py monitor -p /dev/ttyUSB0 | grep "CSI_DATA" > my-experiment-file.csv`

Sir if i run 
**idf.py monitor -p /dev/ttyUSB0** this command runs same as the **idf.py flash monitor** and connect to my AP (wifi router) and I'll start getting CSI_DATA on my serial monitor

 If i run **idf.py monitor -p /dev/ttyUSB0 | grep "CSI_DATA"**  it shows binary file (standad input) matches   and **idf.py monitor -p /dev/ttyUSB0 | grep "CSI_DATA" > sta-file.csv** and in csv file it shows like below image
[image]

And if i run directly **idf.py monitor -p /dev/ttyUSB0 | grep "CSI_DATA" > sta-file.csv** after **idf.py flash monitor** the same thing is coming on the csv file as in the image

StevenMHernandez, 2021-12-13:
Show me a screenshot for: 

1. `idf.py monitor -p /dev/ttyUSB0`

---------

Did you search for [_"binary file (standad input) matches"_](https://unix.stackexchange.com/questions/335716/grep-returns-binary-file-standard-input-matches-when-trying-to-find-a-string)? 

---------

None of the commands I shared should automatically exit. Are they?

kallu1294, 2021-12-14:
> Show me a screenshot for:
>
>     1. `idf.py monitor -p /dev/ttyUSB0`
>
>
> Did you search for [_"binary file (standad input) [...]

No sir they are not exiting automatically just i asked for the information whether we manually need to exit or they automatically exit 

Sir i changed the baud rate now so I'm able to see the **CSI data for STA mode in csv file** .Before the baud rate was 921600 now i made that 115200  i don't think it is because of baud rate.But whatever the data i got in csv file that is different from the data printed on console . I think because of the speed I'm loosing some data but apart from those data at least one  frame should match right.??

And I"m running it for 30sec to 1 min in that time I"m getting more than 50+ frames on my console and more than 250+ frames on csv file

kallu1294, 2021-12-15:
Sir how i can find real and imaginary data from CSI_DATA which i got in csv file..??

kallu1294, 2021-12-17:
Sorry to disturb you sir,, 
Sir i have one more doubt can we get the two sets of same data while receiving the csi data..??

StevenMHernandez, 2021-12-21:
To parse the real/imaginary data, see https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/master/python_utils/parse_csi.py

kallu1294, 2021-12-23:
Sir i gone through python_utils file in your project but i don't know how to run parse_csi.py file 
please help me sir 

And if try to run **idf.py monitor | python ./python_utils/serial_append_time.py > my-experiment-file.csv**  this is showing no cmakelist.txt found as i shared in screenshot
 
[image]

Now I'm only left with plotting the amplitude and phase  please help me out sir

kallu1294, 2022-02-07:
@StevenMHernandez  Thank you so much sir for your source by which i can able to complete my project up to some extent :-) Now just trying to plot it for real time but not getting that could you please help for this..??

kallu1294, 2022-02-08:
Okay thank you so much sir :-)

closed 2022-02-08 by kallu1294