StevenMHernandez/ESP32-CSI-Tool #46: xtensa Error, Ninja failed with exit code 1

jayantrathi, 2021-12-10:
**Describe the bug**
A clear and concise description of what the bug is.

**Tool Settings**
- Baud Rate: [e.g. 1552000]  (using a lower baud rate *WILL* cause issues. **this is the most common issue**)
- WiFi Channel per Device: [e.g. 6]
- FreeRTOS > Tick rate (Hz): 1000

**To Reproduce**
Steps to reproduce the behavior:
installed ESP-IDF v4.0.4
1. Build project(s) `...` passive
2. Flash project(s) `...` passive
3. External device setup (if applicable)
3. See error

**Desktop (please complete the following information):**
 - OS: Windows 10
 - ESP-IDF Version: v4.0.4
 - ESP32-CSI-Tool Git Hash: [e.g. 31f9f1be9f803e7770e568cd9eb951b4887bbb68]

**Additional context**
Add any other context about the problem here.

C:\Users\jayan\Desktop\esp\esp32-csi-tool\passive>idf.py flash monitor
Checking Python dependencies...
Python requirements from C:\Users\jayan\Desktop\esp-idf\requirements.txt are satisfied.
Adding flash's dependency "all" to list of actions
Executing action: all (aliases: build)
Running ninja in directory c:\users\jayan\desktop\esp\esp32-csi-tool\passive\build
Executing "ninja all"...
[0/1] Re-running CMake...
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- ccache will be used for faster recompilation
-- Project version: 8193b75
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/Users/jayan/Desktop/esp/esp32-csi-tool/passive/sdkconfig
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- Adding linker script C:/Users/jayan/Desktop/esp/esp32-csi-tool/passive/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/jayan/Desktop/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/jayan/Desktop/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/Users/jayan/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/jayan/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/jayan/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/jayan/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/jayan/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_ringbuf esp_rom esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/jayan/Desktop/esp-idf/components/app_trace C:/Users/jayan/Desktop/esp-idf/components/app_update C:/Users/jayan/Desktop/esp-idf/components/asio C:/Users/jayan/Desktop/esp-idf/components/bootloader C:/Users/jayan/Desktop/esp-idf/components/bootloader_support C:/Users/jayan/Desktop/esp-idf/components/bt C:/Users/jayan/Desktop/esp-idf/components/coap C:/Users/jayan/Desktop/esp-idf/components/console C:/Users/jayan/Desktop/esp-idf/components/cxx C:/Users/jayan/Desktop/esp-idf/components/driver C:/Users/jayan/Desktop/esp-idf/components/efuse C:/Users/jayan/Desktop/esp-idf/components/esp-tls C:/Users/jayan/Desktop/esp-idf/components/esp32 C:/Users/jayan/Desktop/esp-idf/components/esp_adc_cal C:/Users/jayan/Desktop/esp-idf/components/esp_common C:/Users/jayan/Desktop/esp-idf/components/esp_eth C:/Users/jayan/Desktop/esp-idf/components/esp_event C:/Users/jayan/Desktop/esp-idf/components/esp_gdbstub C:/Users/jayan/Desktop/esp-idf/components/esp_http_client C:/Users/jayan/Desktop/esp-idf/components/esp_http_server C:/Users/jayan/Desktop/esp-idf/components/esp_https_ota C:/Users/jayan/Desktop/esp-idf/components/esp_https_server C:/Users/jayan/Desktop/esp-idf/components/esp_local_ctrl C:/Users/jayan/Desktop/esp-idf/components/esp_ringbuf C:/Users/jayan/Desktop/esp-idf/components/esp_rom C:/Users/jayan/Desktop/esp-idf/components/esp_websocket_client C:/Users/jayan/Desktop/esp-idf/components/esp_wifi C:/Users/jayan/Desktop/esp-idf/components/espcoredump C:/Users/jayan/Desktop/esp-idf/components/esptool_py C:/Users/jayan/Desktop/esp-idf/components/expat C:/Users/jayan/Desktop/esp-idf/components/fatfs C:/Users/jayan/Desktop/esp-idf/components/freemodbus C:/Users/jayan/Desktop/esp-idf/components/freertos C:/Users/jayan/Desktop/esp-idf/components/heap C:/Users/jayan/Desktop/esp-idf/components/idf_test C:/Users/jayan/Desktop/esp-idf/components/jsmn C:/Users/jayan/Desktop/esp-idf/components/json C:/Users/jayan/Desktop/esp-idf/components/libsodium C:/Users/jayan/Desktop/esp-idf/components/log C:/Users/jayan/Desktop/esp-idf/components/lwip C:/Users/jayan/Desktop/esp/esp32-csi-tool/passive/main C:/Users/jayan/Desktop/esp-idf/components/mbedtls C:/Users/jayan/Desktop/esp-idf/components/mdns C:/Users/jayan/Desktop/esp-idf/components/mqtt C:/Users/jayan/Desktop/esp-idf/components/newlib C:/Users/jayan/Desktop/esp-idf/components/nghttp C:/Users/jayan/Desktop/esp-idf/components/nvs_flash C:/Users/jayan/Desktop/esp-idf/components/openssl C:/Users/jayan/Desktop/esp-idf/components/partition_table C:/Users/jayan/Desktop/esp-idf/components/protobuf-c C:/Users/jayan/Desktop/esp-idf/components/protocomm C:/Users/jayan/Desktop/esp-idf/components/pthread C:/Users/jayan/Desktop/esp-idf/components/sdmmc C:/Users/jayan/Desktop/esp-idf/components/soc C:/Users/jayan/Desktop/esp-idf/components/spi_flash C:/Users/jayan/Desktop/esp-idf/components/spiffs C:/Users/jayan/Desktop/esp-idf/components/tcp_transport C:/Users/jayan/Desktop/esp-idf/components/tcpip_adapter C:/Users/jayan/Desktop/esp-idf/components/ulp C:/Users/jayan/Desktop/esp-idf/components/unity C:/Users/jayan/Desktop/esp-idf/components/vfs C:/Users/jayan/Desktop/esp-idf/components/wear_levelling C:/Users/jayan/Desktop/esp-idf/components/wifi_provisioning C:/Users/jayan/Desktop/esp-idf/components/wpa_supplicant C:/Users/jayan/Desktop/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/jayan/Desktop/esp/esp32-csi-tool/passive/build
[1/7] cmd.exe /C "cd /D C:\Users\jayan\Desktop\esp\esp32-csi-tool\pa...*******************************************************************"Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[3/7] Performing build step for 'bootloader'
[0/1] Re-running CMake...
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- Project version: v4.0.4
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/Users/jayan/Desktop/esp/esp32-csi-tool/passive/sdkconfig
-- Adding linker script C:/Users/jayan/Desktop/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/Users/jayan/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/jayan/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Users/jayan/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/jayan/Desktop/esp-idf/components/bootloader/subproject/main/esp32.bootloader.ld
[... 115 lines cut ...]
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/no-rtti/libstdc++-v3/include/bits/char_traits.h:464: undefined reference to `wmemcpy'
c:/users/jayan/desktop/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/jayan/desktop/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(cow-wstring-inst.o): in function `std::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::_M_move(wchar_t*, wchar_t const*, unsigned int)':
[... cut ...] .0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(collate_members_cow.o): in function `std::collate<wchar_t>::_M_transform(wchar_t*, wchar_t const*, unsigned int) const':
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/no-rtti/libstdc++-v3/src/c++98/collate_members_cow.cc:69: undefined reference to `wcsxfrm'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

StevenMHernandez, 2021-12-10:
I think this is the same issue as #32 or #41

Please read through those threads and see if you can find a solution.

It sounds like there is an issue with the current version of the esp-idf v4.0. Unfortunately, I do not have time to track down the issue with that external project.

One solution is to use a esp-idf v4.3 using this pull request #39.

StevenMHernandez, 2022-01-07:
This project is now using esp-idf v4.3 which should have solved this issue. If the problem still persists, please reopen the issue and let me know. Thanks.

closed 2022-01-07 by StevenMHernandez