StevenMHernandez/ESP32-CSI-Tool #48: Customizing the CSI inter-sample interval

elmehdiilli, 2022-01-05:
Hello all,

I could successfully flash the project into my ESP32 devices. I was able to measure and save CSI (channel frequency response) measurements in csv files. Nevertheless, I am still curious to know if we can customize the time interval between each two successive CSI outputs/measurements. Let's say I want it to be 1 second separating each two successive measurements.

Used Baud Rate = 115200

ESP-IDF Version : 4.3

OS: Win10 and Linux (works well in both of them)

Thank you,

StevenMHernandez, 2022-01-05:
When you run `idf.py menuconfig` you should see `ESP32 CSI Tool Config > Packet TX Rate`. You can set the rate to `1` to send CSI frames every one second.

This is only works for `active_sta` because the STA sends CSI frames while the `active_ap` only listens for CSI frames.

Please let me know if this answers your question.

elmehdiilli, 2022-01-07:
@StevenMHernandez 

Hi Steven,

Thank you so much for the hint. It worked perfectly :) . Moreover, changing it in the station side impacts likewise the rate at the AP as well. 

I have another question, though not related to the issue: Is it possible to collect CSI simultaneously from two nodes? Or even having two measurements interleaved by some time interval ? I've been thinking to an AP to which two STA are connected, and such an AP can collect the respective CSIs. Could you please advise in this ?

Thank you once again,

Best,

StevenMHernandez, 2022-01-07:
This is a good question. I cannot guarantee that using two STAs at the same time will work perfectly with the current codebase. However, give it a try and let me know what you find. I cannot work on this immediately, but if you want to look at updating the codebase, it is a very small amount of code and I encourage you to help contribute if you are able to. 

You might want to create a new feature request (a new github issue), then you can close this issue since your new question is unrelated.

closed 2022-01-07 by StevenMHernandez