StevenMHernandez/ESP32-CSI-Tool #50: Monitor garbled

YXW-123, 2022-01-22:
**Describe the bug**
A clear and concise description of what the bug is.

**Tool Settings**
- Baud Rate: [1152000] 
- WiFi Channel per Device: [6]
- FreeRTOS > Tick rate (Hz): [1000]

**To Reproduce**
Steps to reproduce the behavior:
1. Build project(s) `active_ap` 
2. Flash and monitor project(s) active_ap
3. See error like this 
[image]

**Desktop (please complete the following information):**
 - OS: [ Ubuntu20.04]
 - ESP-IDF Version: [4.3]
 - ESP32-CSI-Tool Git Hash: [e.g. 31f9f1be9f803e7770e568cd9eb951b4887bbb68]

**Remark**
There is no error when starting with get-started/hello_world project from examples directory in IDF.

StevenMHernandez, 2022-01-22:
Thank you for checking to see if `get-started/hello_world` works.

Please make sure to follow all of the 7 steps in https://github.com/StevenMHernandez/ESP32-CSI-Tool#configuration-esp-idf. The new version of ESP-IDF v4.3 requires a few extra steps that you might not have done previously when you were using older versions of the ESP-IDF.

Can you confirm that you set all of the 'UART' and 'baud rate' related configuration values correctly?

YXW-123, 2022-01-23:
Sorry I forgot to write clearly to complete the configuration step, I can make sure set all of the 'UART' and 'baud rate' related configuration values correctly. I try to flash and monitor again, but there is the same error.

YXW-123, 2022-01-23:
Thanks for your help! After looking up some information，I find that garbled phenomenon is related to 'baud rate' .Now I use a much lower value(115200) ,this tool can  run correctly!

StevenMHernandez, 2022-01-24:
I am glad this worked for you. I will close the issue.

closed 2022-01-24 by StevenMHernandez