StevenMHernandez/ESP32-CSI-Tool #51: esp_wifi_set_csi(1)    Error

alihayek179, 2022-01-30:
**Describe the bug**
An error occurs in csi_component.h" line 96

> ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40088434
> file: "../main/../../_components/csi_component.h" line 96
> func: void csi_init(char*)
> expression: esp_wifi_set_csi(1)

I replaced "ESP_ERROR_CHECK(esp_wifi_set_csi(1))" with "esp_wifi_set_csi(1)" and it worked fine. 

I tied to track  "esp_wifi_set_csi(1)" to see what is the problem but I could not find the source.

**Tool Settings**
- Baud Rate: 152000
- WiFi Channel per Device: 6
- FreeRTOS > Tick rate (Hz):  not sure

**To Reproduce**
Steps to reproduce the behavior:
Build project(s) :  ACTIVE_AP
Flash project(s) :  ACTIVE_AP

**Desktop (please complete the following information):**
OS: Windows10
ESP-IDF Version: 4.4

Thanks in advance.

tartunian, 2022-01-31:
I have also encountered this same error.

- Baud rate: 115200
- Project: passive
- ESP-IDF: 4.3.2
- OS: Win 10

alihayek179, 2022-02-12:
I solved this problem by selecting o menuconfig > Conponent config > WI-FI  > WiFi CSI (Chanel State Information).

[image]

closed 2022-02-12 by alihayek179

PaulTbbr, 2023-02-15:
Hello, for a student engineering project I need to access raw CSI data, could you tell me more about the hardware setup you used? Do you have any ressources for how to use esp32 in that case?