StevenMHernandez/ESP32-CSI-Tool #52: write CSI to csv file

flyjasonborn, 2022-02-09:
The STA mode ESP32 can collect csi_data, but unfortunatelly cannot write csidata to csv file using cmd "findstr..." or "grep...".Looking forward to your apply!

StevenMHernandez, 2022-02-10:
I need more information. What is there error that you are seeing?

What operating system are you using? Windows uses `findstr` Linux/MacOS uses `grep`.

Run the following and report the results. :
1. `idf.py monitor -p /dev/ttyUSB0` <- What is printed by STA mode?
2. `idf.py monitor -p /dev/ttyUSB0 | grep "CSI_DATA"` <- What is printed with this?
3. `idf.py monitor -p /dev/ttyUSB0 | findstr "CSI_DATA"` <- What is printed with this?

flyjasonborn, 2022-02-10:
1. `idf.py monitor`
[image]

2. `idf.py monitor | grep "CSI_DATA"`
[image]

3. `idf.py monitor | findstr "CSI_DATA"`
[image]

Thank you, Steven

StevenMHernandez, 2022-02-10:
I am sorry, I have no knowledge about Powershell. From what I can gather, powershell does not use pipes as would be expected in something like [bash](https://linuxhint.com/bash_pipe_tutorial/) on macOS or Linux.

I do not have a windows computer to solve this. If you can figure out a solution, please share it here so that others can benefit from your solution. Thank you.
Also, #18 deals with windows compatibility.

StevenMHernandez, 2022-10-17:
Closing issue in favor of keeping #18 open for any issues running on Windows machines