StevenMHernandez/ESP32-CSI-Tool #53: ninja failed with exit code 1

zhy102, 2022-03-03:
**To Reproduce**
Steps to reproduce the behavior:
1. Build project(s) `...` (active_ap, active_sta, passive)
2. Flash project(s) `...` (active_ap, active_sta, passive)
3. External device setup (Eps32 wroom 32u)
3. See error when i run "idf.py flash monitor”：Failed to create temporary file for esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_aead/chacha20poly1305/sodium/aead_chacha20poly1305.c.obj: No such file or directory
[800/918] Building C object esp-idf/json/CMakeFiles/__idf_json.dir/cJSON/cJSON.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

help me，please.

StevenMHernandez, 2022-03-03:
You forgot to add information about your operating system and which version of esp-idf you are using. 

libsodium is not used directly in this project, so I think this is a problem with your esp-idf install. Are you able to flash example projects from esp-idf?

Have you searched for any solutions so far? For example, does this help (https://github.com/espressif/esp-idf/issues/7195)

zhy102, 2022-03-04:
> You forgot to add information about your operating system and which version of esp-idf you are using.
> 
> libsodium is not used directly in this project, so I think this is a problem with your esp-idf install. Are you able to flash example projects from esp-idf?
> 
> Have you searched for any solutions so far? For example, does this help ([espressif/esp-idf#7195](https://github.com/espressif/esp-idf/issues/7195))

The operating system is windows 10 with version of esp-idf v4.3.
The first time I burned and ran all operations were fine. Now it reports the errors at the last ran "idf.py -p COM12 flash monitor" command. Should I delete the original file when I compile it again and burn it again?
ccache: error: Failed to create temporary file for esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_shorthash/siphash24/ref/shorthash_siphash24_ref.c.obj: No such file or directory
[812/970] Building C object esp-idf/freemodbus/CMakeFiles/__idf_freemodbus.dir/common/esp_modbus_master_tcp.c.obj
ninja: build stopped: subThe operating system is windows 10 with version of esp-idf v4.3.
The first time I burned and ran all operations were fine. Now it reports the errors at the last ran "idf.py -p COM12 flash monitor" command. Should I delete the original file when I compile it again and burn it again?
ccache: error: Failed to create temporary file for esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_shorthash/siphash24/ref/shorthash_siphash24_ref.c.obj: No such file or directory
[812/970] Building C object esp-idf/freemodbus/CMakeFiles/__idf_freemodbus.dir/common/esp_modbus_master_tcp.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1command failed.
ninja failed with exit code 1

zhy102, 2022-03-04:
> You forgot to add information about your operating system and which version of esp-idf you are using.
>
> libsodium is not used directly in this project, so I think this is a problem with your esp-idf [...]
**Here is the detailed error message as follows：**
C:\Users\Administrator\.espressif\frameworks\esp-idf-v4.3.2\ESP32-CSI-Tool-master\active_ap>idf.py -p COM12 flash monitor
Executing action: flash
Running ninja in directory c:\users\administrator\.espressif\frameworks\esp-idf-v4.3.2\esp32-csi-tool-master\active_ap\build
Executing "ninja flash"...
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/ESP32-CSI-Tool-master/active_ap/sdkconfig
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- App "active-ap" version: v4.3.2
-- Adding linker script C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/ESP32-CSI-Tool-master/active_ap/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_local_ctrl esp_netif esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/app_trace C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/app_update C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/asio C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/bootloader C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/bootloader_support C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/bt C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/cbor C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/cmock C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/coap C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/console C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/cxx C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/driver C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/efuse C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp-tls C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp32 C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_adc_cal C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_common C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_eth C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_event C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_gdbstub C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_hid C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_http_client C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_http_server C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_https_ota C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_https_server C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_hw_support C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_ipc C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_local_ctrl C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_netif C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_pm C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/esp_ringbuf C:/Users [... cut ...] ministrator/.espressif/frameworks/esp-idf-v4.3.2/components/soc C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/spi_flash C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/Administrator/.espressif/frameworks/esp-idf-v4.3.2/ESP32-CSI-Tool-master/active_ap/build/bootloader
ninja: no work to do.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

StevenMHernandez, 2022-03-04:
"The first time I burned and ran all operations were fine" The project worked before? Did you make any changes at all?

"Should I delete the original file when I compile it again and burn it again?" Sure, go ahead and try this. You do not need to ask permission to try to debug this issue by yourself! :) Let me know what happens.

zhy102, 2022-03-04:
> Yes, I deleted the original file when I compiled it again and burn it again, however, it is still.
So I want to rebuild my development environment now, can I use the updated version to extract CSI，for example? For example，esp-idf-v4.4？

zhy102, 2022-03-04:
> > You forgot to add information about your operating system and which version of esp-idf you are using.
> > libsodium is not used directly in this project, so I think this is a problem with your esp-idf install. Are you able to flash example projects from esp-idf?
> > Have you searched for any solutions so far? For example, does this help ([espressif/esp-idf#7195](https://github.com/espressif/esp-idf/issues/7195))
> 
> The operating system is windows 10 with version of esp-idf v4.3. The first time I burned and ran all operations were fine. Now it reports the errors at the last ran "idf.py -p COM12 flash monitor" command. Should I delete the original file when I compile it again and burn it again? ccache: error: Failed to create temporary file for esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_shorthash/siphash24/ref/shorthash_siphash24_ref.c.obj: No such file or directory [812/970] Building C object esp-idf/freemodbus/CMakeFiles/__idf_freemodbus.dir/common/esp_modbus_master_tcp.c.obj ninja: build stopped: subcommand failed. ninja failed with exit code 1command failed. ninja failed with exit code 1

Are the version of esp-idf-v4.3.2 and esp-idf-release-v4.3 equally suitable to be able to use this ESP32-CSI-Tool？

StevenMHernandez, 2022-03-05:
Both v4.3 should work the same I believe.

Are you able to run any example from the esp-idf project? You need to get that other project working first before you can get my project working.

zhy102, 2022-03-05:
Yes, it is ok when i ran firstly the any example project, even the ESP32-csi-tool. However, the errors are coming when i run this same project. So why？

zhy102, 2022-03-05:
Yes, it is ok when i ran firstly the any example project, even the ESP32-csi-tool. However, the errors are coming when i run this same project again. So why？

zhy102, 2022-03-05:
> Both v4.3 should work the same I believe.
> 
> Are you able to run any example from the esp-idf project? You need to get that other project working first before you can get my project working.

I also found that I had no problem executing the example from the esp-idf project repeatedly. But when I ran the tool, only the first time was correct, after that it all prompted errors as described above. When I reinstalled the whole environment, it still reported no error the first time I ran it, and after that, it reported an error, and the command could not generate a csv file, I was very distressed.

StevenMHernandez, 2022-03-05:
Please do not be distressed. Programming issues, problems with tools and things not working as you expect are all "par for the course" when it comes to "cutting edge research". Your job is to solve these problems. _Embrace the unknown_!

That being said, this problem is very strange. You say it works the first time you run the project. **What command do you run the first time where it works for this project? Also, what is the output of the command? Do you see any CSI data?** After you run that command, you say it fails. **What command do you run when it fails?**

Ignore the CSV file part, it earlier parts of the pipeline are not working, then the CSV file will not work either. You need to break down the problem into the earliest problem first.

zhy102, 2022-03-09:
I recently tried running it under the “”hello_world“” file and he can flash and print normally, but when flashing under the "active_ap" file it reports the following error：
C:\Espressif\frameworks\esp-idf-v4.3.2\ESP32-CSI-Tool-master\active_ap>idf.py flash
```
Executing action: flash
Serial port COM12
Connecting.....
Detecting chip type... ESP32
Running ninja in directory c:\espressif\frameworks\esp-idf-v4.3.2\esp32-csi-tool-master\active_ap\build
Executing "ninja flash"...
[5/114] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodi.../crypto_box/curve25519xchacha20poly1305/box_seal_curve25519xchacha20poly1305.c.obj
FAILED: esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_seal_curve25519xchacha20poly1305.c.obj
ccache C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DCONFIGURED -DHAVE_WEAK_SYMBOLS -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DNATIVE_LITTLE_ENDIAN -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -Iconfig -I../../../components/libsodium/libsodium/src/libsodium/include -I../../../components/libsodium/port_include -I../../../components/libsodium/libsodium/src/libsodium/include/sodium -I../../../components/libsodium/port_include/sodium -I../../../components/libsodium/port -I../../../components/newlib/platform_include -I../../../components/freertos/include -I../../../components/freertos/port/xtensa/include -I../../../components/esp_hw_support/include -I../../../components/esp_hw_support/port/esp32/. -I../../../components/heap/include -I../../../components/log/include -I../../../components/lwip/include/apps -I../../../components/lwip/include/apps/sntp -I../../../components/lwip/lwip/src/include -I../../../components/lwip/port/esp32/include -I../../../components/lwip/port/esp32/include/arch -I../../../components/soc/include -I../../../components/soc/esp32/. -I../../../components/soc/esp32/include -I../../../components/hal/esp32/include -I../../../components/hal/include -I../../../components/esp_rom/include -I../../../components/esp_rom/esp32 -I../../../components/esp_rom/include/esp32 -I../../../components/esp_common/include -I../../../components/esp_system/include -I../../../components/esp32/include -I../../../components/driver/include -I../../../components/driver/esp32/include -I../../../components/esp_ringbuf/include -I../../../components/efuse/include -I../../../components/efuse/esp32/include -I../../../components/xtensa/include -I../../../components/xtensa/esp32/include -I../../../components/espcoredump/include -I../../../components/esp_timer/include -I../../../components/esp_ipc/include -I../../../components/esp_pm/include -I../../../components/vfs/include -I../../../components/esp_wifi/include -I../../../components/esp_wifi/esp32/include -I../../../components/esp_event/include -I../../../components/esp_netif/include -I../../../components/esp_eth/include -I../../../components/tcpip_adapter/include -I../../../components/app_trace/include -I../../../components/mbedtls/port/include -I../../../components/mbedtls/mbedtls/include -I../../../components/mbedtls/esp_crt_bundle/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.3.2\" -DESP_PLATFORM -Wno-unused-function -MD -MT esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_seal_curve25519xchacha20poly1305.c.obj -MF esp-idf\libsodium\CMakeFiles\__idf_libsodium.dir\libsodium\src\libsodium\crypto_box\curve25519xchacha20poly1305\box_seal_curve25519xchacha20poly1305.c.obj.d -o esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_seal_curve25519xchacha20poly1305.c.obj   -c C:/Espressif/frameworks/esp-idf-v4.3.2/components/libsodium/libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_seal_curve25519xchacha20poly1305.c
ccache: error: Failed to create temporary file for esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_seal_curve25519xchacha20poly1305.c.obj: No such file or directory
[7/114] Performing build step for 'bootloader'
ninja: no work to do.
[10/112] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_kx/crypto_kx.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\Espressif\frameworks\esp-idf-v4.3.2\ESP32-CSI-Tool-master\active_ap>idf.py flash
Executing action: flash
Serial port COM12
Connecting....
Detecting chip type... ESP32
Running ninja in directory c:\espressif\frameworks\esp-idf-v4.3.2\esp32-csi-tool-master\active_ap\build
Executing "ninja flash"...
[2/106] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodi.../crypto_box/curve25519xchacha20poly1305/box_seal_curve25519xchacha20poly1305.c.obj
FAILED: esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_seal_curve25519xchacha20poly1305.c.obj
[... cut ...]
ccache: error: Failed to create temporary file for esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_seal_curve25519xchacha20poly1305.c.obj: No such file or directory
[4/106] Performing build step for 'bootloader'
ninja: no work to do.
[7/104] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_pwhash/argon2/argon2-core.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

StevenMHernandez, 2022-03-09:
Thank you for trying the `hello_world` example again. This confirms that ESP-IDF was installed. 

Now can you try the `scan` example (https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/examples/wifi/scan/main/scan.c). This will confirm if ESP-IDF WiFi components are working.

Thank you for formatting the error message. It makes it easier to read. When I search for your error, I found this link (https://esp32.com/viewtopic.php?t=14651). Specifically, they say "Windows is limited to 260 character paths". Have you tried their methods?

zhy102, 2022-03-10:
I appreciate even more your answer, which is crucial for me.
I tried their methods about  "Windows is limited to 260 character paths".
And i tried the ‘’‘’scan ‘’’ example，it is ok , as follows:

C:\Espressif\frameworks\esp-idf-v4.3.2\examples\wifi\scan>idf.py menuconfig
'''
Executing action: menuconfig
Running ninja in directory c:\espressif\frameworks\esp-idf-v4.3.2\examples\wifi\scan\build
Executing "ninja menuconfig"...
[0/1] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v4...if/frameworks/esp-idf-v4.3.2/examples/wifi/scan/sdkconfig"
C:/Espressif/frameworks/esp-idf-v4.3.2/Kconfig:14: warning: IDF_ENV_FPGA has 'option env="IDF_ENV_FPGA"', but the environment variable IDF_ENV_FPGA is not set
Loaded configuration 'C:/Espressif/frameworks/esp-idf-v4.3.2/examples/wifi/scan/sdkconfig'
No changes to save (for 'C:/Espressif/frameworks/esp-idf-v4.3.2/examples/wifi/scan/sdkconfig')
'''
C:\Espressif\frameworks\esp-idf-v4.3.2\examples\wifi\scan>idf.py flash
'''
Executing action: flash
Serial port COM13
Connecting....
Detecting chip type... ESP32
Running ninja in directory c:\espressif\frameworks\esp-idf-v4.3.2\examples\wifi\scan\build
Executing "ninja flash"...
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/Espressif/frameworks/esp-idf-v4.3.2/examples/wifi/scan/sdkconfig
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- App "scan" version: v4.3.2
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/examples/wifi/scan/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_local_ctrl esp_netif esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Espressif/frameworks/esp-idf-v4.3.2/components/app_trace C:/Espressif/frameworks/esp-idf-v4.3.2/components/app_update C:/Espressif/frameworks/esp-idf-v4.3.2/components/asio C:/Espressif/frameworks/esp-idf-v4.3.2/components/bootloader C:/Espressif/frameworks/esp-idf-v4.3.2/components/bootloader_support C:/Espressif/frameworks/esp-idf-v4.3.2/components/bt C:/Espressif/frameworks/esp-idf-v4.3.2/components/cbor C:/Espressif/frameworks/esp-idf-v4.3.2/components/cmock C:/Espressif/frameworks/esp-idf-v4.3.2/components/coap C:/Espressif/frameworks/esp-idf-v4.3.2/components/console C:/Espressif/frameworks/esp-idf-v4.3.2/components/cxx C:/Espressif/frameworks/esp-idf-v4.3.2/components/driver C:/Espressif/frameworks/esp-idf-v4.3.2/components/efuse C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp-tls C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp32 C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_adc_cal C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_common C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_eth C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_event C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_gdbstub C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_hid C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_http_client C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_http_server C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_https_ota C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_https_server C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_hw_support C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_ipc C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_local_ctrl C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_netif C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_pm C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_ringbuf C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_rom C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_serial_slave_link C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_system C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_timer C:/Espressif/frameworks/esp-idf-v4.3.2/components/es [... cut ...]
Writing at 0x0007f394... (76 %)
Writing at 0x00084f03... (80 %)
Writing at 0x0008b80b... (84 %)
Writing at 0x0009548c... (88 %)
Writing at 0x0009c1d4... (92 %)
Writing at 0x000a1c24... (96 %)
Writing at 0x000a6fa9... (100 %)
Wrote 622704 bytes (395751 compressed) at 0x00010000 in 9.1 seconds (effective 547.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
'''

**However, '''ninja: build stopped: subcommand failed. ninja failed with exit code 1'''In the issue above  is still.**

StevenMHernandez, 2022-03-10:
Both `scan` and `active_ap` seem to use libsodium, so it does not make sense that it works with `scan` and not with `active_ap`.

Can you try to run `idf.py fullclean` in your `active_ap` and then run `idf.py build` again?

StevenMHernandez, 2022-10-17:
Closing old issues.

closed 2022-10-17 by StevenMHernandez