StevenMHernandez/ESP32-CSI-Tool #54: Configuring incomplete, errors occurred!

zhy102, 2022-03-06:
The bug is：

**To Reproduce**
Steps to reproduce the behavior:
1. Build project(s) `...` (active_ap, active_sta, passive)
2. Flash project(s) `...` (active_ap, active_sta, passive)
3. External device setup (eps32-wroom-32u)
3. See error

**Desktop (please complete the following information):**
 - OS: [Ubuntu]
 - ESP-IDF Version: [v4.3]
 - ESP32-CSI-Tool Git Hash: [e.g. 31f9f1be9f803e7770e568cd9eb951b4887bbb68]

zhy123@zhy123-Lenovo-XiaoXin-700-15ISK:~/esp/esp-idf/hello_world$ **idf.py build**
The bug is：
Executing action: all (aliases: build)
Running cmake in directory /home/zhy123/esp/esp-idf/hello_world/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/zhy123/esp/esp-idf/hello_world"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Initialising new submodule examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib...
子模组 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib'（https://github.com/espressif/esp-cryptoauthlib.git）已对路径 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib' 注册
正克隆到 '/home/zhy123/esp/esp-idf/examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib'...
子模组路径 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib'：检出 'bb672b0437485fc7420add178299631692b15ac3'
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/zhy123/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/zhy123/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/zhy123/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/zhy123/esp/esp-idf/hello_world/sdkconfig
CMake Error at /home/zhy123/esp/esp-idf/tools/cmake/component.cmake:306 (message):
  Include directory
  '/home/zhy123/esp/esp-idf/components/mbedtls/mbedtls/include' is not a
  directory.
Call Stack (most recent call first):
  /home/zhy123/esp/esp-idf/tools/cmake/component.cmake:478 (__component_add_include_dirs)
  /home/zhy123/esp/esp-idf/components/mbedtls/CMakeLists.txt:4 (idf_component_register)

**-- Configuring incomplete, errors occurred!
See also "/home/zhy123/esp/esp-idf/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1**

And my version of cmake is v3.23.

StevenMHernandez, 2022-03-06:
Your error is:

```
Include directory
'/home/zhy123/esp/esp-idf/components/mbedtls/mbedtls/include' is not a
directory.
```

I think you have not installed esp-idf correctly. I am guessing git submodules are not installed. You should have called `git clone -b release/v4.3 --recursive https://github.com/espressif/esp-idf.git` when you clones esp-idf. Alternatively check this comment (https://stackoverflow.com/a/1032653)

zhy102, 2022-03-07:
> 
Ok, i deleted all files and called ''git clone -b release/v4.3 --recursive https://github.com/espressif/esp-idf.git'' after ran "git submodule update --remote". **Errors:**
zhy123@zhy123-Lenovo-XiaoXin-700-15ISK:~/esp$ git clone -b release/v4.3 --recursive https://github.com/espressif/esp-idf.git
正克隆到 'esp-idf'...
fatal: 无法访问 'https://github.com/espressif/esp-idf.git/'：GnuTLS recv error (-110): The TLS connection was non-properly terminated.
So i called ''git clone -b release/v4.3  https://github.com/espressif/esp-idf.git'' instead of the ''git clone -b release/v4.3 --recursive https://github.com/espressif/esp-idf.git''. Is it okay?
**However, the errors as follows** when i run "idf.py menuconfig":
Executing action: menuconfig
Running cmake in directory /home/zhy123/esp/ESP32-CSI-Tool-master/active_ap/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/zhy123/esp/ESP32-CSI-Tool-master/active_ap"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Initialising new submodule components/asio/asio...
子模组 'components/asio/asio'（https://github.com/espressif/asio.git）已对路径 'components/asio/asio' 注册
正克隆到 '/home/zhy123/esp/esp-idf/components/asio/asio'...
error: RPC 失败。curl 56 GnuTLS recv error (-110): The TLS connection was non-properly terminated.
fatal: 远端意外挂断了
fatal: 协议错误：坏的包头
fatal: 无法克隆 'https://github.com/espressif/asio.git' 到子模组路径 '/home/zhy123/esp/esp-idf/components/asio/asio'
克隆 'components/asio/asio' 失败。按计划重试
正克隆到 '/home/zhy123/esp/esp-idf/components/asio/asio'...
子模组路径 'components/asio/asio'：检出 'f31694c9f1746ba189a4bcae2e34db15135ddb22'
子模组路径 'components/asio/asio'：检出 'f31694c9f1746ba189a4bcae2e34db15135ddb22'
-- Initialising new submodule components/bootloader/subproject/components/micro-ecc/micro-ecc...
子模组 'components/bootloader/subproject/components/micro-ecc/micro-ecc'（https://github.com/kmackay/micro-ecc.git）已对路径 'components/bootloader/subproject/components/micro-ecc/micro-ecc' 注册
正克隆到 '/home/zhy123/esp/esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc'...
子模组路径 'components/bootloader/subproject/components/micro-ecc/micro-ecc'：检出 'd037ec89546fad14b5c4d5456c2e23a71e554966'
-- Initialising new submodule components/bt/controller/lib_esp32...
子模组 'components/bt/controller/lib_esp32'（https://github.com/espressif/esp32-bt-lib.git）已对路径 'components/bt/controller/lib_esp32' 注册
正克隆到 '/home/zhy123/esp/esp-idf/components/bt/controller/lib_esp32'...
fatal: 无法访问 'https://github.com/espressif/esp32-bt-lib.git/'：GnuTLS recv error (-110): The TLS connection was non-properly terminated.
fatal: 无法克隆 'https://github.com/espressif/esp32-bt-lib.git' 到子模组路径 '/home/zhy123/esp/esp-idf/components/bt/controller/lib_esp32'
克隆 'components/bt/controller/lib_esp32' 失败。按计划重试
正克隆到 '/home/zhy123/esp/esp-idf/components/bt/controller/lib_esp32'...
fatal: 无法访问 'https://github.com/espressif/esp32-bt-lib.git/'：GnuTLS recv error (-110): The TLS connection was non-properly terminated.
fatal: 无法克隆 'https://github.com/espressif/esp32-bt-lib.git' 到子模组路径 '/home/zhy123/esp/esp-idf/components/bt/controller/lib_esp32'
第二次尝试克隆 'components/bt/controller/lib_esp32' 失败，退出
CMake Error at /home/zhy123/esp/esp-idf/tools/cmake/git_submodules.cmake:48 (message):
  Git submodule init failed for components/bt/controller/lib_esp32
Call Stack (most recent call first):
  /home/zhy123/esp/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/zhy123/esp/esp-idf/tools/cmake/build.cmake:166 (__build_get_idf_git_revision)
  /home/zhy123/esp/esp-idf/tools/cmake/idf.cmake:50 (__build_init)
  /home/zhy123/esp/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1

StevenMHernandez, 2022-03-07:
Please format you messages better so that I can distinguish your sentences and the error. (Use new lines and add code formatting)

How is it low speed?

StevenMHernandez, 2022-03-07:
This is not a problem with this project, this is a problem with your esp-idf installation. Right? You might want to make sure esp-idf is installed correctly first. Then come back to this project. 

If you have general questions about esp-idf you may have better luck asking with those maintainers.

StevenMHernandez, 2022-03-10:
This looks like a duplicate of your installation problem in #53. Closing. Please reopen if I am incorrect.

closed 2022-03-10 by StevenMHernandez

ericphilbaz, 2023-07-11:
maybe delete the file "sdkconfig.old",   works for me!