StevenMHernandez/ESP32-CSI-Tool #56: wifi not connected. waiting...

gmcd310, 2022-03-13:
I have some questions for you
Run active_ AP, the AP named myssid cannot be found, the computer and mobile phone cannot be scanned, and then run

active_ STA, as shown below:
I (940) Active CSI collection (Station): connect to ap SSID:myssid password:mypassword
CSI will not be collected. Check `idf.py menuconfig  # > ESP32 CSI Tool Config` to enable CSIwifi not connected. waiting...
wifi not connected. waiting...
wifi not connected. waiting...
I (2990) Active CSI collection (Station): Retry connecting to the AP
wifi not connected. waiting...
wifi not connected. waiting..

 idf：v4.3

closed 2022-03-13 by gmcd310

StevenMHernandez, 2022-03-13:
Were there any steps that you took to solve this issue? It might help someone else in the future if you share your debugging steps and your solution. Thanks.

gmcd310, 2022-03-16:
I wrong with some sitting of menuconfig