StevenMHernandez/ESP32-CSI-Tool #59: When using the function "passtive",unable to get data  from AP or STA

YXW-123, 2022-04-03:
**Describe the bug**
When using the function "passtive",I only get a liitle data from AP or STA, even unable to get data.
**Tool Settings**
- Baud Rate: [115200]
- P.S.Using a higher baud rate will be garbled.
- WiFi Channel per Device: [4]
- FreeRTOS > Tick rate (Hz): [1000]

**To Reproduce**
Steps to reproduce the behavior:
1. Build project(s) active_ap, active_sta, passive
2. Flash project(s) active_ap, active_sta, passive to 3 ESP32s
3. Run the above projects
4. when using channel 13 for above 3 projects,I can sniffer some data from other device,and unable to get data from AP or STA. Then,I changed channel  to 4 avoid interference from other devices in the channel.ap AP and STA can work well and print CSI well,but I only get a little data.

**Desktop (please complete the following information):**
 - OS: [Ubuntu]
 - ESP-IDF Version: [4.3]

**Additional context**
Add any other context about the problem here.

Vdeus, 2022-05-09:
I met the same problem. The ESP-IDF CMD printed the following with no CSI data:

-----------------------
ESP32 CSI Tool Settings
-----------------------
PROJECT_NAME: PASSIVE
CONFIG_ESPTOOLPY_MONITOR_BAUD: 115200
CONFIG_ESP_CONSOLE_UART_BAUDRATE: 115200
IDF_VER: v4.3.2
-----------------------
WIFI_CHANNEL: 6
SHOULD_COLLECT_CSI: 1
SHOULD_COLLECT_ONLY_LLTF: 1
SEND_CSI_TO_SERIAL: 1
SEND_CSI_TO_SD: 0
-----------------------

I (685) wifi:wifi driver task: 3ffc00c0, prio:23, stack:6656, core=0
I (686) system_api: Base MAC address is not set
I (686) system_api: read default base MAC address from EFUSE
I (699) wifi:wifi firmware version: eb52264
I (700) wifi:wifi certification version: v7.0
I (700) wifi:config NVS flash: enabled
I (703) wifi:config nano formating: disabled
I (708) wifi:Init data frame dynamic rx buffer num: 32
I (712) wifi:Init management frame dynamic rx buffer num: 32
I (718) wifi:Init management short buffer num: 32
I (722) wifi:Init dynamic tx buffer num: 32
I (726) wifi:Init static rx buffer size: 2212
I (730) wifi:Init static rx buffer num: 10
I (734) wifi:Init dynamic rx buffer num: 32
I (738) wifi_init: rx ba win: 6
I (741) wifi_init: tcpip mbox: 32
I (745) wifi_init: udp mbox: 6
I (749) wifi_init: tcp mbox: 6
I (753) wifi_init: tcp tx win: 5744
I (757) wifi_init: tcp rx win: 5744
I (761) wifi_init: tcp mss: 1440
I (765) wifi_init: WiFi IRAM OP enabled
I (770) wifi_init: WiFi RX IRAM OP enabled
I (776) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (881) wifi:mode : null
I (882) wifi:ic_enable_sniffer
type,role,mac,rssi,rate,sig_mode,mcs,bandwidth,smoothing,not_sounding,aggregation,stbc,fec_coding,sgi,noise_floor,ampdu_cnt,channel,secondary_channel,local_timestamp,ant,sig_len,rx_state,real_time_set,real_timestamp,len,CSI_DATA

StevenMHernandez, 2022-10-17:
Closing old issues. Feel free to open again with more information.

closed 2022-10-17 by StevenMHernandez