StevenMHernandez/ESP32-CSI-Tool #60: OSError: [Errno 22] Invalid argument

AlexXu12256, 2022-04-15:
The script in python_utils does not work and give error says:
Exception ignored in: <_io.TextIOWrapper name='<stdout>' mode='w' encoding='gbk'>
OSError: [Errno 22] Invalid argument

**Tool Settings**
- Baud Rate: [1552000] 
- WiFi Channel per Device: [6]
- FreeRTOS > Tick rate (Hz): [1000]

**To Reproduce**
Steps to reproduce the behavior:
1. Build project(s) `cd (file location)` (active_ap, active_sta)
2. Flash project(s) `idf.py flash monitor` (active_ap, active_sta)
3. External device setup (one flashed as the station, one flashed as the access point)
3. input: C:\Espressif\CODE\ESP32-CSI-Tool-master\active_sta>idf.py -p /COM4 monitor | python ./python_utils/parse_csi.py > tst.csv
4. Error: Exception ignored in: <_io.TextIOWrapper name='<stdout>' mode='w' encoding='gbk'>
OSError: [Errno 22] Invalid argument

**Desktop (please complete the following information):**
[image]

 - OS: [Windows 10]
 - ESP-IDF Version: [e.g. 4.3.2]
 - ESP32-CSI-Tool Git Hash: [not sure]

StevenMHernandez, 2022-04-18:
Because you are using Windows, you might want to look at #18 

It looks like `parse_csi.py` is throwing an exception. I wonder if we add a `try: catch:` if we can identify what input is causing this issue.

Unfortunately I do not have a Windows machine to test this on. @AlexXu12256, do you mind adding exception handling around this line (https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/master/python_utils/parse_csi.py#L20)

```
try:
    csi_string = re.findall(r"\[(.*)\]", l)[0]
except Exception as e:
    print(e)
    print(l)
```

This should print the exception and it should also print out the `l` variable which causes the errror.

AlexXu12256, 2022-04-23:
Hi there, thank you so much for the response, I just have a quick question about which type of OS are you running this code on?

StevenMHernandez, 2022-04-25:
Personally I use macOS. I believe Linux has worked well for users too (because macOS and Linux commands are similar).

closed 2022-07-14 by AlexXu12256

AlexXu12256, 2022-08-08:
Hi there, all my tool settings stay the same as mentioned above and I am currently trying to set up two receivers (sta) and one transmitter (ap) at the same time, but the issue I am having is that the amount of data received in each sta is almost halved compared to a single link connection (one sta, one ap). 
I have tried to change the packet tx rate in the sta menuconfig from 100 to 200 but the receivers are now receiving even fewer data.
I have seen that you have managed to set up two sta with one ap in your papers, is there any changes I need to make in the mentioning, to maintain the same amount of data for 2 links compare to the single link?

Thank you very much