StevenMHernandez/ESP32-CSI-Tool #68: use csi for location

xiaokangkang6, 2022-10-12:
Excuse me, how can I use these data for indoor positioning?  Could you give me any suggestions please?

StevenMHernandez, 2022-10-17:
Thanks for your interest in this project. Unfortunately I will not be able to give you specific steps to follow (there are so many ways to go about performing localization). 

We do have a new paper that discusses localization and surveys many of the required signal processing techniques. The paper is titled: ["WiFi Sensing on the Edge: Signal Processing Techniques and Challenges for Real-World Systems"](https://www.people.vcu.edu/~ebulut/COMST22_WiFi_Sensing_Survey.pdf). It might help give you some ideas for your project.

Outside of that, I would recommend reviewing a few other localization papers and figure out which techniques are most popular. Then you can begin researching how to perform those specific techniques.

Since this question is not related to the CSI data collection tool I am going to close this issue for now. However, if you have any other questions about the tool or find any bugs, please feel free to comment again. Thanks.

xiaokangkang6, 2022-10-18:
Thank you very much for your reply. I'm just using the data collected by your method for fingerprint location. Thanks again for your help！