StevenMHernandez/ESP32-CSI-Tool #71: How to plot the collected CSI data

Gowtham-Gorajana, 2022-11-04:
help me on how to plot the CSI data and extract the csi data into a csv file.

StevenMHernandez, 2022-11-04:
Please see the documentation for `serial_plot_csi_live.py` in the `README.md`. This script shows how to read CSI and plot amplitude. The script was added to the codebase two days ago.

Gowtham-Gorajana, 2022-11-09:
can you please explain how we have to do using two esp32s
it is given in the scrippt for a router and a esp32 but not for 2 esp32s

StevenMHernandez, 2022-11-09:
I am not sure that I understand. Where does it mention "router"? 

Both ESP32 TX and ESP32 RX can print CSI. Thus we can use `serial_plot_csi_live.py` for both TX and RX.
 
Please review the [README.md](https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/master/README.md). If there are any steps that are confusing, please let me know.

Gowtham-Gorajana, 2022-11-10:
while assigning certain configurations to the esp using the command idf.py menuconfig there can certain configurations which are not shown in the way u mentioned like Custom UART and the max val of FREERTOS is 1000 and we cannot make it more than 1000 . 
how to proceed in such type of situations

Gowtham-Gorajana, 2022-11-10:
and also while flashing visualizing and plotting the data where do we have to run the commands , like in which specific directory whether it is active_ap or active_sta?
i am totally new to this field and i am having some basic doubts as i could not get hold of the exact theory related to the work so hope u dont mind me asking simple doubts :)

StevenMHernandez, 2022-11-10:
>while assigning certain configurations to the esp using the command idf.py menuconfig there can certain configurations which are not shown in the way u mentioned like Custom UART and the max val of FREERTOS is 1000 and we cannot make it more than 1000 .

Can you share a screenshot of the differences? For `Component config > FreeRTOS > Tick rate (Hz)` the value should be exactly `1000` so there should not be any problems there. For the custom UART, make sure you set `Serial flasher config > 'idf.py monitor' baud rate > Custom Baud Rate` first before setting the custom baud rate.

Also make sure to consider this notice in the README:

> NOTE: For some systems, baud rate 1552000 does not work. Good alternatives to try are 921600, 1000000, 1152000, and 1500000.
The higher baud rate the better! Baud rate is extremely important to achieve high sampling rates without lag!
If you have a problem, please leave any relevant information such as operating system, esp-idf version, list of all baud rates work and baud rates that do not work etc in https://github.com/StevenMHernandez/ESP32-CSI-Tool/issues/5.

Gowtham-Gorajana, 2022-11-10:
i am using a single laptop and connected it to 2 esp32s and then opened 2 esp-idf windows one for active_ap and other for active_sta .
do i have to simulatenously run the programs ?

Gowtham-Gorajana, 2022-11-10:
after running the command idf.py flash monitor in active_sta i got this 
[image]

StevenMHernandez, 2022-11-10:
From the README documentation:

> Important: It is important that you are able to successfully build and flash the example project from the esp-idf guide onto your own esp32.

(1) Were you able to do this, and do you understand how it works?

(2) Show the full output with active_sta. Did you see any earlier error messages when the esp32 was being flashed?

(3) Try flashing the active_ap in a separate terminal window. Do you see anything?

Gowtham-Gorajana, 2022-11-10:
@StevenMHernandez thank you for the suggestions i can now print the csi data 
But i am having troube colecting it into a csv file and plotting the data .
[image]

Gowtham-Gorajana, 2022-11-10:
@StevenMHernandez i have plotted the data but the graph goes on continuosly and even while collecting the data into the csv file for how much time duration we have to keep it running so that we get the required data.

Gowtham-Gorajana, 2022-11-13:
what can we conclude from the data

Gowtham-Gorajana, 2022-11-13:
For plotting the data the given file is not working can you please give any other tool on how to plot the above data

StevenMHernandez, 2022-11-14:
You mentioned that plotting worked previously. What changed?

Remember: I cannot read your mind and I cannot see what you have tried. You must add enough information so that I can help solve problems with this project. 

You can plot the data with any software like Python (which was already mentioned above) or other software like MATLAB, however you will have to build this code yourself. Please understand that this project assumes that you have some understanding of how to write your own programs.

AsciiPro99, 2022-11-14:
If you have used the CSI parse present in ./python_utils this code will plot the data, not in realtime.